Sloeber/arduino-eclipse-plugin #290: moving to mars

jantje, 2015-07-27:
As mars compiles again. Why would we move to mars?
Why not?
Anyone has a point of view?

dniklaus, 2015-07-27:
Please make at least one release with Luna before getting interstellar - ehem - before going to mars :-)

jantje, 2015-07-28:
@dniklaus 
good argument

misteralex, 2015-08-09:
Hello,

I designed XDAQ as comprehensive ecosystem for Arduino C++ users that includes powerful scientific tools like SciPy and other interesting products like Plotly for web data visualization over Debian and Ubuntu.

Currently XDAQ includes Eclipse Luna release and it works fine. Anyway in these last weeks I had an issue to build Arduino boards from Eclipse.

I fixed the issue updating Eclipse setting environment variables as follow:

From Eclipse > Project Properties > C/C++ Build > Environment:

1) Add variable A.RUNTIME.TOOLS.AVR-GCC.PATH  with value ${A.RUNTIME.IDE.PATH}/hardware/tools/avr
2) Change value of variable A.TOOLS.AVRDUDE.CMD.PATH to ${A.RUNTIME.IDE.PATH}/hardware/tools/avr/bin/avrdude
3) Change value of variable A.TOOLS.AVRDUDE.CONFIG.PATH to ${A.RUNTIME.IDE.PATH}/hardware/tools/avr/etc/avrdude.conf
4) Change value of variable A.COMPILER.PATH/value to ${A.RUNTIME.IDE.PATH}/hardware/tools/avr/bin/

Fixing these variables Arduino C++ works fine as usual. Very nice plugin!

Please let me know more about it @jantje. 

Thanks

Alex

jantje, 2015-08-09:
Hi Alex
Nice to see you like the plugin. 
I'm however a bit unsure about what your message is.
This issue is about moving to mars. I'm not sure whether you are making any statement, comment, thought about moving to mars.

Secondly the work around you describe gives me the impression you are running Arduino IDE 1.6.5 without the nightly. With the nightly these changes are not needed.

Please update me on your intentions.
Best regards
Jantje

misteralex, 2015-08-10:
Hi Jantje

Currently XDAQ works with Arduino nightly (http://arduino.cc/download.php?f=arduino-nightly-linux32.tar.xz) and now latest release is 1.6.6.

How to fix the issue without change manually that Eclipse variables?

This happens with Eclipse Luna so, I thought was interesting also for mars.

Thanks

Alex

jantje, 2015-08-10:
I do not support the Arduino nightly. You need arduino IDE 1.6.5 and the nightly from arduino eclipse plugin for this to work.
The latest version is known to have a issue but that one is different from your issue https://github.com/jantje/arduino-eclipse-plugin/issues/294

misteralex, 2015-08-10:
I fixed the issue. It was related to the project not to the plugin or Eclipse context. I reinstalled the environment (Eclipse+Plugin) and all work as expected. XDAQ will support Mars without any restriction. Thanks.

rlogiacco, 2015-09-08:
I will expand the concept @dniklaus mentioned to suggest a more frequent release: the nightly builds are quite stable so I would suggest to create a team of voluntary testers who will vote for a release and when they reach consensus a new stable is released.
Using the current approach we will probably never get stable releases.

I propose myself as tester for a 64bit setup Windows 7 + Mars + Arduino IDE of your choice. If necessary I can setup a Win32 virtual machine with a similar configuration.

jantje, 2015-09-08:
@rlogiacco 
txs (again) :-)
There are actually 3 problems right now for making releases.
1 is lack of voluntary testers (thanks to be a volunteer) 
2 is that I still do not have a good (enough to spend little time) graps on how to make a release. The biggest problem here is the V2 update site. The second biggest is the release creation in github.
3 is the fact I don't need the stable release and given the support from the community (responses on requests for testing, donations to make release, and now visible at patreon) it looks to me the community is not "really interested" as well.

All these things make that I'm constantly looking at how much time I invest in the community (living with the drawback of not having a relatively new release) to how much time it would cost me to do a release.

Some figures in August 2015 the remind page of the nightly has been shown 1931 times and from the stable 2232 times. Looks like more than 50% is still on the more than 1 year old stable version.

As to the stability of the nightly. There have been some major hick-ups in the nightly (I hope the latest build fixes #301 and #302 which make the plugin unusable). The idea of a stable version is to spare newbies from bad starting experience due to this kind of hick-ups.
In other words I do not want to replace the nightly by the stable. I make mistakes and I do not want more stress than I already experience for the nightly. 

To conclude:
Yes it is -long due- time for a release. I simply do not get motivated enough to give it priority.

Best regards
Jantje

rlogiacco, 2015-09-08:
:smile: I do understand you perfectly @jantje and I'm using the nightly as well, but OSS is a looping world: if you get better releases you get more attention which brings you more contributors which allows you to release more often, better code and more features.
You have done a lot so far and I'm not asking you to do more, but may be somebody can step up and take the release process out of your shoulders.
As usual I will be delighted to do it for the project, but I can't do both testing and release so we'll require somebody else to take my place as a tester.
I have an update site for another open source project, which means I have an idea on how the thing works (more or less) and I know Github enough to be able to perform a release and share the binary.

So, if you agree, I can take over the release burden: do we have volunteers for the testing team? 

How can you reach your user base so to ask for volunteers?

dniklaus, 2015-09-08:
I'm willing to do some testing:
* on Win 7 64bit
* on Win 7 32bit
* on Linux 32bit (virtual machine)
Do we have some written testcases? Otherwise I'll write some as a start.
Who else would be able to do tests on other system configurations?
* Mac OS X ?
* Linux 64 bit ?

jantje, 2015-09-08:
>As usual I will be delighted to do it for the project, but I can't do both testing and release so we'll require somebody else to take my place as a tester.

cool :-)
I added you and @wimjongman  as collaborators.

As to testers: I post on the arduino forum with requests for testers. Like I did here: http://forum.arduino.cc/index.php?topic=79595.msg2366886#msg2366886 and I tweet about it.
For the boardsmanager change I tweeted and blogged about it but I don't think that had any outcome. Though as visible in the issues, there were some serious issues.

In the past I have included people who tested before in the test mails.
I will trace back the people who have tested in the past and I'll send the list to you.

The V2 issue is as follows: Having an existing update site (on my server of which I can send you a copy) How do you update the update site with a release.
I mean for the product I simply rename the "tested and approved version" file adding a version number. The nightly gets updated every night by jenkins to the latest (old versions are deleted) but for the V2 update site I want all the old releases to remain available. I have no clue to what that process looks like.

dniklaus, 2015-09-08:
Great! I'm pleased being a collaborator now!
Having a list of potential testers who could help would be great!

jantje, 2015-09-08:
@dniklaus 
>Great! I'm pleased being a collaborator now!

I actually added @rlogiacco but as you are so pleased ... I added you as well.
:-)

dniklaus, 2015-09-08:
Ok, sorry for misunderstanding this:-) But, nevertheless...

jantje, 2015-09-08:
A lot of good things happened by mistake ;-)

rlogiacco, 2015-09-09:
Good!!!
Would it be good to have a google group dedicated to users so we can reach out the user base more easily?
Can we use the plugin reminder to ask the users to join the test team? Something like:

**NEW stable release incoming** 
*We need your help, join our test team!*
*If you are willing to dedicate some time to the project, we need help to verify everything works nicely: a few minutes of your time can make the change: register here.*

The link might direct the users to a Google Forms page collecting environment information and email addresses.

Does it make any sense?

jantje, 2015-09-09:
@rlogiacco 
It does make sense to me :-)
I'll work on the list of former testers tonight.

misteralex, 2015-09-10:
@jantje
I already appreciate your product with Debianinux and currently I am working on XDAQ, a comprehensive ecosystem Arduino (nightly release based) over C++/Eclipse context. This plugin is part of my default toolchain. It also include ArduinoUnit as well as other open source products.

From my experience arduino-eclipse-plugin works fine on Debian (Wheezy, Jessie) and Ubuntu (release 14.04 and 15.04).

I also implemented two small tools (xdaq-eclipse-import.py and xdaq-eclipse-make-ino.sh). I found useful integrate xdaq-eclipse-import.py to update version of already existing Arduino Eclipse projects to local version of Arduino toolchain (e.g. from 1.6.5 to 1.6.6).

So, currently I could collaborate as tester for Debian and Ubuntu (Linux 32bit on virtual machine).

Thanks

rlogiacco, 2015-09-10:
@misteralex would you mind submitting this google form? I'm trying to collect info for all the testers.
http://goo.gl/forms/pxHR7XrCuA

Obviously anybody willing to contribute is welcome to fill the form :smile:

misteralex, 2015-09-10:
@rlogiacco
Done!!!

Thanks :)

pep, 2015-09-12:
I am a new user of the arduino-eclipse-plugin and with little concise information available as to what version of os/eclipse/arduino-ide/and arduino-eclipse-plugin works with what I went ahead naively and loaded the latest of everything. ie win10 64bit/eclipse mars 64bit/ arduino 1.6.5/ nightly arduino-eclipse-plugin updated from within eclipse. 

I also intended to do esp8266 development primarily to have easier access to libraries via eclipse rather than using the Arduino IDE which I have projects working fine using the arduino-esp8266 libraries. I found Neil Kolban's Eclipse Cheat Sheet pdf.

**** The main issue I currently have is with Eclipse not finding include files mainly like the issue: https://github.com/jantje/arduino-eclipse-plugin/issues/10, as has been mentioned in your blog and other people much earlier before going back to early version of the Arduino IDE and the eclipse plugin. Do you have any idea how to resolve this problem after the removal of windows->preferences->C/C++->indexer->Files to index upfront back in earlier versions of eclipse.

Additionally, I think it would be good to have some sort of a matrix that showed what environmental configuration is know to work well, ie tested and what isn't yet tested. Also alot of the FAQ and blogs entries, flow charts etc are out of date and no longer relate to current software versions. Everything is all over the place.

I thought I would put this here because it seemed to be a recent thread, you were taking about Mars the latest version and where to go from here, and I don't understand the issue process, how to submit issues etc and i would be good to have a clear description of how to contribute to the project and not waste peoples time. I would also like to be a tester on the above mentioned configurations.

Regards,
Phil.

jantje, 2015-09-12:
Ho Phil
Most of what you ask is available on http://eclipse.baeyens.it
The info there is not outdated.
But if you use the nightly you can not really expect updated info in a nice process-able way.

As to what runs with what. That is the toughest question. I made a patreon page about that.

Issue 10 should no longer exist. So maybe something is wrong on your system. Do you have the very latest nightly? #301 and #302 are real game spoilers.

Best regards
Jantje

rlogiacco, 2015-09-14:
@pep want to lend an hand and participate to testing for a new release or improving the documentation? We are currently looking for help: http://goo.gl/forms/pxHR7XrCuA

pep, 2015-09-15:
Roberto,
I already filled in the questionnaire for 2 environments, just let me know
how I can help.
Regards,
Phil.

pep, 2015-09-15:
Thanks @jante for your reply, I hope to get some time today to try to get things working and follow your suggestions. I am happy to be a patreon in some way if I can get the Eclipse Arduino plugin working effectively. I appreciate all your hard work and understand the difficulty in configuration management with so many changing dependencies. Regards, Phil.

jantje, 2015-09-16:
There are about 3500 builds a day with the plugin. That is quite a lot of people using it. 
Basically this means: If it doesn't work quickly there is something wrong.

pep, 2015-09-16:
@jante,

I deleted my previous eclipse installation that I had installed the plugin
into and updated.
I downloaded your win64.2015-09-08_02-06-02.tar.gz file and extracted the
tar twice with 7zip.
I loaded the latest 1.6.5-r5 Arduino IDE, got the warning about version
being later than 1.6.5 when setting up the Arduino IDE path. Got a lot of
errors in the error log, so editted the lib/version file to remove the -r5
in it to get rid of the errors.
copied the make.exe from 1.5.7 to the hardware/tools/avr/utils/bin/make.exe
added .ino as a C++ file type so the IDE plugin recognises .ino files,
renaming to a .cpp file worked but this made more sense.
setup a net project with the .ino file from the Arduino IDE project, added
a library to the selected project that was needed.
selected the project
verify and upload the sketch with the com port configured to the esp8266
board.
everything worked

so I think the main issue is that the download has over steps implemented
in it that the eclipse plugin process/update didn't have.
help from neil kolbans cheat sheet helped alot to resolve each issue.

Regards,
Phil.

jantje, 2015-09-16:
Great news :-)
There is nothing special in the product compared to the plugin but it is lots handier to have everything in one bundle and it also makes sure you have the arduino perspective on by default. 
Do you have a link to the cheat sheet that helped you?

pep, 2015-09-17:
@jante,

Neil Kolban wrote a book on the ESP8266 and inside the
http://neilkolban.com/tech/wp-content/uploads/2015/09/Kolbans-Book-on-the-ESP8266-September-2015.pdf
file page 156 there is a section called "Building ESP Arduino apps using
the Eclipse IDE". He did have a separate document somewhere called
Arduino-Eclipse-Cheat-Sheet.pdf I think but I can't seem to find it
separately on google anymore. I think I found it on a forum somewhere maybe
the ESP8266 one? So I attached it to the email. Hope if helps a little....

I would like to learn about eclipse plugins as you did... can you recommend
any resources that you have come across that a seminal to learn eclipse
programming quicker?

How do you know how many Arduino Eclipse plugin builds are done per day?
Does the plugin have an internet monitor built in to send a message on each
build or something?

Regards,
Phil.

rlogiacco, 2015-09-17:
@pep each time a build is performed a counter in the plugin is updated, the same used to show up the remider popup which appears every now and then. The content of the reminder popup is hosted on Jantje's web server: you should be receiving an updated reminder page starting from yesterday which asks people to submit the form I've proposed above.
So the short answer is _yes, it is remotely monitored_.

jantje, 2015-09-17:
I remember neil Kolban now. I helped him out with the arduino eclipse part of the book, the link to the book is indeed on the forum

remotely monitored sounds heavy. When you do a build a http request is done for a non -existing web page on a web server on a low priority thread. The logs of the server tell me the hit count of the page and so I know how many builds are done. The same counts for opening the serial monitor, opening the scope and uploading to a board.

jantje, 2015-12-11:
The product now compiles with mars.

closed 2015-12-11 by jantje