Sloeber/arduino-eclipse-plugin #293: No support for assembly files in arduino/core

nkolban, 2015-08-05:
I am trying to use the Arduino Eclipse Plugin with the ESP8266 Arduino package.  This package includes a file called "cont.S" which is source assembly code.  It appears that the arduino-eclipse plugin doesn't scan for ".S" files to include in its Makefile rules for building the core environment.  I am hoping that we can add support for both .S files and ideally ... test with the ESP8266 package and boards.

jantje, 2015-08-05:
great an assembly tester :-)
I added assembly support in issue #121 but as I do not have any assembly experience and the original requester never tested it I'm not sure it works.
The fix is from 14 jan 2014 so any release after that should pick up the .S files.

nkolban, 2015-08-06:
You are awesome!!  I downloaded the latest nightly and it worked perfectly.  I had been using the stable.  I'll be curious to hear when the next stable will be shipped.  I am now testing YOUR arduino-eclipse support with the ESP8266 functions ... I am VERY excited!!

closed 2015-08-06 by nkolban

jantje, 2015-08-06:
cool. Nice to know it works.

About the stable release.... there is quite some work involved in making one, and .... I don't need it. Nor does the community that delivers content. 
The users that need it are the "users only" who can support the community financially. And, financial support is very limited. Because I can say whatever I want about how much financial support I get I recently made a patreon account to make the financial side of this project more visible. That is both how much I get and how much the "users" get.
I put 100$ per month on "Every half year a stable release of the arduino eclipse plugin. (january and July)"
Feel free: https://www.patreon.com/jantje?ty=h

greg-ware, 2015-10-10:
Ah, okay, good to know that stable is not the latest!