Sloeber/arduino-eclipse-plugin #294: Combiner problem (arduino nightly + eclipse mars)@ubuntu 15.04

skip777, 2015-08-05:
Hello,

I've just started to play some arduino with pro mini version and after few successful attempts on my windows machine i've decided to move to linux. I've installed ubuntu 15.04 and eclipse mars edition with arduino nightly build and of course your plugin. The first attempt to build a test project (just to blink the builtin led) failed with the following problem:

1:18:51 **** Build of configuration Release for project Test ****
make all 
Starting combiner
"/usr/share/arduino-nightly/hardware/tools/avr/bin/avr-gcc" -Os -Wl,--gc-sections -mmcu=atmega328p  -o "/home/emil/workspace/Test/Release/sketch/Test.elf"    ./.ino.cpp.o   arduino.ar   "/home/emil/workspace/Test/Release/arduino.ar" "-L/home/emil/workspace/Test/Release" -lm
/usr/share/arduino-nightly/hardware/tools/avr/bin/../lib/gcc/avr/4.8.1/../../../../avr/bin/ld: cannot open output file /home/emil/workspace/Test/Release/sketch/Test.elf: No such file or directory
collect2: error: ld returned 1 exit status
makefile:79: commands for 'Test.elf' have failed
make: *** [Test.elf] Error 1

21:18:52 Build Finished (took 551ms)

The first thought is there's a problem with "sketch" folder where the *.eep and *.elf files should be placed. But unfortunately there is no command to create this folder, and I don't know if there should be such.

Regards,
Emil

jantje, 2015-08-06:
I haven't tested mars yet. 
The arduino nightly is not supported.
Which version of the plugin do you use?

skip777, 2015-08-08:
Hello, 

Sorry for the delay, but I'm still fightin' now with HC-05 bluetooth on ubuntu 15.04... it's kinda tough.
Back to the subject:
1. It seems to be strictly "nightly" problem - after installing Arduino IDE 1.6.5 everything works flawless (using normal FTDI on USB connection)
2. The plugin version I've installed is: 2.4.0.201508050214.

Regards,
Emil

misteralex, 2015-08-10:
Hello

I checked the plugin for Ubuntu 15.04 and Arduino 1.6.6 nightly both for Luna and Mars. It works fine.

So, I will update XDAQ to support Eclipse Mars release without any restriction. The plugin works very well both for Ubuntu 15.04 and Debian (Jessie/Wheezy).  

I suggest to reinstall Eclipse and the plugin (or use XDAQ Setup).

jantje, 2015-08-10:
@misteralex 
I'm sorry but Arduino IDE 1.6.6 hourly will no work with the arduino eclipse plugin.

misteralex, 2015-08-10:
@jantje use XDAQ scripts over Ubuntu or Debian and you will check fastly that Arduino toolchain 1.6.6 + Eclipse Luna/Mars + your plugin works fine !!! 

I checked it !!! It works without issues. Please consider that I am talking about Debian based platforms. I did not check it on Windows.

Here some relevant XDAQ configuration info:
System info            #1 SMP Debian 3.16.7-ckt11-1 (2015-05-24)  
GCC Toolchain       4.9.2                 
Gnome Desktop     3.16.0-1              
Java                       1.7.0_75              
Python                   2.7.10                
Arduino IDE           1.6.6                 
Eclipse IDE            4.5.0 (Mars)

jantje, 2015-08-26:
so Can we close this?

jantje, 2015-09-18:
The Arduino core team was so kind to accept the issue and fixed it https://github.com/arduino/Arduino/issues/3807
So this can be closed.

closed 2015-09-18 by jantje

skip777, 2015-09-18:
OK - let's close this one. Thanks for help and Your time spent over the plugin development. Respect :)