Sloeber/arduino-eclipse-plugin #296: Generated .ino.cpp breaks ESP8266 includes

nkolban, 2015-08-08:
I am hunting a problem.  I am finding that if I create a sketch for the ESP8266 Arduino environment then the following is actually compiled:

    //This is a automatic generated file
    //Please do not modify this file
    //If you touch this file your change will be overwritten during the next build
    //This file has been generated on 2015-08-08 18:33:35
    
    #include "Arduino.h"
    #include <Arduino.h>
    #include "Arduino.h"
    #include "user_interface.h"
    void setup() ;
    void loop() ;
    
    
    #include "main.ino"

The detailed issue is the line which includes the header called "user_interface.h".  In ESP programming, this has to be included in C++ as

    extern "C"

I am not sure which component is responsible for this injection nor how to change it.  Can we look into this to see whether it is the Eclipse Arduino project including it incorrectly or if it is the Arduino ESP generating badly?

jantje, 2015-08-09:
My personal opinion is that it is a user_interface.h problem.
I advice to add following to user_interface.h
```cpp
#ifndef __USER_INTERFACE_H__
#define __USER_INTERFACE_H__
...
#ifdef __cplusplus
extern "C" {
#endif
....
#ifdef __cplusplus
}
#endif
#endif
```

nkolban, 2015-08-09:
Thanks for response.  I am not keen on changing code supplied by other packages.  I did find a perfect workaround (for me).   I renamed my "myfile.ino" to "myfile.cpp" and all worked perfectly.

closed 2015-08-09 by nkolban

jantje, 2015-08-09:
The .ino format has some drawbacks. The biggest is that I need to build the .ino.cpp file which is done as a pre build step. So each time you build the .ino file is compiled even when it hasn't changed.
If you do not want ot touch the file you can always create a my_user_interface.h file that looks like
````cpp
#ifdef __cplusplus
extern "C" {
#endif
#include "user_interface.h"
#ifdef __cplusplus
}
#endif
#endif
````