Sloeber/arduino-eclipse-plugin #297: Plugin causes NullPointerException when building/reconfiguring non-Arduino CDT projects (showstopper!) - but with fix attached

jgow, 2015-08-21:
Firstly - a very big thank you for your time and effort in writing this plugin. Eclipse is my main development environment, I develop across different platforms and it is extremely beneficial not to have to use yet another different IDE for the odd Arduino project...

Summary of issue:
With the Arduino plugin installed, non-Arduino CDT projects throw a NullPointerException when rebuilding/reconfiguring

Environment:
64-bit OpenSUSE with Eclipse Mars, CDT plugin (and many others). Arduino plugin from yesterday's nightly build.

To reproduce:
Install nightly build of Arduino plugin. Create any Arduino project and build it. Then create a different CDT project for a different platform other than Arduino (in my case this was a Makefile project from existing cross-GCC code for an embedded platform). Then try to build the non-Arduino project. The result is an error dialog (see image)
[image]

The logfile shows the exception occurring in 

it.baeyens.arduino.tools.ArduinoHelpers.setTheEnvironmentVariablesSetTheDefaults(ArduinoHelpers.java:519) (see image)

[image]

and stems from the fact that in a non-Arduino project, of course the platform file path is not set at all (as would  be expected) - however the plugin is still trying to initialize Arduino environment variables on a non-Arduino project. Tracing this back, the issue is in the ConfigurationChangeListener for the plugin. This is registered with Eclipse and is being called back on changes to all CDT projects - but the code in the handleEvent member of  it.baeyens.arduino.listeners.ConfigurationChangeListener does not discriminate between Arduino and non-Arduino CDT projects.

The fix:

I modified it.baeyens.arduino.listeners.ConfigurationChangeListener.handleEvent to grab the project type and check to see if it had one of the unique Arduino natures - only proceeding with the initialization if it is an Arduino project: see code fragment below:

public void handleEvent(CProjectDescriptionEvent event) {
	// we are only interested in about to apply
	// if (event.getEventType() != CProjectDescriptionEvent.ABOUT_TO_APPLY)
	// return;
	// if (true)
	// return;
	ICProjectDescription projDesc = event.getNewCProjectDescription();
	IProject project = event.getProject();
	boolean isArduino=false;
	try {
		isArduino=project.hasNature(ArduinoConst.ArduinoNatureID);
	} 
	catch(Exception e)
	{
	}
	
	if (projDesc.getActiveConfiguration() != null && isArduino) {

        // the rest of the function is unchanged.

With this in place the plugin performs almost flawlessly (i say 'almost' - because the make process errors out in the build if .ino files are used. It works perfectly with .c or .cpp files - and in my installation .ino files are registered as a CDT source file extension - but this is a separate issue).

The caveat:
I am not a die-hard Eclipse plugin developer (and certainly not a die-hard Java programmer) so I do apologize in advance if this is not the most elegant solution, or indeed the 'best' one - but it does work and I needed a quick fix to get on with some embedded development. Just thought I would share the fix if anyone is encountering the same problem.

Hope this helps....

jantje, 2015-08-21:
:-) another good example of an issue report.
Indeed the ConfigurationChangeListener should only be called for Arduino projects.
Your fix is what is needed. I'm pushing a -untested- fix

jantje, 2015-08-22:
Have you been able to test the nightly?

jgow, 2015-08-22:
Just have tested it - on a fresh Eclipse install. Looks like the issue has been fixed! Thanks for your help.

jantje, 2015-08-24:
txs for the issue and testing :-)

closed 2015-08-24 by jantje