Sloeber/arduino-eclipse-plugin #299: java NullPointer exception after compile is complete

lakidd, 2015-08-24:
So, when starting the IDE, the workspace loads and all of my projects are there.
I click compile and it compiles fine.
After it finishes I get a pop up w/ 4 Java Nullpointer exceptions about Serializing CDT project settings.

http://imgur.com/euMZBMn

I do have a number of both arudino and esp8266arduino projects open. It doesn't seem to affect anything though...everything keeps working but I'm guessing it's still not good :-)

Details of the error is here
````
org.eclipse.core.runtime.CoreException: 
	at org.eclipse.cdt.internal.core.settings.model.CProjectDescriptionManager$5.run(CProjectDescriptionManager.java:526)
	at org.eclipse.core.internal.resources.Workspace.run(Workspace.java:2313)
	at org.eclipse.cdt.internal.core.settings.model.CProjectDescriptionManager.runAtomic(CProjectDescriptionManager.java:519)
	at org.eclipse.cdt.internal.core.settings.model.CProjectDescriptionManager.access$2(CProjectDescriptionManager.java:517)
	at org.eclipse.cdt.internal.core.settings.model.CProjectDescriptionManager$4.run(CProjectDescriptionManager.java:498)
	at org.eclipse.core.internal.jobs.Worker.run(Worker.java:54)
Caused by: java.lang.NullPointerException
	at it.baeyens.arduino.tools.ArduinoHelpers.setTheEnvironmentVariablesAddtheBoardsTxt(ArduinoHelpers.java:651)
	at it.baeyens.arduino.tools.ArduinoHelpers.setTheEnvironmentVariables(ArduinoHelpers.java:805)
	at it.baeyens.arduino.listeners.ConfigurationChangeListener.handleEvent(ConfigurationChangeListener.java:30)
	at org.eclipse.cdt.internal.core.settings.model.CProjectDescriptionManager.notifyListeners(CProjectDescriptionManager.java:2212)
	at org.eclipse.cdt.internal.core.settings.model.AbstractCProjectDescriptionStorage.fireAboutToApplyEvent(AbstractCProjectDescriptionStorage.java:284)
	at org.eclipse.cdt.internal.core.settings.model.SetCProjectDescriptionOperation.executeOperation(SetCProjectDescriptionOperation.java:66)
	at org.eclipse.cdt.internal.core.model.CModelOperation.execute(CModelOperation.java:341)
	at org.eclipse.cdt.internal.core.model.CModelOperation.run(CModelOperation.java:607)
	at org.eclipse.core.internal.resources.Workspace.run(Workspace.java:2313)
	at org.eclipse.cdt.internal.core.model.CModelOperation.runOperation(CModelOperation.java:638)
	at org.eclipse.cdt.internal.core.settings.model.AbstractCProjectDescriptionStorage.setProjectDescription(AbstractCProjectDescriptionStorage.java:203)
	at org.eclipse.cdt.internal.core.settings.model.CProjectDescriptionStorageManager.setProjectDescription(CProjectDescriptionStorageManager.java:149)
	at org.eclipse.cdt.internal.core.settings.model.CProjectDescriptionManager.setProjectDescription(CProjectDescriptionManager.java:885)
	at org.eclipse.cdt.internal.core.settings.model.CProjectDescriptionManager.setProjectDescription(CProjectDescriptionManager.java:846)
	at org.eclipse.cdt.internal.core.settings.model.CProjectDescriptionManager$6.run(CProjectDescriptionManager.java:573)
	at org.eclipse.cdt.internal.core.settings.model.CProjectDescriptionManager$5.run(CProjectDescriptionManager.java:523)
	... 5 more
````

jantje, 2015-08-24:
I had this problem and then it went away.
If I remember correctly it is caused by projects not properly configured (read the arduino board is bonkers)
A nicer message (saying so) would be good.

jantje, 2015-08-24:
I said may fix ...

lakidd, 2015-08-24:
No probs, nice to know what's causing it. I went thru my open projects and 'applied' the configuration and it's gone away. Thanks !

jantje, 2015-08-24:
Then it will be "fixed #299"

commit 67100d55 (2015-08-24): Log warning in case of not found board. May fix #299
closed 2015-08-24 by jantje