Sloeber/arduino-eclipse-plugin #301: 1st time user feedback

barrystaes, 2015-09-03:
Some feedback, and a problem i encountered while trying the plugin for the first time.
(on Linux 32bit with eclipse Kepler already installed)

Documentation on website:  Way to many short pages. After a lot of headscratching i found out that what i'd expect (given the menu options) to be at http://eclipse.baeyens.it/how_to.shtml#/c was instead on page http://eclipse.baeyens.it/how_to.shtml#/c

Creating a new project resulted seemed to work, but Finish gives a `URI is not  hierarchical` error dialog. I have no idea how to troubleshoot that..

Explaining how to open an existing project, or import sources, or can i just open and program it?

jantje, 2015-09-04:
Hi
Thanks for the feedback. I'm sorry you had so much trouble getting things to work. You are right that when you know how to do things it is easy but when you are new it is so easy to get lost in the forest.
Being new to eclipse it is hard to understand what is going on.
Anyways. What I understand from your comments is 
1: It is not always clear what is in the nightly and what is in the product.
2: You prefer long html pages to the short versions.
3: You got an error "URI is not hierarchical" in the nightly which I have never seen before.
4: It is not always clear where to find the information.
5: Compilation is slow on your laptop.

I accept all there remarks but there is little I can/want to do about it. Part of it is related to open software part of it is related to eclipse and part of it is related to the shortcommings of the arduino eclipse plugin community part of it is related to mistakes made in the past.

Practically I can not advice you on 32bit versus 64 bit on windows as I mainly run on Linux. Linux does not have the compatibility mode 32bit running on 64 bit os. It is unclear to me why you switched from Linux to windows in your description.

V2 windows 32 bit got lost in the mist of times. It was not a decision, it just happened.

I'll try to see whether I can reproduce the "URI is not hierarchical" error.

The problem with clicking finish again is "normal". Basically the project was created and I prefer having a wonky project where "something went wrong" to deleting the project when something goes wrong making it impossible to create a project when the "something goes wrong" is consistent.

V2 is getting really old. Basically due to lack of sponsoring it is hard to motivate me to make new versions. Basically right now the nightly is the way to go. I made a patreon account so it is visible what I get for all the work I put in the community. You will bump into it one day.

"build before upload", " the documentation mentioning i should not enter port confused me here" are both nightly only.

importing is a cdt and eclipse functionality (as such documented partly in the cdt documentation and partly in the eclipse documentation). It can be done in many ways. My preferred way is drag and drop. Note that you can link or copy. The chose is up to you depending on your setup.

>!ENTRY it.baeyens.arduino.core 1 0 2015-09-04 01:27:46.397
!MESSAGE The folder /home/barry/.arduino15/packages does not contain any files.

means you have no packages installed on your system. Which is weird as you are talking about due which needs a package. It is a simple warning and normal if indeed no packages have been installed. It may be related to the "URI is not hierarchical" problem. Note that the boards manager support is brand new.

Best regards
Jantje

barrystaes, 2015-09-05:
**URI is not hierarchical**
I've seen the "URI is not hierarchical" error on both Linux and Windows. Latest Arduino IDE with your NIGHTLY and V2.
Linux since works, but i dont remember how/why exactly, but i think because i used the V2 product download there.
As a last resort for Windows i just now tried the latest nightly http://eclipse.baeyens.it/download/product/win64.2015-09-03_02-06-01.tar.gz with the latest but pre-1.5.6 Arduino IDE http://arduino.cc/download.php?f=/arduino-1.5.5-r2-windows.zip and again, consistently the same error. Its Windows 10, if that matters. Running eclipse as administrater did not help.

**V2 vs NIGHTLY**
So V2 is deprecated, and NIGHTLY can be regarded as stable?
I suggest to at least mention that on these pages on V2, saving people new to this a lot of trouble;
- http://eclipse.baeyens.it/installAdvice.shtml > V1?V2? Product? Eclipse
- http://eclipse.baeyens.it/installAdvice.shtml > V2
- http://eclipse.baeyens.it/how_to.shtml#/c > Install V2 plugin
- http://eclipse.baeyens.it/how_to.shtml#/d > Install V2 product
Also, these URLs illustrate how the install info is scattered over 2 URLs and 4 pages.

I switched to Windows because that PC's hardware is better than my old Linux laptop, which noticeably struggles doing code completion or even compile an empty project.

barrystaes, 2015-09-05:
About the Arduino path being forgotten - I now think its because its stored in the workspace?

I routinely try a new workspace because the docs said somewhere that i should not uncheck the "use default location" option when creating a new project, since its not supported by the plugin.

I am still trying to create a new eclipse project with my own source files in there.

barrystaes, 2015-09-05:
**Opening existing project**
Another thing that i could not wrap my head around, is how to open an existing project in Eclipse. It turns out they can be imported, by choosing File > Import Files > General > Existing Projects
I put this out here so other new users can benefit from it. I'd appreciate an "open project" option very much, but i take thats the turf of eclipse.

jantje, 2015-09-05:
>V2 vs NIGHTLY
So V2 is deprecated, and NIGHTLY can be regarded as stable?
I suggest to at least mention that on these pages on V2, saving people new to this a lot of trouble;

Currently the nightly is pretty stable as I'm working towards a new release. http://forum.arduino.cc/index.php?topic=79595.msg2366886#msg2366886. So it is not a general statement.
As you experienced there are still "unknown" problems.

>Ah, settings are stored in Eclipse workspace
About eclipse sometimes forgetting the Arduino path - i've now seen this occur on Linux also, and now think its because its stored in the workspace? Turns out it forgets every setting.. including the perspective, file types, and even down to the editor font size. I had no idea that eclipse works that way, sorry!

Yes and that is intentionally. This allows to have a workspace for each arduino ide version you still use. Otherwise multiple eclipse installations would be needed.

>"use default location" 

You can uncheck it in the nightly. Though I do not really see why you would want to do so.
With the nightly you can create/use projects outside of the workspace. With last stable you can not.

>But all in all this tries moving your project inside the workspace folder, so you're not actually opening the other project, but creating a copy in a different location. To save things in VCS you'll have to copy files, or put a workspace folder inside your VCS apparently.

Somewhere in the prosess you can say "link to files".

jantje, 2015-09-06:
I removed the folder /home/jantje/.arduino15 from my system and created a sketch and all went fine.
I currently see no way how to reproduce this issue.
Can you try to get traces or so on this issue?

barrystaes, 2015-09-06:
Ok. I know of no log files expect the one with errors i quoted.

To trace this plugin error, do i need to build it myself from source, or can i just enable something in my current eclipse installation?

jantje, 2015-09-06:
Normally errors come with a trace and if there is a trace it is also logged in the error log 
See http://eclipse.baeyens.it/faq.shtml#/troubleshooting "How can I find out more of the problem at hand?"
You could try debugging. you can attach a debugger to a running eclipse but it is lots easier just to compile and debug. See the readme file on this page on how to do so https://github.com/jantje/arduino-eclipse-plugin
I have tried several scenario's to reproduce but I only get #302 which will need fixing before the release.

jantje, 2015-09-07:
I found this is related to #302 
Basically it works fin in a development/debug environment but not when the plugin is deployed as a jar file. I do not understand why the error is inconsistent "URI is not hierarchical" versus "setup corrupt"
I think I will disable this code for now and think for a good solution.

barrystaes, 2015-09-14:
Sorry for the late reply. I wanted to say i'll try to look into this wednesday, but it appears you've already cracked it. I no longer have a working Due at hand though, i let the magic smoke escape. :zap: :dash: 

If you'd like me to test the plugin by then, do let me know!

jantje, 2015-09-14:
No problem.
We are looking for testers https://docs.google.com/forms/d/12laG01sOcWG23Zaixur4v31MFcZ66ObJL_8XvRVgZZA/viewform?c=0&w=1

closed 2015-09-14 by jantje