Sloeber/arduino-eclipse-plugin #303: Missing LINK_OBJ for .ino and .pde sources in subdir.mk

Louis2016, 2015-09-14:
Hi,

With nightly build 2.4 and every other version, it isn't possible to build project with .ino or .pde files.
Even if .ino and .pde are declared as C/C++ files, they are ignored by make.
subdir.mk is created but it lacks an "LINK_OBJ" entry for the corresponding .o object file and thus the compilation command is never executed.

Regards

jantje, 2015-09-14:
It works fine on linux.
Have you read this blog on how it works?
I explain that .ino and .pde files are not processed in the make file and how it works.

Louis2016, 2015-09-14:
Yes. But .ino.cpp is not created. subdir.mk is created with .ino not .ino.cpp. LINK_OBJ is missing. As a result, .ino are not compiled.

Workarounds:
- renaming .ino to .cpp works
- adding LINK_OBG to subdir.mk and unsetting automatic makefile generation also works

So?

Louis2016, 2015-09-14:
All tests under window 7 32 bits.

jantje, 2015-09-16:
just wondering
Are you using the cdt build command or the arduino verify command?

Louis2016, 2015-09-16:
I'm using the CDT build command (didn't know about the arduino verify command, will try it and let you know)

jantje, 2015-09-29:
Now I remember this is a duplicate of #191

closed 2015-09-29 by jantje