Sloeber/arduino-eclipse-plugin #304: Stable version 2.3 is unusable as it

Louis2016, 2015-09-14:
Hi,

The stable 2.3 version which is proposed by default to all newcomers is unusable, at least in Windows:
-  the environment variable "A.RUNTIME.TOOLS.AVR-GCC.PATH" is not defined!
- so any compilation will failed due to "/bin/avr-g++" not found

Regards

jantje, 2015-09-14:
Indeed the nightly does not support 1.6.5
Here is a list of issues that have been fixed in the nightly
https://github.com/jantje/arduino-eclipse-plugin/issues?q=is%3Aissue+is%3Aclosed+label%3A%22status%3A+fixed+in+nightly%22
This is an impressive list as there hasn't been a release in a long time.
@rlogiacco is setting up a test to make a new release. Maybe you can yoin the test effort.

Louis2016, 2015-09-14:
The default version that's offered for download is 2.3. Then, it just doesn't work.

Coming to nightly isn't that natural.

Leaving it like that will just push away most newcomers... The idea is great but with as it doesn't work at first, many potential users will just switch to another solution...

Do you think it makes any sense to offer a non working version as default download?

Louis2016, 2015-09-14:
Ok. I think I got the point now. (Let's use only .cpp for now as .ino support is broken, issue #303).

Eclipse Arduino 2.3 + arduino IDE 1.5.5_r2:
- It works: environment variables are properly defined and "make" is present

Eclipse Arduino 2.3 + arduino IDE 1.6.5:
- "make" is missing
- A.RUNTIME.TOOLS.AVR-GCC.PATH is missing

-> Fix for Arduino IDE 1.6.5:

- copy "make.exe" from 1.5.5 into {1.6.5 path}/hardware/tools/avr/utils/bin/
- define A.RUNTIME.TOOLS.AVR-GCC.PATH to "${A.RUNTIME.HARDWARE.PATH}/tools/avr"

Regards.

rlogiacco, 2015-09-14:
Hi @Louis2016 and thanks a million for your feedback! 
We are setting up a test team to prepare a new stable and some help on the website documentation is very welcome.
I you want to join us please fill the following form http://goo.gl/forms/pxHR7XrCuA or submit a pull request for any documentation improvement you think might help get people onboard.

jantje, 2015-09-15:
>Coming to nightly isn't that natural.

I fully accept that. On the other hand.... making a release for a community that doesn't show appreciation is neither.

jantje, 2015-12-07:
As 2.4 has been released I'm closing this one

closed 2015-12-07 by jantje