Sloeber/arduino-eclipse-plugin #306: Remove the ArduinoIDE dependency

rlogiacco, 2015-09-16:
Since most of the confusion and issues are related to the many possible mixtures of Eclipse, the plugin itself and the Arduino IDE, I wonder if we could remove the Arduino IDE dependency and bring straight into the plugin all the necessary dependencies. I believe this solution, which I'm not sure if it is viable, could potentially:
* solve most of the troubles for newcomers with a much simpler installation process
* remove some configuration steps and installation path related problems
* remove most of the cluttering code which deals with path separators file system related
* reduce the amount of work which is brought into this project for each ArduinoIDE release

Most of the changes that are being applied to the ArduinoIDE do not really have impact on the plugin, they are changing the way they detect boards, changing the toolchain, adding library management, etc..
We are not leveraging that code and we do not benefit from the new features, but we strive to keep contact with the latest version because it's what most of the people download. If we embed what we need, we can decide what and when we want to bring in a feature and pull an update... If that doesn't occur, we should have no problems. What really needs to be in sync are a few files, for what I understand: boards.txt, libraries and includes.

Is my reasoning correct?

jantje, 2015-09-16:
I like the idea but there is a bit more than boards.txt, libraries and includes.
The arduino IDE also delivers the toolchains.
If we find a nice way to bundle the avr toolchains (I would limit the toolchain to avr -at least for a start-) and a make that would make for a great "getting started product".
I'm interested in hearing @wimjongman his opinion on this.

wimjongman, 2015-09-16:
What exactly do you mean by "Remove the ArduinoIDE dependency"? 

Do you propose to get rid of the IDE and let people install the plugins in their own whatever flavor they are using which in turn pulls in all relevant dependencies?

wimjongman, 2015-09-16:
"Most of the changes that are being applied to the ArduinoIDE do not really have impact on the plugin, they are changing the way they detect boards, changing the toolchain, adding library management, etc.."

We only make changes to the plugin. We never make changes to the IDE which, IIRC, is just a copy of the C IDE.

jantje, 2015-09-16:
Hi Wim
What Roberto is proposing is to "add the arduino IDE content we need to the plugin"
As such we would have a product that contains everything but can only run arduino IDE X.Y.Z
This would remove the need for configuring the plugin.
Jantje

jantje, 2015-09-16:
The first step could be to add make and avr to the plugin
for avr we need the boards.txt and the platform.txt and the avr toolchain.
As far as I can remember the plugin doesn't need anything else from the arduino IDE install.

jantje, 2015-09-16:
Och yes the plugin also uses the cores and the libraries.

wimjongman, 2015-09-16:
You mean remove the need to separately install the Arduino IDE from Arduino right? Yes that would be cool. I am sorry I thought you were referencing to our IDE when you said Arduino IDE.

rlogiacco, 2015-09-16:
Thanks @jantje for clarifying my point of view: you nailed it.
Yes @wimjongman, I was suggesting to get rid of the separate install and bring inside what we need, the bare minimum. 
I would bring in the contributed libraries as well (OneWire, SoftSerial, etc...), that should fix tons of issues and reduce the amount of support requests: we will be able to publish a fully functional build system with no external dependencies, which is the root cause of most of the problems and newcomers bouncing away.

With regards to the avr toolchain, that's correct @jantje, that will be needed along with the make command. I'm wondering: if the ArduinoIDE is not including it anymore, what did they use to replace it? A Java implementation? If so we can switch to use that and remove another dependency. If not viable we can embed it in the plugin...

I don't have all the info, but I believe if we will not need to create command executions by concatenating strings, most of our troubles and code clutter should go away...

jantje, 2015-09-16:
Don't be sorry. The title is not 100% clear but we are still at 0.01 ;-)
So we all agree it would be cool.
As arduino is having the boards manager I think we should keep all the plugins and products we have and add a starters avr only product.
If we could put the avr toolchain, core and libraries in a plugin that would make release management a lot easier.
if we could add make to the core plugin we would have: 1 product download is enough for arduino avr based boards.

thinking about this Dough has been working on his arduino eclipse plugin and has support for the board manager files. It may be a good idea to look on how he did it. if we process the boards manager files ourselves at least we get rid of the .arduino15 folder

thinking more about this I think we should think more about this ;-)

jantje, 2015-09-16:
Warning brain dump
suppose we support boards manger formatted files 
1) We should not have platform specific issues as they should be solved. 
2) We could dump the files in the eclipse installation so we get rid of .arduino15.
3)We could offer a list of "plugin approved boards manager files" for novices and "any file for advanced users.
4)We could check at startup of the plugin to see whether boards are installed and if no boards are installed we simply show the add boards as described above.
5) we do not need to point to a Arduino IDE.
6) If we fix the make issue we would have a product that works right after download and unzip.

if we could reuse dough his code this should be relatively easy and fast to implement.
If we fix the make issue
end of warning braindump

anyone knows a pointer to dough his arduino plugin?

jantje, 2015-09-16:
found the link https://github.com/eclipse/cdt/tree/master/toolchains/arduino

Louis2016, 2015-09-16:
Very clever move, indeed.

What is required:
- avr toolchain
- make (don't know if internal CDT build can handle this instead of make)
- arduino core
- 'variants" board pins
- compile flags (inspired from boards.txt and platform.txt)
- avr libs
- optionally arduino bootloaders for advanced users
- access to the arduino libraries from the community

What would be cool too is to separate the core and avr libs builds from the user projects builds (dependency projects)

bertenvdb, 2015-09-16:
> solve most of the troubles for newcomers with a much simpler installation process
> remove some configuration steps and installation path related problems
As This would be great.

rlogiacco, 2015-09-17:
@Louis2016 would you be able to check the CDT make capability? There should probably be some documentation on the Eclipse website...

@bertenvdb could you please check the licenses and verify their compatibility, including the avr toolchain?

Louis2016, 2015-09-19:
@rlogiacco 
I've been able to succesfully build an Arduino project with the Eclipse CDT Managed Internal Builder:
- Eclipse CDT 8.7 Fresh installation
- C++ Project with Internal Builder and Cross GCC toolchain

- Toolchain defined with avr- prefix pointing to Arduino IDE 1.6.5-r5 toolchain
- Toolchain manual settings with appropriate flags and target atmega328p (Arduino nano)
- Linked folder to "core" and "variants/<model>"
- Added corresponding include in settings (note that avr include is auto discovered by the CDT Cross gcc provider)

The Eclipse CDT project builds both the "Arduino Core library" and the target app in one step:
- One Build configuration with Artifact "Static library" and excluding target project to build the core library
- One Build configuration with Artifact "Executable"  and excluding core to build the target app. This one has a dependency to the core library and is defined as active.
- A post build step has been defined on the target app build to generate the .hex file

Then triggered a "Build project" which automatically launch the core library build and link the whole stuff and generate the hex file.

So:
- The entire build process can be handled by Eclipse MBS Internal Build
- The core library and target app can be separated or handled in the same project

jantje, 2015-09-19:
Thanks for the investigation

If I understand correctly you have 2 projects. 1 for the Arduino and one for your sketch.
If so: This is the setup that existed in V1 of the plugin. We had to drop this (easy CDT oriented) way of working due to the fact that each sketch has defines that need to be used for building the core.
One of the tough things about the plugin was getting the core and the sketch in the same project.
If not: please explain the difference with the plugin and why it is interesting.

Using the internal builder sounds interesting. I assume we no longer need make.exe if we can use that. Do you know the difference between the internal and external builder from the point of the project? I never looked at it closely as Arudino used to come with a make tool but it is probably better to use the internal builder than to add make.

bertenvdb, 2015-09-19:
>@bertenvdb could you please check the licenses and verify their compatibility, including the avr toolchain?

@rlogiacco I'd love to, but I'm not even nearly competent enough to do this as most things you guys talk about sound like Chinese to me.
I'm struggling for more than a week now to get the plugin working. That's why I thought it would be great to have this:
>solve most of the troubles for newcomers with a much simpler installation process
remove some configuration steps and installation path related problems.

I'm a newcomer :-) When I'm a bit more familiar with all of this, I'd be happy to sign up as a test user.

Louis2016, 2015-09-19:
@jantje
In this test, the arduino core and the application are in the same eclipse project (core and include pointers are defined in a way similar to the one used by the plugin).
There are two eclipse "build configuration" as explained that compile the target app and its core dependency if needed.

I think the defines that are needed to build the core are not "sketch" related but "board" related (boards.txt).

The main difference are: 
- no specific toolchains. Cross Gcc is used instead.
- setting or adding a board is just a matter of adding a "build configuration" to the core and to the application with the corresponding flags and includes.

jantje, 2015-09-19:
@Louis2016 
I'm seriously wondering whether you are using the arduino eclipse plugin V2 (as it should be) because it sounds to me you are using something different.

For instance
>setting or adding a board is just a matter of adding a "build configuration" to the core and to the application with the corresponding flags and includes.

This is adding changing 2 configurations so it is more work than with the plugin.

Louis2016, 2015-09-19:
@jantje
If I understand it right, this thread is about plugin improvement...

jantje, 2015-09-19:
@Louis2016 
It is indeed about improvements. What I'm trying to say: if in the current plug-in changing/adding a board means changing/adding 1 configuration and in the improved version changing/adding a board means changing/adding 2 configuration; I'm failing to see the improvement.

Louis2016, 2015-09-19:
@jantje
Please see my edit.
The plugin doesn't use "Build configuration" that much. The test has been driven "by hand" from a clean CDT installation, as a proof of concept.
In a plugin implementation, the underlying details would be hidden from the user perspective. The user would just choose a board from a board manager that will configure the "build configurations".

As the things are simply exposed to CDT, the MBS and indexer are able to perform their job without any under the hood tweaking.
When things have been put in place by say the "board manager", the eclipse CDT components are all accessible by advanced users that can further tweak things (like changing compile flags, using other cores, ...).

jantje, 2015-09-20:
@Louis2016 
Now I get your problem :-)
I need some history to explain this.
When I started this plugin I had a huge experience in writing C++ code and very little in writing java code.
So when I bumped into some bugs in CDT code in non virtual methods in the ArduinoGnuMakefileGenerator.java class I fixed it with my limited java knowledge (copy the code and rename the classes). 

I reported the bugs to the cdt community on the mail list and on their bug list inclusive a description what is the problem and how it could be fixed. I hoped the cdt community would fix it. 
If I remember correctly 2 are bugs and 1 is an improvement (request)
https://bugs.eclipse.org/bugs/show_bug.cgi?id=422376
https://bugs.eclipse.org/bugs/show_bug.cgi?id=422377
https://bugs.eclipse.org/bugs/show_bug.cgi?id=422378
The 3 issues are all related to being able to compile both the archive and the sketch in the same project. 

Until now the CDT community didn't pick up my bugs and my efforts to try to compile the cdt stuff and become a contributor to cdt have failed. Given the fact that dough is trying to make a new build engine I don't think these changes will ever be picked up.
Recently I checked the files and they had changed since I reported the issues about 2 years ago but not content wise. So my fixes should still be valid.

To conclude: I pulled in part (or all of the MBS -I don't claim to know that-) to fix 3 minor issues
The whole code change is about adding 4 lines (of which one is a closing bracket) and there is probably a far better java-way to do this.

jantje, 2015-09-20:
@Louis2016 
In the previous post I told the story in a chronological way and I feel it is not getting fully to the point I want to make. So here the architectural take.

The MBS only exists in the arduino eclipse plugin due to bugs in the MBS; I reported the bugs to the CDT community. I would love CDT to apply the proposed fixes so we can get rid of the copy in the plugin.

For the building I am only using standard CDT integrations. The Arduino toolchain exists for following reasons:
1) to point to the gcc compiler based on the arduino platform.txt and boards.txt using configuration environment variables.
2) Introduce a "filtered" building sequence to allow to archive the core and libraries before linking the hex file. (using name providers)
3) Adding other Arduino build recipe steps. (like getting the file size) (in plugin.xml)
4) Stuff I may have forgotten.

I have been told that there are better implementations for the MBS problem. We will accept pull requests that are better implementations. Getting the bug fixes in CDT is my preferred solution.

Best regards
Jantje

rlogiacco, 2015-09-20:
@jantje I'm on the other side of the barricade here, with lot of experience in Java and just a little of C, which makes me struggle to understand the last 5 posts. I truly believe we can make this together!

@Louis2016 I believe I understood your point, would you be able to help us implement it? I truly believe @jantje faced the issues he reported, but looking at the problem from a different perspective might bring in a different solution. Keep your thoughts flowing!

rlogiacco, 2015-09-24:
Along this idea of using build configurations let me throw in something additional: Mars launchbar http://mcuoneclipse.com/2015/09/04/new-in-eclipse-mars-launchbar/ 

By properly creating build configs for the boards we might be able to fulfill a few features... @Louis2016 have you had the opportunity to investigate on the issues reported by @jantje ?

jantje, 2015-12-17:
FYI
I started this. As it is a huge change I put it on another branch.
The boardsmanager integration works already. The solution still needs lots of stitches for a smooth experience.

jantje, 2015-12-22:
For those who want a sneak preview. It starts comming together in the bigChange branch.
What is there:
The plugin installs the arduino avr platform at first startup.
ino and pde are registered automatically
You can add and remove platforms via preferences->arduino.
You can create a new sketch based on hardware libraries and private libraries.
You can add a library to a project
So basically only the library manager needs to be added before we can start testing :-)

jantje, 2015-12-30:
Here are download locations for the product.
http://eclipse.baeyens.it//download/big_change/linux32.2015-12-30_01-54-23.bigchange.tar.gz
http://eclipse.baeyens.it//download/big_change/linux64.2015-12-30_01-54-23.bigchange.tar.gz
http://eclipse.baeyens.it//download/big_change/mac64.2015-12-30_01-54-23.bigchange.tar.gz
http://eclipse.baeyens.it//download/big_change/win32.2015-12-30_01-54-23.bigchange.tar.gz
http://eclipse.baeyens.it//download/big_change/win64.2015-12-30_01-54-23.bigchange.tar.gz
The update site is located at: eclipse.baeyens.it/nightly_big_change

Feedback is welcomed

wimjongman, 2016-01-01:
I have downloaded the program and I noticed a job running. This job took a look time to complete without any visual explanation except that it was running. I assume that this job is fetching and installing the Arduino IDE.

After around 20 minutes the IDE asked me to allow a port to be opened on the local network. 

I suggest to make this job a user job and start it with a window. The user can then "Run in background". I also suggest to give some visual feedback on what the job is doing.

I then hooked in an Arduino UNO and uploaded a sketch. It all worked flawlessly. I have a new disk and I did not install the Arduino IDE previously.

I do have an issue with special hardware. I have a Helios board that I can not get the IDE to recognize if I point it to the hardware config of that board [1] (Dutch)

Also, since you guys have gone all this way, why not also install the make.exe automatically?

My board was recognized automatically without installing any drivers or setting up a com port. Very impressive compared to the hassle that I remembered from before. 

All in all. It looks great.

### Possible "Issues"
* Install job is too obscure and runs very long. An impatient/ignorant user will probably restart before the install has completed. I am not sure if the install is restartable.
* Possible additional hardware recognize problems
* Example sketches: When creating a new sketch only a limited set of examples is shown (e.g. "blink" is missing)
* Install/Fetch of make.exe
 

[1] http://www.heliosgadget.nl/

jantje, 2016-01-02:
Hi Wim
Thanks for testing and the feedback.
>I have downloaded the program and I noticed a job running. This job took a look time to complete without any visual explanation except that it was running. I assume that this job is fetching and installing the Arduino IDE.

There is a job that does so; so I guess that is the job you saw. At next run it should be quick.

>After around 20 minutes the IDE asked me to allow a port to be opened on the local network. 

This is probably related to the discovery services to find network enabled boards. This is started immediately after the boards installation job. Can you tell me which port it asked to open?
https://github.com/jantje/arduino-eclipse-plugin/blob/bigChange/it.baeyens.arduino.core/src/it/baeyens/arduino/ui/Activator.java#L86

>I suggest to make this job a user job and start it with a window. The user can then "Run in background". I also suggest to give some visual feedback on what the job is doing.

I would love that to. However I need someone with more eclipse experience to give me some hints on how to. Now I start a job in the activator. At that point in time the gui is not yet available.
https://github.com/jantje/arduino-eclipse-plugin/blob/bigChange/it.baeyens.arduino.core/src/it/baeyens/arduino/ui/Activator.java#L93

>I do have an issue with special hardware. I have a Helios board that I can not get the IDE to recognize if I point it to the hardware config of that board 1

I'll have a look. But don't hold your breath as it hits on 2 difficult problems.
1) How to handle old hardware setups that relied on the avr toolchain delivered with the arduino ide.
2) Do I support multiple private hardware locations or not.
Buy the way: If the boards.txt is not visible at all it is probably related to the java bug and the user folder. Have you checked the hardware paths in the libraries to point to the correct location?

>Also, since you guys have gone all this way, why not also install the make.exe automatically?

I have only been working on linux so far for this. I already downloaded a make.exe but the path still needs to be set correctly. In other words: coming soon.

>All in all. It looks great.

txs nice to hear.

>Possible "Issues"

Except for the examples; all of them are known and will be fixed. 
For the examples I still need to find a good maintainable solution.

Best regards
Jantje

wimjongman, 2016-01-04:
I'm not sure if you noticed but the installer fails with an invalid cast.

java.lang.ClassCastException: org.eclipse.core.runtime.Path cannot be cast to java.nio.file.Path
        at it.baeyens.arduino.managers.ArduinoManager.downloadAndInstall(ArduinoManager.java:163)
        at it.baeyens.arduino.managers.ArduinoManager.startup_Pluging(ArduinoManager.java:98)
        at it.baeyens.arduino.ui.Activator$2.run(Activator.java:102)
        at org.eclipse.core.internal.jobs.Worker.run(Worker.java:55)

java.nio.file.NoSuchFileException: C:\Users\jongw\git\arduino-eclipse-plugin\it.baeyens.arduino.product\target\products\it.baeyens.arduino.product\win32\win32\x86_64\eclipseArduino\arduinoPlugin\libraries\library_index.json
        at sun.nio.fs.WindowsException.translateToIOException(Unknown Source)
        at sun.nio.fs.WindowsException.rethrowAsIOException(Unknown Source)
        at sun.nio.fs.WindowsException.rethrowAsIOException(Unknown Source)
        at sun.nio.fs.WindowsFileSystemProvider.newByteChannel(Unknown Source)
        at java.nio.file.spi.FileSystemProvider.newOutputStream(Unknown Source)
        at java.nio.file.Files.newOutputStream(Unknown Source)
        at java.nio.file.Files.copy(Unknown Source)
        at it.baeyens.arduino.managers.ArduinoManager.loadLibraryIndex(ArduinoManager.java:233)
        at it.baeyens.arduino.managers.ArduinoManager.internalLoadIndices(ArduinoManager.java:73)
        at it.baeyens.arduino.managers.ArduinoManager.loadIndices(ArduinoManager.java:182)
        at it.baeyens.arduino.managers.ArduinoManager.startup_Pluging(ArduinoManager.java:83)
        at it.baeyens.arduino.ui.Activator$2.run(Activator.java:102)
        at org.eclipse.core.internal.jobs.Worker.run(Worker.java:55)

!ENTRY org.eclipse.core.jobs 4 2 2016-01-04 12:10:57.817
!MESSAGE An internal error occurred during: "Arduino installer job".
!STACK 0

wimjongman, 2016-01-04:
Also, what code formatter are you using? Can you switch to the default code formatter or provide formatting details on the project level?

To supply formatting on the project level, select a project and press ALT+ENTER. This will take you to a screen where you can override the preferences on the project level. You have to do this for all projects.

When devs work with different formatting rules you get all weird commits.

jantje, 2016-01-04:
I know about the wierd commits. :-( 
I have been fighting your pull request on de dialog box for the installer and I don't know what is on github and have a merge conflict in my workspace. I gzave up last time (to late) I'l give it a try again now
.
The formatting is available in the project. But I'll switch to the default JAVA.
I'm thinking of using UTF-8 as text encoding and unix newlines.
Is there an objections to this?

jantje, 2016-01-05:
@wimjongman 
Of your possible issues 1 is fixed by you and 4 is fixed by me.
For 2 (Possible additional hardware recognize problems): have you checked whether it works if you change the hardware locations in the preferences?
For 3 (Example sketches:) There are 2 types of example sketches missing. The ones of the default libraries and the real examples.
For the default libraries I'm wondering what would be best: Show the preferences and let the user decide which libraries and versions to install or install the latest version of a subset of libraries.

wimjongman, 2016-01-05:
I must admit that I am not sure how it is supposed to work for a different Arduino based board.

This is my setup:

[image]

And my drivers dir looks like this:

[image]

wimjongman, 2016-01-05:
btw did you fix that ClassCastException?

wimjongman, 2016-01-06:
Answer: No it is still there. Do you have some stale file that still needs to be committed?

!ENTRY org.eclipse.core.jobs 4 2 2016-01-06 11:14:40.073
!MESSAGE An internal error occurred during: "Arduino installer job".
!STACK 0
java.lang.ClassCastException: org.eclipse.core.runtime.Path cannot be cast to java.nio.file.Path
        at it.baeyens.arduino.managers.ArduinoManager.downloadAndInstall(ArduinoManager.java:163)
        at it.baeyens.arduino.managers.ArduinoManager.startup_Pluging(ArduinoManager.java:98)
        at it.baeyens.arduino.ui.Activator$2.run(Activator.java:102)
        at org.eclipse.core.internal.jobs.Worker.run(Worker.java:55)

jantje, 2016-01-06:
The folder to give is c:/Users/jongw/Arduino/hardware
After that select apply and then go arduino->boards and you should be able to add your board

For the classpath. 
Here is the problem
https://github.com/jantje/arduino-eclipse-plugin/blob/bigChange/it.baeyens.arduino.core/src/it/baeyens/arduino/managers/ArduinoManager.java#L163
It should state something like
Path makePath = Paths.get (ConfigurationPreferences.getPathExtensionPath().append("make.exe").toString());
I fixed that. So there may be more fixed unfixed issues. I'll look at it tonight.

jantje, 2016-01-08:
Fixed the above issue and added support for downloading examples.
When creating a new sketch you have the examples and the library examples.
Thanks to wim the startup procedure is looking nice to :-)

Next for me is testing the old hardware way for helios and Teensy.
I think if this is done and tested we are ready to move the big change to the nightly.

jantje, 2016-01-11:
@wimjongman 
The helios board is a setup for 1.0.4 :-(
If you want to use this board I advice to make a json file.That should be relatively easy and is reusable.

wimjongman, 2016-01-12:
Thanks, I will look into that.

closed 2016-04-01 by jantje