Sloeber/arduino-eclipse-plugin #307: Platform.txt Error

DQvsRA, 2015-09-19:
When i try to start new project i get this error message
[image]

jantje, 2015-09-19:
I've seen somethng similar before but I can not recall the root cause.
Which arduino IDE version and plugin version are you using?

DQvsRA, 2015-09-20:
I use 
Eclipse IDE for C/C++ Developers
Version: Mars Release (4.5.0)
Build id: 20150621-1200

Actually this problem does not effect on project.

jantje, 2015-09-20:
Which version of the plugin are you using? 
Which version of the Arduino IDE are you using?
Are you using Teensy?

jantje, 2015-10-17:
closing as to long open without input

closed 2015-10-17 by jantje