Sloeber/arduino-eclipse-plugin #309: I have problem running compile

DQvsRA, 2015-09-20:
Actually spend all my weekend trying to understand how to run program on Arduino from Eclipse and still have no result. Please help. 

I use Arduino Nano V3 on Ubuntu with Eclipse Mars (4.5)

This is what i got with Arduino for Eclipse.
[image]

And i also try AVR tool for C
Tutor from here http://www.instructables.com/id/How-to-get-started-with-Eclipse-and-AVR/
but without Settings->AVRDude enabled
First i have problem with "Load from MCU" it give an error
[image]

Next, if i try to ignore this step and start to compile my code i get that error 
[image]

In Arduino IDE all sketches loaded fine and all works as it should be, but ide is weak, and i want to use eclipse with my work.

I add custom ttyUSB0 as port.
I also try to reinstall official libuseb drives and also no results.

Can you help me with this, please.

jantje, 2015-09-20:
You are mixing up many things which make it impossible for me to help you with your current situation.
So I propose to start from scratch.
Install the nightly linux project from here http://eclipse.baeyens.it/nightly-linux.html
And then do configure, create sketch, and add library from as described here
http://eclipse.baeyens.it/how_to.shtml.
As soon as something goes wrong stop and ask the question.

DQvsRA, 2015-09-27:
Ok. I reinstall plugin from nighty builds (Eclipse-Help-Install New Software)
But still have nothing to work. The eclipse show me an error when i create and try to compile a project. 
Here is it:

[image]
[image]

jantje, 2015-09-27:
Which arduino IDE version are you using?

DQvsRA, 2015-09-28:
I use ARDUINO 1.6.5 64bit. And it compile and upload code very well, but it's weak of features for writing code.

jantje, 2015-09-28:
I see you installed avr plugin and the arduino eclipse plugin. As you are clearly a newbie to the arduino eclipse plugin *I strongly dis-advice to do so*.
I guess you configured avr but not the plugin.
I just tested here -on a linux 64 bit machine- with the arduino eclipse plugin nightly and arduino IDE 1.6.5 and I get the error before configuration and I do not get the error after configuration. As the error message states "Plugin is not properly configured. Please reconfigure plugin." it looks like a pretty good message.

closed 2015-09-28 by jantje