Sloeber/arduino-eclipse-plugin #310: Can't access: http://www.baeyens.it/eclipse/v2

mdilger, 2015-09-25:
Hi I am trying to set up Eclipse as per Neil Kolbains instructions on 2015-06-28

For some reason Eclipse can't find http://www.baeyens.it/eclipse/v2

Any ideas??

Also, I am happy to be a tester once I get verything running.

Cheers,
Mike

mdilger, 2015-09-25:
Sorted.  I had an internet issue.

closed 2015-09-25 by mdilger

jantje, 2015-09-25:
good :-)
For the testing. Please fil out this form. https://docs.google.com/forms/d/12laG01sOcWG23Zaixur4v31MFcZ66ObJL_8XvRVgZZA/viewform
Jantje

mdilger, 2015-09-25:
done, earlier. Cheers, Mike

jantje, 2015-09-25:
:-)

jantje, 2015-09-25:
Och yes
There are a couple of people subscribed with a wrong email address. If you didn't receive a google groups invite yet .... well you may be one of them.

mdilger, 2015-09-25:
Thinking about it,  I may have got distracted at "work" earlier today and not submitted the form.  Anyway, I have just completed the form again.  Cheers, Mike.