Sloeber/arduino-eclipse-plugin #312: Can't start project

DQvsRA, 2015-09-29:
Hello
OS: Windows 7
Eclipse: Mars 4.5
Arduino: 1.6.5
Hardware: Arduino Nano
Arduino-Eclipse: installed from market.

I can't even create a project. 

[image]
[image]

DQvsRA, 2015-09-29:
The same thing when i install from nightly.
P.S. My Arduino Folder is not in default path, i don't install any programs on C drive

jantje, 2015-09-29:
I have checked in change to investigate this issue.
Can you update your plugin tomorrow and try again?

jantje, 2015-10-11:
has the change helped you?

DQvsRA, 2015-10-15:
Actually i tune and prepare eclipse to work with "AVR for eclipse" with .h files from Arduino core and it's work fine.

closed 2015-10-15 by DQvsRA