Sloeber/arduino-eclipse-plugin #314: Used in Linux ok but something does not work under Windows

mr-miky, 2015-10-04:
I'm working on a project that uses Arduino DUE.
At work I use Winblows XP or 7 while at home I use Fedora 22.
At home everything works and the plugin generates Environment (C/C++ build) and Arduino compiler settings (C/C ++ Preprocessor general ...) has within it all that is needed.
Under Winblows, "Environment" it is correct but "Arduino compiler settings" is empty.
The project compiles successfully but while editing the code syntax highlighting Report mistake anywhere.
The interesting thing is that if I use AVR microcontrollers instead of SAM everything works properly.
I thought it was a problem of paths and moved everything relating sam (tools and hardware) together with AVR in C:\Program Files\Arduino\hardware\arduino and C:\Program Files\Arduino\hardware\tools but nothing changes.
As a last ditch effort I copied in workspace\.metadata\.plugins\org.eclipse.cdt.core the file Encoder_Due.language.settings.xml  that is under linux.
I fixed the various paths to reflect the configuration under Winblows and for the security I made it read only.
Miraculously the project is loaded, compiles and highlighting syntax/errors works BUT only until the first compilation since for some reason are regenerated the wrong "Arduino compiler settings".
If I close the project and reopen it reloads the "Arduino compiler settings" in the language.settings.xml "donated" by linux.
screenshots 1.2 and 3 are those of fedora and the rest of Winblows
[image]
[image]
[image]
[image]
[image]
[image]

mr-miky, 2015-10-04:
I forgot ...

linux:
   eclipse Mars 4.5
   arduino ide 1.6.5
   Arduino eclipse nightly buld

Winblows XP , 7 :
   eclipse Mars.1 Release (4.5.1)
   arduino ide 1.6.5-r5
   Arduino eclipse nightly buld

jantje, 2015-10-04:
Indeed there seems to be something wrong with the discovery due to the use of spaces which need quotes .....
Do you agree that compilation and upload work but the indexer reports lots of errors?

mr-miky, 2015-10-04:
I confirm, compilation works.
If errors indexing make that, for example, every uint8_t is underlined in red then the answer is yes.
I have not tried to upload on Arduino DUE, I focused on the above issue.
I did the upload only under linux and I forgot to tell you that I had problems related to the fact that while the Arduino IDE uses bossac 1.3a Arduino eclipse uses bossac 1.6. I worked around the problem by replacing the executable file in the folder 1.6-Arduino with the one contained in 1.3a-Arduino.

Using bossac 1.6 happens that once programmed Arduino DUE is first reset but soon after the flash is re deleted with the resulting  block of microcontroller...

closed 2015-10-04 by mr-miky

mr-miky, 2015-10-05:
It works !!!
I just updated the plugin and open the project under Winblows and everything works as it should including indexing and uploading to Arduino DUE.

I can confirms that under Winblows is used bossac 1.3a

Starting upload
using arduino loader
Starting reset using 1200bps touch process
Continuing to use COM12
Ending reset using 1200bps touch process

Launching C:\Documents and Settings\m.giorato\Dati applicazioni\Arduino15\packages\arduino\tools\bossac\1.3a-arduino/bossac.exe --port=COM12 -U false -e -w -v -b C:\Documents and Settings\m.giorato\workspace\Encoder_Due/Release/Encoder_Due.bin -R 
Output:
Erase flash
Write 65840 bytes to flash 
.....
.....
.....
Verify successful
Set boot flash true
CPU reset.
1.3a-arduino/bossac.exe finished
upload done

jantje, 2015-10-05:
Thanks for the feedback
>If errors indexing make that, for example, every uint8_t is underlined in red then the answer is yes.

So the answer is yes.
> I did the upload only under linux and I forgot to tell you that I had problems related to the fact that while the Arduino IDE uses bossac 1.3a Arduino eclipse uses bossac 1.6. I worked around the [...]

This is a bit alarming and confusing to me. The plugin should use the same upload program (in casu bossac) as the arduino IDE. As you can see from your own post just above I use the full path to the Arduino install.
If I look at my linux system I see the following
in eclipse for upload
>Launching /home/jan/.arduino15/packages/arduino/tools/bossac/1.6-arduino/bossac

from the command line
>bash-4.1$ /home/jan/.arduino15/packages/arduino/tools/bossac/1.6-arduino/bossac -h
Usage: bossac [OPTION...] [FILE]
Basic Open Source SAM-BA Application (BOSSA) Version 1.6-arduino
Flash programmer for Atmel SAM devices.
Copyright (c) 2011-2012 ShumaTech (http://www.shumatech.com)

So it looks as if the arduino ide (on my system) uses bossac 1.6-arduino.
This may be related to having another version installed in the boards manager but the plugin asks the arduino IDE the boards manager information. 
Can you take a look?

mr-miky, 2015-10-06:
I double-checked and under Winblows the arduino ide uses bossac 1.3a even if 1.6 present.

C:\Documents and Settings\m.giorato\Dati applicazioni\Arduino15\packages\arduino\tools\bossac\1.3a-arduino/bossac.exe -i -d --port=COM12 -U false -e -w -v -b .....

Dir of :  C:\Documents and Settings\m.giorato\Dati applicazioni\Arduino15\pkages\arduino\tools\bossac

8/09/2015  18.01    <DIR>          .
8/09/2015  18.01    <DIR>          ..
8/09/2015  18.01    <DIR>          1.3a-arduino
8/09/2015  17.58    <DIR>          1.6-arduino

as you can see there is the 1.6 but does not use it ...

When I go home tonight I'll check what Arduino IDE uses under Linux ...

mr-miky, 2015-10-06:
I checked at home and under fedora linux 22  arduino ide 1.6.5-r5 uses bossac1.3a .

Forzo il reset aprendo e chiudendo a 1200bps la porta /dev/ttyACM0
/home/michele/.arduino15/packages/arduino/tools/bossac/1.3a-arduino/bossac -i -d --port=ttyUSB0 -U false -e -w -v -b /tmp/build58556995896283239.tmp/sketch_oct06a.cpp.bin -R

mike-o-tronic, 2015-10-21:
Hi Jantje!

Sorry if I'm wrong here, but I'm having the same issue with my Arduino Micro here. "Arduino compiler settings" are empty and so I'm getting wrong warnings and errors. Compiling and flashing works fine. Screenshots would look the same as miky's. I have no Linux system to test though.

I'm using version win32.2015-10-05_02-06-49 in Windows 7 32.

I got around most errors by adding the avr, gcc and gcc-fixed libraries manually, but the indexer is still missing the board specific defines. I tried to add them manually to the Arduino compile options but no effect on that (i.e. -D__AVR_ATmega32U4__ to get registers for serial interface from the avr/io.h).

jantje, 2015-10-21:
@mike-o-tronic 
Mike what you are describing is another issue.
Please create a new issue and add images and the toolchain output to the new issue.
Please try another board to see whether it is board related or not.
Are we talking about a new project or a "migrated" project?