Sloeber/arduino-eclipse-plugin #315: Unable to compile blink.cpp for chipkit max32

mr-miky, 2015-10-06:
hi jantje,

I came across another oddity.
I can not compile for chipKIT Max32. The build fails because in the string of compilation there are two quotes out of place.

"/usr/share/arduino-1.6.5-r5/hardware/chipkit-core/pic32/compiler/pic32-tools/bin/pic32-g++"  -O2 -c -mno-smart-io -fno-exceptions -ffunction-sections -fdata-sections -g3 -mdebugger -Wcast-align -fno-short-double -fframe-base-loclist -mprocessor=32MX795F512L -DF_CPU=80000000L  -DARDUINO=10605 -D_BOARD_MEGA_  -DMPIDEVER=16777998 -DMPIDE=150 ""    -I"/usr/share/arduino-1.6.5-r5/hardware/chipkit-core/pic32/cores/pic32" -I"/usr/share/arduino-1.6.5-r5/hardware/chipkit-core/pic32/variants/Max32" -MMD -MP -MF"ck_test.cpp.d" -MT"ck_test.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "../ck_test.cpp"  -o  "ck_test.cpp.o"   -Wall

I Tried the above command in a shell and fails. I removed the two quotes after -DMPIDE=150 and the compilation and the compilation occur properly.

The file platform.txt of chipKIT is a bit different from those of Arduino IDE, use two colons to indicate the space, I do not know if this might be causing the problem.

pic32.library.core.path=./hardware/pic32/cores/pic32
pic32.library.path=./hardware/pic32/libraries
pic32.ldscript=chipKIT-BL-mega.ld
pic32.ldcommon=chipKIT-application-COMMON.ld
pic32.compiler.define=::-DMPIDEVER=16778004::-DMPIDE=150::
pic32.build.variant=Default_100
pic32.core.header=WProgram.h

mr-miky, 2015-10-06:
I forgot ...

linux:
eclipse Mars 4.5
arduino ide 1.6.5-r5
Arduino eclipse nightly buld

jantje, 2015-10-07:
the double double quote is indeed the problem. It is a bit weird as I remove them.
Can you point me to the platform.txt and boards.txt used?

mr-miky, 2015-10-07:
Boards.txt

############################################################
cerebot32mx4.name=Cerebot 32MX4
cerebot32mx4.group=Cerebot

# new items
cerebot32mx4.platform=pic32
cerebot32mx4.board=_BOARD_CEREBOT_32MX4_
cerebot32mx4.build.board=BOARD_CEREBOT_32MX4_

cerebot32mx4.board.define=
cerebot32mx4.ccflags=ffff
cerebot32mx4.ldscript=chipKIT-application-32MX460F512.ld
# end of new items

cerebot32mx4.upload.protocol=stk500v2
cerebot32mx4.upload.maximum_size=520192
cerebot32mx4.upload.speed=115200
cerebot32mx4.upload.tool=pic32prog

cerebot32mx4.bootloader.low_fuses=0xff
cerebot32mx4.bootloader.high_fuses=0xdd
cerebot32mx4.bootloader.extended_fuses=0x00
cerebot32mx4.bootloader.path=not-supported
cerebot32mx4.bootloader.file=not-supported
cerebot32mx4.bootloader.unlock_bits=0x3F
cerebot32mx4.bootloader.lock_bits=0x0F

cerebot32mx4.build.mcu=32MX460F512L
cerebot32mx4.build.f_cpu=80000000L
cerebot32mx4.build.core=pic32
cerebot32mx4.build.variant=Cerebot_32MX4

############################################################
############################################################
cerebot32mx7.name=Cerebot 32MX7
cerebot32mx7.group=Cerebot

# new items
cerebot32mx7.platform=pic32
cerebot32mx7.board=_BOARD_CEREBOT_32MX7_
cerebot32mx7.build.board=_BOARD_CEREBOT_32MX7_

cerebot32mx7.board.define=
cerebot32mx7.ccflags=ffff
cerebot32mx7.ldscript=chipKIT-application-32MX795F512.ld
# end of new items

cerebot32mx7.upload.protocol=stk500v2
cerebot32mx7.upload.maximum_size=520192
cerebot32mx7.upload.speed=115200
cerebot32mx7.upload.tool=pic32prog

cerebot32mx7.bootloader.low_fuses=0xff
cerebot32mx7.bootloader.high_fuses=0xdd
cerebot32mx7.bootloader.extended_fuses=0x00
cerebot32mx7.bootloader.path=not-supported
cerebot32mx7.bootloader.file=not-supported
cerebot32mx7.bootloader.unlock_bits=0x3F
cerebot32mx7.bootloader.lock_bits=0x0F

cerebot32mx7.build.mcu=32MX795F512L
cerebot32mx7.build.f_cpu=80000000L
cerebot32mx7.build.core=pic32
cerebot32mx7.build.variant=Cerebot_32MX7

############################################################
############################################################
cerebot_mx3ck.name=Cerebot MX3cK
[... 1331 lines cut ...]
pic32.compiler.ar.cmd=pic32-ar
pic32.compiler.ar.flags=rcs
pic32.compiler.objcopy.cmd=pic32-objcopy
pic32.compiler.objcopy.eep.flags=-O::ihex::-j::.eeprom::--set-section-flags=.eeprom=alloc,load::--no-change-warnings::--change-section-lma::.eeprom=0
pic32.compiler.elf2hex.flags=-a
pic32.compiler.elf2hex.cmd=pic32-bin2hex
#pic32.compiler.elf2hex.flags=-O::ihex::-R::.eeprom
#pic32.compiler.elf2hex.cmd=pic32-objcopy
pic32.compiler.c.elf.flags=-Os::-Wl,--gc-sections::-mdebugger::-mno-peripheral-libs::-nostartfiles
pic32.compiler.c.elf.cmd=pic32-g++
pic32.compiler.cpudef=-mprocessor=
#pic32.compiler.ldflags=
pic32.compiler.ldscript=chipKIT-BL.ld
#pic32.compiler.includepath=/opt/local/pic32/install-image/pic32mx/include
pic32.compiler.upload.cmd=
pic32.compiler.includepath={0}/hardware/pic32/compiler/pic32-tools/pic32mx/include
pic32.compiler.upload.flags=
#pic32.library.path=./hardware/pic32/cores/pic32
pic32.library.core.path=./hardware/pic32/cores/pic32
pic32.library.path=./hardware/pic32/libraries
pic32.ldscript=chipKIT-BL-mega.ld
pic32.ldcommon=chipKIT-application-COMMON.ld
pic32.compiler.define=::-DMPIDEVER=16778001::-DMPIDE=150::
pic32.build.variant=Default_100
pic32.core.header=WProgram.h

jantje, 2015-10-07:
I see what the problem is
```java
	String ret= compilerCommand.replaceAll("[^\\\\]\"\"", "").replaceAll("  ", " "); // remove
									     // "" except \""
									     // and
									     // double
									     // blanks
```
For the other problem I have replaced the regular expression "\"\"" by ("[^\\\\]\"\""
As far as I know this means replace all double quotes not preceded by \
the \"" now remain but apparently so do the ""
 the "" is caused by
recipe.cpp.o.pattern="{compiler.path}{compiler.cpp.cmd}" {compiler.cpp.flags} -mprocessor={build.mcu} -DF_CPU={build.f_cpu} -DARDUINO={runtime.ide.version} -D_{build.board} {compiler.define} "{compiler.cpp.extra_flags}" {build.extra_flags} {includes} "{source_file}" -o "{object_file}"

containing "{compiler.cpp.extra_flags}"  and 
compiler.cpp.extra_flags=
resulting in ""

input on the regex is appreciated.

jantje, 2015-10-08:
I tried a test here and the regex expression seems to be fine.
I fear you'll have to do a debugging session to find out why the "" are not removed.
By the way I think it is a bad idea to have double quotes  around {compiler.cpp.extra_flags} in recipe.cpp.o.pattern

mr-miky, 2015-10-09:
well, you could just do a check if the string resulting command contains "" and ask if remove them (this time only or always). In the configuration menu 'Arduino' add a tick "remove empty quotes".
You could even issue a warning to check platform.txt and then the user fix it.
I could not find an option to activate a log of actions of the plugin, I could load the sources of the plugin on eclipse but I never made debugging of a plugin.
You can download the package "chipKIT core"  here ... http://chipkit.net/wiki/index.php?title=ChipKIT_core

jantje, 2015-10-09:
A debugging session is the first thing to do. I can't code around a problem if I do not know what the problem is.
I currently do not have time to debug a board I do not own.
Debuging is easy just set a breakpoint at the regex that should remove the double quoted (by double clicking or right clicking) and start a debugging session by pressing the bug icon

jantje, 2015-11-26:
closing as usage error

closed 2015-11-26 by jantje