Sloeber/arduino-eclipse-plugin #316: Problem with ESP8266/Arduino and Eclipse

lrmoreno007, 2015-10-13:
Hi, i'm using Eclipse/Arduino to program a ESP8266/Arduino, everithing go OK except when upload with esptool.

Eclipse don't get the parameter: tools.esptool.upload.params.verbose = -vv   established in platform.txt

To make it run OK, I must add a new line in platform.txt with a generic " upload.verbose = -vv ".  But this can affect to other upload tools. 

Why eclipse don't read this parameter? how to solve?

Thanks in advance.

jantje, 2015-10-13:
I responded in the esp thread. I think there is a problem in the platform.txt file
https://github.com/esp8266/Arduino/pull/781

jantje, 2015-10-13:
:-)

jantje, 2015-10-17:
If you think there is nothing to fix in the plugin can you close this issue?

lrmoreno007, 2015-10-19:
Continue in esp8266/Arduino#781

closed 2015-10-19 by lrmoreno007