Sloeber/arduino-eclipse-plugin #317: Arduino Compiler Settings missing

mike-o-tronic, 2015-10-22:
"Arduino compiler settings" are empty even though "Enviroment variables" are set (see screenshots). As a result so I'm getting wrong warnings and errors from Eclipse, while the compiler does its job perfectly fine without complaints. Flashing works fine as well.

I went through all the boards listed and this issues occurs with the following ones:
Esplora, Leonardo, Robot Control, Robot Motor, Yun, Micro and LillyPad USB

So I'm sure this problem is related to the MCU since all of these boards are using the ATmega32U4. Unfortunately I don't have a Linux system available, so I can't test if it's OS specific. I'm currently using version win32.2015-10-05_02-06-49 in Windows 7 32.

I got around most errors by adding the avr, gcc and gcc-fixed libraries manually, but the indexer is still missing the board specific defines. I tried to add them manually to the Arduino compile options (i.e. -D__AVR_ATmega32U4__ to get registers for serial interface from the avr/io.h) but that didn't help.

[image]
[image]
[image]

jantje, 2015-10-22:
can you delete the DUMP files in the root of your workspace and the reselect the arduino boards and select apply in the project properties?
can you provide the console output of a build?
Have you installed a new version of arduino ide or avr toolchain?

mike-o-tronic, 2015-10-22:
DUMP files were not existing. I created a new workspace, before I checked all the boards. Console output is fine. No errors! But post a log later.
Using Arduino 1.6.5-R5

jantje, 2015-10-22:
there must be dump files.
A new workspace  a good way to get rid of them as well.

mike-o-tronic, 2015-10-23:
Yes, I meant there were no because I created a new workspace. Deleted them again and reapplied board settings, but the result is the same: No Arduino Compiler Settings, lots of problems shown in the editor.

But the toolchain works fine and gets all the flags:

mike-o-tronic, 2015-10-23:
08:07:47 **** Incremental Build of configuration Release for project ArduinoMicro ****
make all 
Building file: ../ArduinoMicro.cpp
Starting C++ compile
"C:\Programs\arduino-1.6.5-r5\hardware\tools\avr/bin/avr-g++" -c -g -Os -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10605 -DARDUINO_AVR_MICRO -DARDUINO_ARCH_AVR  -DUSB_VID=0x2341 -DUSB_PID=0x8037 '-DUSB_MANUFACTURER="Unknown"' '-DUSB_PRODUCT="Arduino Micro"'   -I"C:\Programs\arduino-1.6.5-r5\hardware\arduino\avr\cores\arduino" -I"C:\Programs\arduino-1.6.5-r5\hardware\arduino\avr\variants\micro" -MMD -MP -MF"ArduinoMicro.cpp.d" -MT"ArduinoMicro.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "../ArduinoMicro.cpp"  -o  "ArduinoMicro.cpp.o"   -Wall
Finished building: ../ArduinoMicro.cpp
 
Building file: C:/Programs/arduino-1.6.5-r5/hardware/arduino/avr/cores/arduino/CDC.cpp
Starting C++ compile
"C:\Programs\arduino-1.6.5-r5\hardware\tools\avr/bin/avr-g++" -c -g -Os -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10605 -DARDUINO_AVR_MICRO -DARDUINO_ARCH_AVR  -DUSB_VID=0x2341 -DUSB_PID=0x8037 '-DUSB_MANUFACTURER="Unknown"' '-DUSB_PRODUCT="Arduino Micro"'   -I"C:\Programs\arduino-1.6.5-r5\hardware\arduino\avr\cores\arduino" -I"C:\Programs\arduino-1.6.5-r5\hardware\arduino\avr\variants\micro" -MMD -MP -MF"arduino/CDC.cpp.d" -MT"arduino/CDC.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "C:/Programs/arduino-1.6.5-r5/hardware/arduino/avr/cores/arduino/CDC.cpp"  -o  "arduino/CDC.cpp.o"   -Wall
Finished building: C:/Programs/arduino-1.6.5-r5/hardware/arduino/avr/cores/arduino/CDC.cpp
 
Building file: C:/Programs/arduino-1.6.5-r5/hardware/arduino/avr/cores/arduino/HID.cpp
Starting C++ compile
"C:\Programs\arduino-1.6.5-r5\hardware\tools\avr/bin/avr-g++" -c -g -Os -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10605 -DARDUINO_AVR_MICRO -DARDUINO_ARCH_AVR  -DUSB_VID=0x2341 -DUSB_PID=0x8037 '-DUSB_MANUFACTURER="Unknown"' '-DUSB_PRODUCT="Arduino Micro"'   -I"C:\Programs\arduino-1.6.5-r5\hardware\arduino\avr\cores\arduino" -I"C:\Programs\arduino-1.6.5-r5\hardware\arduino\avr\variants\micro" -MMD -MP -MF"arduino/HID.cpp.d" -MT"arduino/HID.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "C:/Programs/arduino-1.6.5-r5/hardware/arduino/avr/cores/arduino/HID.cpp"  -o  "arduino/HID.cpp.o"   -Wall
Finished building: C:/Programs/arduino-1.6.5-r5/hardware/arduino/avr/cores/arduino/HID.cpp
 
Building file: C:/Programs/arduino-1.6.5-r5/hardware/arduino/avr/cores/arduino/HardwareSerial.cpp
Starting C++ compile
"C:\Programs\arduino-1.6.5-r5\hardware\tools\avr/bin/avr-g++" -c -g -Os -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10605 -DARDUINO_AVR_MICRO -DARDUINO_ARCH_AVR  -DUSB_VID=0x2341 -DUSB_PID=0x8037 '-DUSB_MANUFACTURER="Unknown"' '-DUSB_PRODUCT="Arduino Micro"'   -I"C:\Programs\arduino-1.6.5-r5\hardware\arduino\avr\cores\arduino" -I"C:\Programs\arduino-1.6.5-r5\hardware\arduino\avr\variants\micro" -MMD -MP -MF"arduino/HardwareSerial.cpp.d" -MT"arduino/HardwareSerial.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "C:/Programs/arduino-1.6.5-r5/hardware/arduino/avr/cores/arduino/HardwareSerial.cpp"  -o  "arduino/HardwareSerial.cpp.o"   -Wall
Finished building: C:/Programs/arduino-1.6.5-r5/hardware/arduino/avr/cores/arduino/HardwareSerial.cpp
 
Building file: C:/Programs/arduino-1.6.5-r5/hardware/arduino/avr/cores/arduino/HardwareSerial0.cpp
Starting C++ compile
"C:\Programs\arduino-1.6.5-r5\hardware\tools\avr/bin/avr-g++" -c -g -Os -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10605 -DARDUINO_AVR_MICRO -DARDUINO_ARCH_AVR  -DUSB_VID=0x2341 -DUSB_PID=0x8037 '-DUSB_MANUFACTURER="Unknown"' '-DUSB_PRODUCT="Arduino Micro"'   -I"C:\Programs\arduino-1.6.5-r5\hardware\arduino\avr\cores\arduino" -I"C:\Programs\arduino-1.6.5-r5\hardware\arduino\avr\variants\micro" -MMD -MP -MF"arduino/HardwareSerial0.cpp.d" -MT"arduino/HardwareSerial0.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "C:/Programs/arduino-1.6.5-r5/hardware/arduino/avr/cores/arduino/HardwareSerial0.cpp"  -o  "arduino/HardwareSerial0.cpp.o"   -Wall
Finished building: C:/Programs/arduino-1.6.5-r5/hardware/arduino/avr/cores/arduino/HardwareSerial0.cpp
 
Building file: C:/Programs/arduino-1.6.5-r5/hardware/arduino/avr/cores/arduino/HardwareSerial1.cpp
Starting C++ compile
"C:\Programs\arduino-1.6.5-r5\hardware\tools\avr/bin/avr-g++" -c -g -Os -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10605 -DARDUINO_AVR_MICRO -DARDUINO_ARCH_AVR  -DUSB_VID=0x2341 -DUSB_PID=0x8037 '-DUSB_MANUFACTURER="Unknown"' '-DUSB_PRODUCT="Arduino Micro"'   -I"C:\Programs\arduino-1.6.5-r5\hardware\arduino\avr\cores\arduino" -I"C:\Programs\arduino-1.6.5-r5\hardware\arduino\avr\variants\micro" -MMD -MP -MF"arduino/HardwareSerial1.cpp.d" -MT"arduino/HardwareSerial1.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "C:/Programs/arduino-1.6.5-r5/hardware/arduino/avr/cores/arduino/HardwareSerial1.cpp"  -o  "arduino/HardwareSerial1.cpp.o"   -Wall
Finished building: C:/Programs/arduino-1.6.5-r5/hardware/arduino/avr/cores/arduino/HardwareSerial1.cpp
 
Building file: C:/Programs/arduino-1.6.5-r5/hardware/arduino/avr/cores/arduino/HardwareSerial2.cpp
Starting C++ compile
[... cut ...]
.data               40   8388864
.text             4280         0
.bss               111   8388904
.comment            17         0
.debug_aranges    1008         0
.debug_info      31262         0
.debug_abbrev     6141         0
.debug_line       5547         0
.debug_frame      2940         0
.debug_str        8513         0
.debug_loc       16085         0
.debug_ranges     1104         0
Total            77048

Finished building target: ArduinoMicro
 

08:08:29 Build Finished (took 41s.864ms)

jantje, 2015-10-23:
I see nothing specific in the console output.
Let's hope it was a one off.

mike-o-tronic, 2015-10-23:
Just realized something else... when I delete the dump files the ArduinoUno project give me the attached error messages. I would expect to get the same error messages for ArduinoMicro project, but I don't. Maybe that's another hint of what's wrong.

Definitely I can reproduce the error for all boards having the ATmega32U4 even with new workspace and new Arduino Eclipse version. I had the same issue with version win32.2015-09-08_02-06-02 and changed to win32.2015-10-05_02-06-49 after reading about fix #314.

Description | Resource | Path | Location | Type
--------------- | ------------- | ------ | ------------ | ----
make: *** No rule to make target `clean'. | ArduinoUno |  |   | C/C++ Problem
Program "\bin\avr-g++" not found in PATH | ArduinoUno |  | Project Properties, C++ Preprocessor Include.../Providers, [Arduino Compiler Settings] options | C/C++ Scanner Discovery Problem
Program "\bin\avr-gcc" not found in PATH | ArduinoUno |  | Project Properties, C++ Preprocessor Include.../Providers, [Arduino Compiler Settings] options | C/C++ Scanner Discovery Problem

nitinsane, 2015-10-27:
I have the same problem with the latest (2015/10/05) update. On windows, I have setup Arduino IDE in c:\apps and my libraries are in C:\home\arduino\library folder. It's not picking up anything from libraries.

jantje, 2015-10-27:
@nitinsane 
can you post the result of 
dir C:\home\arduino\library

jantje, 2015-10-27:
@mike-o-tronic 
I think I found the problem. -DUSB_PRODUCT is specified for all the boards you claim have the problem and only for those. Moreover they all contain a space in the name.
So it may be that a similar fix as #314 is needed for these boards.

mike-o-tronic, 2015-10-27:
I removed the space from the name in the board.txt to see if that would solve the issue, but it doesn't.
Toolchain reports flags as:
-DUSB_VID=0x2341 -DUSB_PID=0x8037 '-DUSB_MANUFACTURER="Unknown"' '-DUSB_PRODUCT="ArduinoMicro"'

But the problem with missing comipler settings and indexer reporting errors is the same

jantje, 2015-10-27:
I'm prettty busy right now. I'll have a look on a windows machine later this week.

jantje, 2015-10-30:
I just tested it and it is space related. 
Because spaces are used you need quotes and then quote the quotes which is different from linux and windows and in windows even different from the interpretation engine ...
All very confusing
This should work
-DUSB_VID=0x2341 -DUSB_PID=0x8037 -DUSB_MANUFACTURER="Unknown" -DUSB_PRODUCT="ArduinoMicro"
I want to do some more testing and then I'll push a fix

jantje, 2015-10-30:
The nightly should contain the fix in about 4 hours.
can you test the fix?

mike-o-tronic, 2015-10-30:
New version works perfectly fine on a new workspace!

An old workspace doesn't get updated automatically and gave me some odd effects like a new project would take long to build and has Binaries and Archives right after creating the project.

To update a workspace I had to create a new workspace, setup the locations for Arduino IDE and libraries, (eventually exit Eclipse), copy post_processing_boards.txt and post_processing_platform.txt from the new workspace to old one.

closed 2015-10-30 by mike-o-tronic

jantje, 2015-10-30:
Txs
To update a workspace you need to reselect the arduino ide in the preferences-> select apply -> ok and then open the project properties arduino-> apply->ok

nitinsane, 2015-10-30:
after upgrading to the latest build, everything works now.
Thanks Jantje.

nitinsane, 2015-10-30:
yes, after latest build, it works now.