Sloeber/arduino-eclipse-plugin #318: Arduino IDE Path can't be found

heathhes, 2015-10-28:
I am using:
MAC OS El Capitan
Arduino 1.6.5
Eclipse IDE for C/C++ Developers
  Version: Mars.1 Release (4.5.1)
  Build id: 20150924-1200

The problem is that when I go to select where the Arduino IDE is located ( /Applications/Arduino-1.6.5)
in the Eclipse Preferences   (Eclipese -> Prefernces-> Arduino), the Arduino path IDE will not recognize the path.  I get the error "Arduino folder is not correct!".  Several other people have also had this problem on the Arduino Forum.

jantje, 2015-10-28:
Have you seen this? http://trippylighting.com/teensy-arduino-ect/arduino-eclipse-plugin/arduino-eclipse-ide-2-4-arduino-1-6-3-teensyduino-1-22-mac-windows-installation/

TrippyLighting, 2015-10-28:
BTW, two Mac users on the Teensy forum have recently confirmed that my instructions also still work with the newest releases of Teensyduino and the Arduno IDE.

heathhes, 2015-10-29:
I see the links for the Teensy but it does not relate to my issue. Here is the link (http://forum.arduino.cc/index.php?topic=345878.new#new) to the Arduino forum where others are having the same issue as well.

jantje, 2015-10-29:
I think you are using the latest stable and missed this 
"In the 1.5 series there has been a change in the folder structure on mac from arduino IDE 1.5.7 onwards. See this issue for more detail. "
in the mac part of this page http://baeyens.it/eclipse/installAdvice.shtml
see #180

heathhes, 2015-10-29:
Ok, there seems to be an issue with post Arduino 1.5.7 path. I am not super savy when it comes to this typed of issue.  Is there anyway you could make it more clear on how to correct the issue? I have reviewed the installAdvice #180, but I can't figure out what to do. If you could make it more obvious what to do to make it more clear to correct the issue I would be glad to share this information via a small tutorial and also contribute some $ to the Eclipse Arduino plugin (assuming it works).  Thank you very much for your time and effort.  I am really hoping I can sort this out.

TrippyLighting, 2015-10-30:
There is no issue with the post 1.5.7 path. If you would actually bother to really follow the steps provided in the installation tutorial and use the download links in the tiutorial I provide on my blog http://trippylighting.com/teensy-arduino-ect/arduino-eclipse-plugin/arduino-eclipse-ide-2-4-arduino-1-6-3-teensyduino-1-22-mac-windows-installation/
It will work.
Today I loaded down and installed  the most current Arduino 1.6.5 IDE and the most recent NIGHTLY build of the Arduino Eclipse IDE and it worked out of the box. This works with Teensyduino and without.

jantje, 2015-10-30:
@heathhes 
like @TrippyLighting is stating: use the nightly build of the arduino eclipse plugin.

jantje, 2015-11-26:
Closing assuming this is a usage error.

closed 2015-11-26 by jantje