Sloeber/arduino-eclipse-plugin #319: Arduino IDE Path can't be found

heathhes, 2015-10-28:
I am using:
MAC OS El Capitan
Arduino 1.6.5
Eclipse IDE for C/C++ Developers
  Version: Mars.1 Release (4.5.1)
  Build id: 20150924-1200

The problem is that when I go to select where the Arduino IDE is located ( /Applications/Arduino-1.6.5)
in the Eclipse Preferences   (Eclipese -> Prefernces-> Arduino), the Arduino path IDE will not recognize the path.  I get the error "Arduino folder is not correct!".  Several other people have also had this problem on the Arduino Forum.

closed 2015-10-28 by heathhes

jantje, 2015-10-28:
Nice to see you fixed it yourself
As other people have the same problem I think it is a good idea for you to share how you fixed it.
:-)

jantje, 2015-10-28:
duplicate of #318

heathhes, 2015-10-29:
I actually never figured it out.  I had just accidentally posted identical topmost 318 so I ‘resolved’ post 319 to remove it. I still can’t get the arduino plugin for eclipse to work on a mac. Thanks for your response.