Sloeber/arduino-eclipse-plugin #320: Plugin not working with Arduino IDE 1.6.5-r5 !

MrOnkelChiller, 2015-10-30:
Couldn't get it working.
Tell me if i'm wrong. Maybe its just my fault.

jantje, 2015-10-30:
For Mac you need the nightly version of the arduino eclipse plugin.
Are you using the nightly of the arduino eclipse plugin?

TrippyLighting, 2015-10-30:
If that is your fault is hard to evaluate given that you've posted NO detail. Not posting any detail of what process you actually followed is definitely your fault ;-)

These instructions have proven to work with the newest versions of everything. If you are not using Teensy boards, just come ego the installation of it.

http://trippylighting.com/teensy-arduino-ect/arduino-eclipse-plugin/arduino-eclipse-ide-2-4-arduino-1-6-3-teensyduino-1-22-mac-windows-installation/

Use the download links provided in these instructions. as Jantje has mentioned the Link for the Arduino Eclipse IDE links to the Nightly builds.

MrOnkelChiller, 2015-10-30:
@jantje i'm using the nightly product of the arduino eclipse plugin. 
The configuration and that stuff works great. Also the uploading to the mega seems to work correctly, but the program isnt running on the mega then. I tested it with the blink example. 

Il post in more detail when I am home from work.
Sorry for bad english, im from germany. :)

TrippyLighting, 2015-10-30:
Kein Problem. Wir können die Unterhaltung auch gerne in Deutsch fortsetzen ;-)

jantje, 2015-10-30:
@MrOnkelChiller 
@TrippyLighting has been a plugin user on the mac from day one. He is the best person in the world to help you out. It is due to him that the mac experience is as good as it is.

I fully agree with TrippyLighting that you're lacking to provide correct information (which completely goes against my understanding of 'deutsche Gründlichkeit'), Providing statements like "it doesn't work" without specifying any details is typically used by trolls  (read more about trolls here: https://en.wikipedia.org/wiki/Internet_troll.)

This makes that people who work and offer services for free (like TrippyLighting  and me) get seriously upset by issues as this one. We can only guess whether you are a troll or you don't have the respect for our work by not doing your homework and not providing "needed and correct information to help you out".

If you are not a troll and still want to continue using the plugin please provide following info:
A screen-shot of eclipse with the projects visible
The content of the console (text dump no screen-shot) from the build
The content of the console (text dump no screen shot) from the upload

jantje

MrOnkelChiller, 2015-10-30:
Well...
Now i want to explain myself.
I spend 4 hours yesterday trying to get stuff running. I got huffy because it still isn't wanting to run. (that Grammer)
I thought that it might be because the -r5 version of the IDE is very new, and it isnt compatible yet.
Todays morning at work i googled around a bit and was excited to see your site on Github here.
I posted my problem and thought that the answer would be something like 'the IDE is pretty new, 	compatibility will come in 3 days'.
But now that i know that this isnt a general problem, but it is just a problem for me, cards changed and I definitely would have posted with all requiered details if i knew that before. 

So when i get home from work i will search that together and post it here. 
I'm unhappy about sounding like a troll, that wasn't the aim. :D Because im definitely not a Internet troll. I'm the one who hates them. 

Also I really have respect for your work. I know how much time you put into such an Plugin until it works correct, esepcially when there are multiple versions of the IDE and so on. 
I'm doing an apprenticeship as Software engineer for the second year now. I know the feeling when you coded at a programm for a long time, you give it to the people you made it for and they dont respect the  power and endurance you put into writing the program. 

Anyway, I'm sorry for seeming like a troll (Hate them)
I will post all the detail in about 2 hours.

MrOnkelChiller, 2015-10-30:
The Screenshot:
[image]

Build Text Dumb:
```
16:08:32 **** Incremental Build of configuration Release for project Waermeschrank ****
make all 
Building file: ../.ino.cpp
Starting C++ compile
"/bin/avr-g++" -c -g -Os -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10604 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR     -I"/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino" -I"/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/variants/mega" -MMD -MP -MF".ino.cpp.d" -MT".ino.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "../.ino.cpp"  -o  ".ino.cpp.o"   -Wall
/bin/sh: /bin/avr-g++: No such file or directory
make: *** [.ino.cpp.o] Error 127

16:08:32 Build Finished (took 72ms)
```

Upload Text Dump:
```
Starting upload
using arduino loader
Starting reset using DTR toggle process
Flushing buffer
Toggling DTR
Continuing to use /dev/tty.usbmodem1411
Ending reset using DTR toggle process

Launching /bin/avrdude -C/etc/avrdude.conf -patmega2560 -cwiring -P/dev/tty.usbmodem1411 -b115200 -D -Uflash:w:/Users/LuisReckziegel/Documents/workspace/Waermeschrank/Release/Waermeschrank.hex:i 
Output:
upload failed
```

jantje, 2015-10-30:
From this info I can clearly see the build fails.
Can you provide the "DUMP_arduino_avr_mega_ArduinoIDE.tmp" file in the root of your workspace?

MrOnkelChiller, 2015-10-30:
[image]

I can't find such an file. :/

jantje, 2015-10-30:
How did you create the project?

MrOnkelChiller, 2015-10-30:
@jantje 
New -> Projekt -> New Arduino Sketch -> Entered Name -> Entered Borad specifications -> select code: sample sketch -> Finish

jantje, 2015-10-30:
Can it be you upgraded the plugin after you created the sketch?
Can you execute the "after upgrade actions" as described here http://baeyens.it/eclipse/how_to.shtml#/n?
Does that make a difference?

MrOnkelChiller, 2015-10-30:
I don't think i did that.. But now the file is there...

```
#This file has been generated automatically.
#Please do not change.
#PREFDUMP#
last.sketch.count=1
#Ignoring this line : update.check=true
#Ignoring this line : platform.auto_file_type_associations=true
editor.window.width.default=500
editor.window.height.default=600
editor.window.width.min=400
editor.window.height.min=290
#Ignoring this line : editor.code_folding=true
editor.font=Monaco,plain,12
#Ignoring this line : editor.antialias=true
#Ignoring this line : editor.external=false
#Ignoring this line : editor.caret.blink=true
#Ignoring this line : editor.invalid=false
#Ignoring this line : editor.linenumbers=true
#Ignoring this line : editor.keys.alternative_cut_copy_paste=false
#Ignoring this line : editor.keys.shift_backspace_is_delete=true
#Ignoring this line : editor.keys.home_and_end_travel_far=true
#Ignoring this line : console=true
console.output.file=stdout.txt
console.error.file=stderr.txt
console.lines=4
#Ignoring this line : console.auto_clear=true
console.length=500
#Ignoring this line : editor.tabs.expand=true
editor.tabs.size=2
#Ignoring this line : editor.indent=true
editor.divider.size=0
#Ignoring this line : editor.save_on_verify=true
run.options=
#Ignoring this line : run.options.memory=false
run.options.memory.initial=64
run.options.memory.maximum=256
run.display=1
#Ignoring this line : export.application.platform=true
#Ignoring this line : export.application.fullscreen=false
#Ignoring this line : export.application.stop=true
#Ignoring this line : export.applet.separate_jar_files=false
#Ignoring this line : export.delete_target_folder=true
#Ignoring this line : preproc.save_build_files=false
#Ignoring this line : preproc.color_datatype=true
#Ignoring this line : preproc.web_colors=true
#Ignoring this line : preproc.enhanced_casting=true
#Ignoring this line : preproc.substitute_floats=true
#Ignoring this line : preproc.substitute_unicode=true
#Ignoring this line : preproc.output_parse_tree=false
preproc.imports.list=java.applet.*,java.awt.Dimension,java.awt.Frame,java.awt.event.MouseEvent,java.awt.event.KeyEvent,java.awt.event.FocusEvent,java.awt.Image,java.io.*,java.net.*,java.text.*,java.util.*,java.util.zip.*,java.util.regex.*
run.present.bgcolor=#666666
run.present.stop.color=#cccccc
#Ignoring this line : run.present.exclusive=true
target_package=arduino
target_platform=avr
board=mega
software=ARDUINO
build.warn_data_percentage=75
programmer=arduino:avrispmkii
upload.using=bootloader
#Ignoring this line : upload.verify=true
serial.port=/dev/tty.usbmodem1411
serial.databits=8
serial.stopbits=1
serial.parity=N
serial.debug_rate=9600
editor.languages.current=de_DE
runtime.ide.path=/Applications/Arduino.app/Contents/Java
runtime.ide.version=10605
boardsmanager.additional.urls=
#Ignoring this line : build.verbose=false
builtin_platform_is_newer=10605
compiler.warning_level=default
custom_cpu=mega_atmega2560
#Ignoring this line : editor.update_extension=true
last.ide.1.6.2.daterun=1446218668
last.ide.1.6.2.hardwarepath=/Users/LuisReckziegel/Downloads/ArduinoComplete.app/Contents/Java/hardware
last.ide.1.6.4.daterun=1446148916
last.ide.1.6.4.hardwarepath=/Applications/Arduino.app/Contents/Java/hardware
last.ide.1.6.5.daterun=1446221448
last.ide.1.6.5.hardwarepath=/Applications/Arduino.app/Contents/Java/hardware
last.screen.height=1440
last.screen.width=2560
last.serial.location=1514,267,640,671
last.sketch0.location=1350,23,1215,1343,1070
last.sketch0.path=/Users/LuisReckziegel/Downloads/ArduinoComplete.app/Contents/Java/examples/01.Basics/Blink/Blink.ino
last.sketch1.location=145,300,988,725,474
last.sketch1.path=/Users/LuisReckziegel/Downloads/Arduino.app/Contents/Java/examples/01.Basics/Blink/Blink.ino
recent.sketches=/Applications/Arduino.app/Contents/Java/examples/01.Basics/Blink/Blink.ino,/Users/LuisReckziegel/Documents/Arduino/libraries/Adafruit_MCP9808/examples/mcp9808test/mcp9808test.ino,/Users/LuisReckziegel/Downloads/Arduino.app/Contents/Java/examples/01.Basics/AnalogReadSerial/AnalogReadSerial.ino,/Users/LuisReckziegel/Downloads/Arduino.app/Contents/Java/examples/01.Basics/Blink/Blink.ino
serial.line_ending=2
serial.port.file=tty.usbmodem1411
sketchbook.path=/Users/LuisReckziegel/Documents/Arduino
update.id=-6627725000697164080
#Ignoring this line : upload.verbose=false
runtime.os=macosx
runtime.tools.avrdude.path=/Applications/Arduino.app/Contents/Java/hardware/tools/avr
runtime.tools.avrdude-6.0.1-arduino5.path=/Applications/Arduino.app/Contents/Java/hardware/tools/avr
runtime.tools.avr-gcc.path=/Applications/Arduino.app/Contents/Java/hardware/tools/avr
runtime.tools.avr-gcc-4.8.1-arduino5.path=/Applications/Arduino.app/Contents/Java/hardware/tools/avr
runtime.platform.path=/Applications/Arduino.app/Contents/Java/hardware/arduino/avr
runtime.hardware.path=/Applications/Arduino.app/Contents/Java/hardware/arduino
```

jantje, 2015-10-30:
I guess the build works now as well.
And so should the upload

jantje, 2015-11-26:
Closing assuming this was a usage error

closed 2015-11-26 by jantje