Sloeber/arduino-eclipse-plugin #321: Several (non blocking) issues found on Arduino Eclipse IDE with Arduino 1.6.5-r5 and Teensyduino 1.26 os:Mac

universpharmacie, 2015-11-01:
Hi from France!
Got it up and running without problems first.
Things became strange when I went away from simple blink things and wanted to compile/upload a more complex software which compiled well in the Arduino IDE before.

a) Why isn't it sufficient (like in the original Arduino IDE) to simply write "#include <EEPROM.h>"? I got message of not found Symbols until I learned (after loosing hours to check paths and everything) that an additional (and in my eyes redundant) step was needed: Adding the same library again to the project through the menu.

b) Same for local .h files, simply copied over from the Arduino Sketch directory into the Eclipse project directory: simply #include "local includes.h" gave errors when I wanted to use things defined in that file. But when I used the project wizard to create a new header file and copied/pasted everything into it, all was fine. Strange, isn't it?

I thought that Arduino Eclipse IDE should be a drop in replacement for the original Arduino IDE and that it would be at least as smart as the latter concerning the #include handling. Perhaps it is, but for me, it looks more complicated and not logical.

What is really annoying for me, is that at each upload towards the Teensy 3.1 processor, a modal error message pops up, telling me that the serial monitor is not available. That is normal for the Teensy, since the code upload is not done via USBserial, but switching temporarily the USB to a different mode. The serial port comes automatically back, once the upload terminated. Since that is standard behavior for the Teensy 3.x and LC processors for years, I think that the Arduino Eclipse IDE should be aware of that and stop/restart the serial connection silently. 

Please accept my apologies if all that sounds very negative, but I had read some nice reviews about Arduino Eclipse and decided to go for it, hoping for a seamless transition and eased work. At both points, I'm somewhat disappointed and hope for helpful and encouraging hints.

If whatever is needed, logs, error messages and so on, don't hesitate to ask for, I will provide it.

Thank you in advance
Thierry

jantje, 2015-11-02:
I decided to no longer respond to github issues -and other questions- due to a lack of support. 
As a patron you can post a message on the patreon forum stating you support a github issue. This allows me to support people who support me.
https://www.patreon.com/posts/3700469

jantje, 2015-11-02:
Hi Thierry
A bunch of good questions/remarks you have.
Lets go through them one by one.
a) Why isn't it sufficient (like in the original Arduino IDE) to simply write "#include "?
The Arduino IDE uses a strict folder setup, this allows the arduino IDE to "search -and modify the build path- before build" for the needed libraries.
If you are a advanced C/C++ programmer it is unlikely that you want to follow the strict folder structure of Arduino. Technically C/C++ does not support something like "search -and modify the build path-before build"
Also the "search -and modify the build path- before build" had (and has?) many problems and side effects. 

Therefore I opted to have the developer list the libraries needed (which is the C/C++ way anyway -like in define in .h implement in .cpp-). Note that you can see all libraries you use in the libraries subfolder of your project -You know nothing else is used-.
Due to the multiple select in the menu->arduino->add arduino library functionality it should be fast and easy to add libraries.

b) Same for local .h files, simply copied over from the Arduino Sketch directory into the Eclipse project directory ...Strange, isn't it?
From your story I think you copied the file in the directory of the project under the workspace folder.
If so indeed eclipse will not pick it up and add it to the project. As such the indexer will not recognize the definitions in the header. The compilation should work fine though.
How to fix it. Select the project and press F5 or right click and select refresh. The refresh action is something to do each time there is a difference between disk and project.
But better is to prevent it :-)
If you drag and drop the file to the project in eclipse eclipse will ask you wether you want to link the file or copy the file. In this case the indexer and compiler wil both work fine.

>I thought that Arduino Eclipse IDE should be a drop in replacement for the original Arduino IDE and that it would be at least as smart as the latter concerning the #include handling. Perhaps it is, but for me, it looks more complicated and not logical.

I'm sorry but the plugin is not the same as the arduino IDE. If it were what would be the added value.
I Have a special section in the faq on the difference between the two. http://eclipse.baeyens.it/faq.shtml#/ide_vs_plugin
While rereading I noticed that I have not yet removed the fact the plugin did not support ino files.

And yes the plugin looks more complicated and follows the logic of source code development. The arduino IDE focusses on simplicity and arduino development.
Compare it to driving a 2WD to a 4WD. A 4WD is more complex (even if it is only a 4WD on off button) but it'll get you over rougher terrain.

>What is really annoying for me, is that at each upload towards the Teensy 3.1 processor, a modal error message pops up, telling me that the serial monitor is not available.

This sounds like a bug. There should not be a error message. Which os are you using?

>Since that is standard behavior for the Teensy 3.x and LC processors for years,

It has been like that because Paul and I programmed it that way. Arduino refused this for a very long time.

>Please accept my apologies if all that sounds very negative, but I had read some nice reviews about Arduino Eclipse and decided to go for it, hoping for a seamless transition and eased work. At both points, I'm somewhat disappointed and hope for helpful and encouraging hints.

I hope this feedback is helpfull. If you provide me with your OS and a image of the modal dialog box for Teensy I'll look into that one.

Best regards
Jantje

universpharmacie, 2015-11-02:
Hi Jantje, thank you already for these long explanations. I'm not yet an advanced C/C++ developer. In my professional life I started as a dBaseIV/Clipper developer (you see I'm an old fart...) and I am now doing php development for more than 10 years, but C/C++ is a recently acquired hobby.
One of the reasons, I wanted to switch over to Arduino Eclipse was in fact the ability to better organize my source code in different subfolders, but I started from my php experience where includes are resolved either from the default include path or from the relative path given in the include statement without further declaration. Ok, I will get used to that project management concept, I think.

Concerning the Serial Monitor "bug", I'll give you everything you ask for later when I'll be back home (still in the office for the moment). What I find still strange, is that I have a bunch of errors and warnings in the error console each time I start Arduino Eclipse. Perhaps these can give you already a hint?
I don't understand the framework errors at all, they are mostly about your plugin. The warnings about an invalid Arduino configuration is also strange since I made sure to set the complete and correct Arduino/Teensy parameters in the Preferences and in each Project properties.

I just wanted to attach the error log file, but apparently I have not enough write permissions to this repository, so I will do a copy and paste in the next message.

universpharmacie, 2015-11-02:
Error.log file (I'm running OS X 10.11.2 El Capitan build 15c27e)
!SESSION 2015-11-02 18:07:18.709 -----------------------------------------------
eclipse.buildId=unknown
java.version=1.8.0_60
java.vendor=Oracle Corporation
BootLoader constants: OS=macosx, ARCH=x86_64, WS=cocoa, NL=fr_FR
Framework arguments:  -perspective it.baeyens.arduino.application.perspective -keyring /Users/thierryfrenkel/.eclipse_keyring -showlocation
Command-line arguments:  -os macosx -ws cocoa -arch x86_64 -perspective it.baeyens.arduino.application.perspective -keyring /Users/thierryfrenkel/.eclipse_keyring -showlocation

!ENTRY it.baeyens.arduino.common 4 0 2015-11-02 18:07:19.725
!MESSAGE FrameworkEvent ERROR
!STACK 0
org.osgi.framework.BundleException: Could not resolve module: it.baeyens.arduino.common [399]
  Another singleton bundle selected: osgi.identity; osgi.identity="it.baeyens.arduino.common"; type="osgi.bundle"; version:Version="2.4.0.201511010208"; singleton:="true"

	at org.eclipse.osgi.container.Module.start(Module.java:434)
	at org.eclipse.osgi.container.ModuleContainer$ContainerStartLevel.incStartLevel(ModuleContainer.java:1582)
	at org.eclipse.osgi.container.ModuleContainer$ContainerStartLevel.incStartLevel(ModuleContainer.java:1561)
	at org.eclipse.osgi.container.ModuleContainer$ContainerStartLevel.doContainerStartLevel(ModuleContainer.java:1533)
	at org.eclipse.osgi.container.ModuleContainer$ContainerStartLevel.dispatchEvent(ModuleContainer.java:1476)
	at org.eclipse.osgi.container.ModuleContainer$ContainerStartLevel.dispatchEvent(ModuleContainer.java:1)
	at org.eclipse.osgi.framework.eventmgr.EventManager.dispatchEvent(EventManager.java:230)
	at org.eclipse.osgi.framework.eventmgr.EventManager$EventThread.run(EventManager.java:340)

!ENTRY it.baeyens.arduino.core 4 0 2015-11-02 18:07:19.730
!MESSAGE FrameworkEvent ERROR
!STACK 0
org.osgi.framework.BundleException: Could not resolve module: it.baeyens.arduino.core [401]
  Another singleton bundle selected: osgi.identity; osgi.identity="it.baeyens.arduino.core"; type="osgi.bundle"; version:Version="2.4.0.201511010208"; singleton:="true"

	at org.eclipse.osgi.container.Module.start(Module.java:434)
	at org.eclipse.osgi.container.ModuleContainer$ContainerStartLevel.incStartLevel(ModuleContainer.java:1582)
	at org.eclipse.osgi.container.ModuleContainer$ContainerStartLevel.incStartLevel(ModuleContainer.java:1561)
	at org.eclipse.osgi.container.ModuleContainer$ContainerStartLevel.doContainerStartLevel(ModuleContainer.java:1533)
	at org.eclipse.osgi.container.ModuleContainer$ContainerStartLevel.dispatchEvent(ModuleContainer.java:1476)
	at org.eclipse.osgi.container.ModuleContainer$ContainerStartLevel.dispatchEvent(ModuleContainer.java:1)
	at org.eclipse.osgi.framework.eventmgr.EventManager.dispatchEvent(EventManager.java:230)
	at org.eclipse.osgi.framework.eventmgr.EventManager$EventThread.run(EventManager.java:340)

!ENTRY it.baeyens.arduino.monitor 4 0 2015-11-02 18:07:19.733
!MESSAGE FrameworkEvent ERROR
!STACK 0
org.osgi.framework.BundleException: Could not resolve module: it.baeyens.arduino.monitor [402]
  Another singleton bundle selected: osgi.identity; osgi.identity="it.baeyens.arduino.monitor"; type="osgi.bundle"; version:Version="2.4.0.201511010208"; singleton:="true"

	at org.eclipse.osgi.container.Module.start(Module.java:434)
	at org.eclipse.osgi.container.ModuleContainer$ContainerStartLevel.incStartLevel(ModuleContainer.java:1582)
	at org.eclipse.osgi.container.ModuleContainer$ContainerStartLevel.incStartLevel(ModuleContainer.java:1561)
	at org.eclipse.osgi.container.ModuleContainer$ContainerStartLevel.doContainerStartLevel(ModuleContainer.java:1533)
	at org.eclipse.osgi.container.ModuleContainer$ContainerStartLevel.dispatchEvent(ModuleContainer.java:1476)
	at org.eclipse.osgi.container.ModuleContainer$ContainerStartLevel.dispatchEvent(ModuleContainer.java:1)
	at org.eclipse.osgi.framework.eventmgr.EventManager.dispatchEvent(EventManager.java:230)
	at org.eclipse.osgi.framework.eventmgr.EventManager$EventThread.run(EventManager.java:340)

!ENTRY it.baeyens.cdt.refactor 4 0 2015-11-02 18:07:19.735
!MESSAGE FrameworkEvent ERROR
!STACK 0
org.osgi.framework.BundleException: Could not resolve module: it.baeyens.cdt.refactor [403]
  Another singleton bundle selected: osgi.identity; osgi.identity="it.baeyens.cdt.refactor"; type="osgi.bundle"; version:Version="1.0.0.201511010208"; singleton:="true"

	at org.eclipse.osgi.container.Module.start(Module.java:434)
	at org.eclipse.osgi.container.ModuleContainer$ContainerStartLevel.incStartLevel(ModuleContainer.java:1582)
	at org.eclipse.osgi.container.ModuleContainer$ContainerStartLevel.incStartLevel(ModuleContainer.java:1561)
	at org.eclipse.osgi.container.ModuleContainer$ContainerStartLevel.doContainerStartLevel(ModuleContainer.java:1533)
	at org.eclipse.osgi.container.ModuleContainer$ContainerStartLevel.dispatchEvent(ModuleContainer.java:1476)
	at org.eclipse.osgi.container.ModuleContainer$ContainerStartLevel.dispatchEvent(ModuleContainer.java:1)
	at org.eclipse.osgi.framework.eventmgr.EventManager.dispatchEvent(EventManager.java:230)
	at org.eclipse.osgi.framework.eventmgr.EventManager$EventThread.run(EventManager.java:340)

!ENTRY it.baeyens.arduino.common 4 0 2015-11-02 18:07:19.739
[... 80 lines cut ...]
!MESSAGE Operation details
!SUBENTRY 1 org.eclipse.equinox.p2.operations 4 10011 2015-11-02 18:09:14.902
[... cut ...]
!ENTRY it.baeyens.arduino.core 2 0 2015-11-02 18:21:54.225
!MESSAGE The project dVoxUniCore Has an invalid arduino board configuration.

!ENTRY it.baeyens.arduino.core 2 0 2015-11-02 18:21:54.261
!MESSAGE The project dVoxUniCore Has an invalid arduino board configuration.

!ENTRY it.baeyens.arduino.core 2 0 2015-11-02 18:23:05.546
!MESSAGE The project dVoxUniCore Has an invalid arduino board configuration.

!ENTRY it.baeyens.arduino.core 2 0 2015-11-02 18:23:05.564
!MESSAGE The project dVoxUniCore Has an invalid arduino board configuration.

universpharmacie, 2015-11-02:
Here is the the so-called "invalid" arduino board configuration of the project:
[image]

jantje, 2015-11-02:
You really have an eye for detail. Great :-)).
I have seen these framework errors before. A clean on the build server and a update fixed that last time.
So I did a clean on my build server and relaunched the build.
Lets hope this will fix it again. I have set my build server now to do a clean before build.

The "Has an invalid arduino board configuration" is a warning and is related to the new boards manager. I added the warning after a bug fix, I have been in doubt whether I would keep the warning or not.
If you switch arduino IDE version in the preferences you will get this warning for nearly each and every project. However if you remove a board in the boards manager and you have a project that uses this board this warning will appear and it is actually an error.
I'll make it a info message instead of a warning message and see how it goes. What do you think?

universpharmacie, 2015-11-02:
I can't decide the latter for you. What I'd try to do if I were in your position is most probably to add a kind of integrity check which verifies the coherence of the selected board for a project with the boards.txt selected in the preferences overtime the project is opened and either display nothing if everything is coherent or display an error if the selected board can not be found in the boards.txt, saying exactly that. But I don't know if it's possible. An additional idea would be to parse all present boards.txt together and make the board selector's list a UNION (in the sql way) of all. But that would still require the coherence check above.

By the way... I'm using Arduino 1.6.5-r5 (the most recent AFAIK) and I keep getting the following message when clicking on "Apply" or "OK"...
[image]

And here is what happens when I upload to the Teensy 3.1:
[image]
while there are no other warnings in the upload console:
Starting upload
using arduino loader

Launching /Applications/Arduino.app/Contents/Java/hardware/tools/teensy_post_compile -file=dVoxUniCore -path=/Users/thierryfrenkel/Documents/ArduinoEclipse/dVoxUniCore/Release -tools=/Applications/Arduino.app/Contents/Java/hardware/tools -board=TEENSY31 -reboot 
Output:
teensy_post_compile finished
upload done

jantje, 2015-11-02:
Thanks for the input on the warning issue. Basically what you describe is what is going on. When the board is not found in the boards file it gives the warning. This code is there to try to make things work after configuration changes.

From what I see you are on a mac.
If I remember correctly. As the upload does not need a usb port you simply do not specify the usb port in the project properties-> arduino (having it blank is fine)
I can't test as I have no teensy configured right now.

TrippyLighting, 2015-11-02:
Well, if you need a Teensy, that can be arranged. Just let me know what varinant(s)

jantje, 2015-11-02:
@TrippyLighting
Txs :-). I still have teensy's in the house but I havn't got the teensy installed on the system I'm currently on.
As you are both a mac expert and a Teensy expert: Do you recognize this issue?

universpharmacie, 2015-11-02:
I just discovered that I need to set the serial port in the project properties. Apparently, the teensy loader needs it. If I let the field empty, the code will not be uploaded and the Teensy just continue running although the upload console tells that the code upload had worked. Mystère...

universpharmacie, 2015-11-02:
BTW: The Serial port problem popup comes only if Serial is used in the code. No Serial, no problem, even if that's not helpful. Is there a way to send status and debug messages to another Stream as you can do in Java? Kind of System.out.println() which is not blocking and you haven't to wait for a port to be ready...

jantje, 2015-11-02:
I need to look into the serial issue.
>Kind of System.out.println() which is not blocking and you haven't to wait for a port to be ready...

As you are on a stand alone board you can only write to connected devices like serial connected via usb or a lcd screen.
But this is a question for the teensy/arduino forum not the plugin.

jantje, 2015-11-04:
@universpharmacie 
I saw I didn't do a stack dump in this case. So I added the stack-trace to the logging. Can you try tomorrow with the new version and provide the stack-trace?
I hope to have some time to look at it tomorrow.
Am I right you are on a mac?

universpharmacie, 2015-11-04:
Thank you, I will try it tomorrow. Yes I'm on a 13" MacBook Pro (Early 2011) running OS X 10.11.2 El Capitan.

jantje, 2015-11-04:
Just so you know. It should fail but with the stack-trace I should be able to find out what he is trying to do.
I think the plugin is trying to reopen the serial connection to soon. I hope the stack-trace confirms that.

jantje, 2015-11-04:
I just tried here (linux 64 bit; arduini IDE 1.6.5 TeensyDuino 1.25 teensy 2.0++) 
All seems to work fine
The serial monitor closes and reopens when I upload.
No dialog boxes with error.
I hope the trace will show something

universpharmacie, 2015-11-04:
Just updated. Made a build and upload test. Error continues showing up. No stack trace available. The different Framework errors persist, too...

jantje, 2015-11-04:
Is it already tomorrow for you? The build is not yet done. It will take at least 5 more hours before it is ready.

universpharmacie, 2015-11-04:
You might be right about this being a timing issue because the serial error pops up very quickly. You should know that the Teensy 3.1 has a much quicker processor core (32bit ARM Cortex M4 @72MHz) than the Teensy 2.0++(8bit AVR @16MHz) 

GREAT DISCOVERY: While adding delay(whatever) after Serial.begin(57600) before checking if(Serial.available) did not change the problem, adding delay(10) before Serial.begin(57600) makes the problem disappear. With delay (5) it re-appears. It seems that the Teensy is booting too quickly and that a too early Serial.begin() is causing the problem. Hope that helps! :-)

universpharmacie, 2015-11-04:
For me, it is 0h30 now, which is "tomorrow". I made an extra effort to stay up... :-)

universpharmacie, 2015-11-04:
Another little, but annoying thing. When one of the console windows (upload,serial,error) has the Mouse Focus and you have a project open in the main window and then you click on the green "Upload Sketch" button in the Menu bar, another error pops up: 
[image]

jantje, 2015-11-05:
I noticed the "project not found" to, and I find it annoying to. However eclipse does not have the concept "current active project" or "currently selected project". The current code is the best i could come up with right now.

I think it is strange that a delay(10); in the sketch solves the problem. I'll try here with a teensy 3.1

jantje, 2015-11-05:
I tried with a 3.0 (don't have a 3.1 lying around where I am right now) with 96MH overclock (to make it as fast as possible) and I got a dialog box once (not the same error as yours ), all seemed to work fine though. The dialog popped-up when trying to reopen the serial monitor.

If I close the teensy uploader before the upload, It looks like the serial monitor is ok after the upload; but it is not reporting new information. When I close the uploader; the serial monitor starts working again.
The fact the serial monitor only starts working after the teensy uploader is stopped is normal because the plugin waits for the uploader to "stop working" which is "closes down".
I don't see a fix for this issue. I'll discuss with Paul.

I also see lots of "The project teensy Has an invalid arduino board configuration." which seems to be Teensy related. I'll take a look at it.

and the "!MESSAGE FrameworkEvent ERROR" has not gone away or is back.
Again something for me to look at.

Were you able to get a stack trace already?

jantje, 2015-11-05:
I posted on the teensy forum for the uploader issue with the serial monitor. My explanation there should make things clearer for you on what is going on. It contains 2 possible work arounds

jantje, 2015-11-05:
The "The project teensy Has an invalid arduino board configuration."  message is logged in some situations it should not be logged. Will check in a fix.

universpharmacie, 2015-11-05:
I saw and commented you post on the pjrc forums. :-)
I'm just updating (again) my Arduino Eclipse but I'll only be able to check it out tonight when I'll be at home. I can't fiddle around with Teensys at my day job...

wimjongman, 2015-11-05:
One strategy is to make a selection listener that keeps track of the last selected project.

universpharmacie, 2015-11-05:
Here is the promised stack trace:
jssc.SerialPortException: Port name - /dev/tty.usbmodem253381; Method name - openPort(); Exception type - Port not found.
	at jssc.SerialPort.openPort(SerialPort.java:167)
	at it.baeyens.arduino.arduino.Serial.connect(Serial.java:167)
	at it.baeyens.arduino.monitor.views.SerialMonitor.ResumePort(SerialMonitor.java:521)
	at it.baeyens.arduino.common.Common.StartSerialMonitor(Common.java:448)
	at it.baeyens.arduino.tools.uploaders.UploadSketchWrapper$UploadJobWrapper.run(UploadSketchWrapper.java:175)
	at org.eclipse.core.internal.jobs.Worker.run(Worker.java:54)

jantje, 2015-11-05:
That is indeed at reopening the serial monitor.
It is wierd that the delay in your sketch fixes the problem. This because it means the serial port is available later.

universpharmacie, 2015-11-06:
I admit that it's weird, but what can I do, out of testing and reporting feedback? 
Basically, I can live with that error message, it's not blocking. But in the last 30 years of software developing, I've learned that if you don't care enough about non-blocking issues and warnings which seem unimportant for the moment, things risk to get worse over time...

universpharmacie, 2015-11-06:
Just installed the latest nightly build. Framework Errors persist in the refactor, common, monitor and core modules of the plugin.
The stack trace looks always similar:
org.osgi.framework.BundleException: Could not resolve module: it.baeyens.cdt.refactor [452]
  Another singleton bundle selected: osgi.identity; osgi.identity="it.baeyens.cdt.refactor"; type="osgi.bundle"; version:Version="1.0.0.201511060208"; singleton:="true"

	at org.eclipse.osgi.container.Module.start(Module.java:434)
	at org.eclipse.osgi.container.ModuleContainer$ContainerStartLevel.incStartLevel(ModuleContainer.java:1582)
	at org.eclipse.osgi.container.ModuleContainer$ContainerStartLevel.incStartLevel(ModuleContainer.java:1561)
	at org.eclipse.osgi.container.ModuleContainer$ContainerStartLevel.doContainerStartLevel(ModuleContainer.java:1533)
	at org.eclipse.osgi.container.ModuleContainer$ContainerStartLevel.dispatchEvent(ModuleContainer.java:1476)
	at org.eclipse.osgi.container.ModuleContainer$ContainerStartLevel.dispatchEvent(ModuleContainer.java:1)
	at org.eclipse.osgi.framework.eventmgr.EventManager.dispatchEvent(EventManager.java:230)
	at org.eclipse.osgi.framework.eventmgr.EventManager$EventThread.run(EventManager.java:340)

universpharmacie, 2015-11-06:
But nice to see that the Arduino configuration thing is fixed. I noticed the extended interface in the Preferences-Arduino screen. :-)

jantje, 2015-11-06:
>I admit that it's weird, but what can I do, out of testing and reporting feedback?
Basically, I can live with that error message, it's not blocking. But in the last 30 years of software developing, I've learned that if you don't care enough about non-blocking issues and warnings which seem unimportant for the moment, things risk to get worse over time...

I fully agree.

jantje, 2015-11-06:
I know it probably sounds borring could you test (again? -wait 4 hours for the build though). I fixed #323 and this is related to your issue at 2 levels.
1 there is/should be a hang
2 I tripled the time to find the serial port

jantje, 2015-11-06:
I made a new issue for the project not found issue based on @wimjongman his input.
I will look at it but I'm not planning to fix this in the next release as testing has already started.
If I got it right then only the "Framework Errors" remain open here.

universpharmacie, 2015-11-07:
Updated and tested. The timing issue seems to be basically solved, thank you. I still get the com port error message in about 1 of 20 compile/upload runs, but that's only when the teensy loader gets stuck and shows the invitation to press the program button of the teensy to force the boot loader mode. The same happens occasionally in the Arduino IDE context, so that seems not longer to be an Eclipse problem.
Thus, only the Framework errors persist, yes.

jantje, 2015-11-07:
To fix the "bootloader got stuck" I need the change request I asked for here 
https://forum.pjrc.com/threads/31325-Teensy-upload-and-the-arduino-eclipse-plugin

jantje, 2015-11-12:
I have discussed in length with @wimjongman on the "Framework Errors" and the conclusion is as follows:
The error appears after the plugin has been updated. The origin of the error is OSGI which reports that the "version of the plugin that is installed that becomes active"
This is a warning/info not an error.
OSGI probably does so because old versions of the plugin remain in the eclipse installation even though they should not. Even the -clean option does not remove them.
My conclusion is that this is a minor issue that may be related to the plugin but may also be related to OSGI.
If we ever find a way to avoid the creation of the error in the error log we will investigate the effort to avoid the error.

jantje, 2016-02-13:
I'm getting plenty of these error again in V3 (after updating). 
I'm willing to throw in a Arduino mega ADK for whomever can fix this :-)

wimjongman, 2016-02-13:
Interesting ;)

jantje, 2016-02-17:
I'm serious: A arduino mega ADK for whomever fixes this issue.

jantje, 2016-02-20:
When Upgrading my linux system eclipse refuses with this error.
This looks related to the untrained eye

```
Cannot complete the install because of a conflicting dependency.
  Software being installed: Arduino Eclipse Extensions 3.0.0.201602191959 (it.baeyens.arduino.feature.feature.group 3.0.0.201602191959)
  Software currently installed: Eclipse e4 Rich Client Platform 1.3.101.v20150204-1353 (org.eclipse.e4.rcp.feature.group 1.3.101.v20150204-1353)
  Only one of the following can be installed at once: 
    Core Runtime 3.10.0.v20140318-2214 (org.eclipse.core.runtime 3.10.0.v20140318-2214)
    Core Runtime 3.11.1.v20150903-1804 (org.eclipse.core.runtime 3.11.1.v20150903-1804)
  Cannot satisfy dependency:
    From: Arduino eclipse plugin 3.0.0.201602191959 (it.baeyens.arduino.core 3.0.0.201602191959)
    To: bundle org.eclipse.core.runtime 3.11.1
  Cannot satisfy dependency:
    From: Arduino Eclipse Extensions 3.0.0.201602191959 (it.baeyens.arduino.feature.feature.group 3.0.0.201602191959)
    To: it.baeyens.arduino.core [3.0.0.201602191959]
  Cannot satisfy dependency:
    From: Eclipse e4 Rich Client Platform 1.3.101.v20150204-1353 (org.eclipse.e4.rcp.feature.group 1.3.101.v20150204-1353)
    To: org.eclipse.core.runtime [3.10.0.v20140318-2214]
```

universpharmacie, 2016-02-20:
I had the same, after updating once with the nightly 3.0 plugin, it never would update again because of dependency problems. What I did was simply trashing everything and re-installing the newest complete nightly Arduino-Eclipse dmg. Since then, all problems have gone.

wimjongman, 2016-02-21:
So you install the product and then you update it using the update site correct?

universpharmacie, 2016-02-21:
I try to update it from the "Help"->"Check for updates" menu. But since I did the fresh install of the latest nightly dmg (http://eclipse.baeyens.it/download/product/mac64.2016-02-20_17-19-59.tar.gz) it hasn't found any updates. At the end, there is now another error message: "No repository found at http://eclipse.bayens.it/V3"

universpharmacie, 2016-02-21:
My wrong, I forgot to go into Preferences and selecting the nightly update site. Now it found updates and installs these without problems. :)

wimjongman, 2016-02-21:
>   Software being installed: Arduino Eclipse Extensions 3.0.0.201602191959 (it.baeyens.arduino.feature.feature.group 3.0.0.201602191959)
  Software currently installed: Eclipse e4 Rich Client Platform 1.3.101.v20150204-1353 (org.eclipse.e4.rcp.feature.group 1.3.101.v20150204-1353)
  Only one of the following can be installed at once: 
    Core Runtime 3.10.0.v20140318-2214 (org.eclipse.core.runtime 3.10.0.v20140318-2214)
    Core Runtime 3.11.1.v20150903-1804 (org.eclipse.core.runtime 3.11.1.v20150903-1804)

What happens here is the following:

1. You try to install the plugin.
2. The plugin requires o.e.core.runtime 3.11 See [1]
3. Installed is o.e.core.runtime 3.10
4. o.e.core.runtime cannot be uninstalled because of other dependencies it has

Resolution: Most of the times, the version number is too restricted. For example, there is no need for the plugin to depend on 3.11. In general only the major version number is important but basically you can also get rid of the version number altogether .

[1] https://github.com/jantje/arduino-eclipse-plugin/blob/master/it.baeyens.arduino.core/META-INF/MANIFEST.MF#L47

jantje, 2016-02-21:
@wimjongman 
Do I understand correctly that you do not think the upgrade problem is related to the framework error?

wimjongman, 2016-02-21:
It is not related.

jantje, 2016-02-22:
@wimjongman 
Your fix fixed the problem on my linux but unfortunately my system did not really upgrade due to the framework errors.
`org.osgi.framework.BundleException: Could not resolve module: it.baeyens.arduino.core [1271]
  Another singleton bundle selected: osgi.identity; osgi.identity="it.baeyens.arduino.core"; type="osgi.bundle"; version:Version="2.4.0.201512061821"; singleton:="true"

	at org.eclipse.osgi.container.Module.start(Module.java:434)
	at org.eclipse.osgi.container.ModuleContainer$ContainerStartLevel.dispatchEvent(ModuleContainer.java:1498)
	at org.eclipse.osgi.container.ModuleContainer$ContainerStartLevel.dispatchEvent(ModuleContainer.java:1)
	at org.eclipse.osgi.framework.eventmgr.EventManager.dispatchEvent(EventManager.java:230)
	at org.eclipse.osgi.framework.eventmgr.EventManager$EventThread.run(EventManager.java:340)`
As you can see he is actually using 2.4.0.201512061821 instead of a the 3.0 version of today

wimjongman, 2016-02-22:
Yes because it is not related ;)

Can you tell me how I can reproduce this? You install V2 of the product en then upgrade with V3 of the plugins and after this you get this message? Is it breaking something or is it just an error? Maybe we should open a separate issue instead of abusing this poor issue.

jantje, 2016-02-22:
The issue I just reported is exactly the same "Framework error"
It is really easy to reproduce.
Download and install a nightly version (but not the latest). I used win64.2016-02-17_18-40-37.tar.gz
once startedup activate the nightly site as update site.
update the plugin
after restarting look in the error log
voila

the dump of my log is below:
`!SESSION 2016-02-22 16:56:04.632 -----------------------------------------------
eclipse.buildId=unknown
java.version=1.8.0_66
java.vendor=Oracle Corporation
BootLoader constants: OS=win32, ARCH=x86_64, WS=win32, NL=en_GB
Framework arguments:  -perspective it.baeyens.arduino.application.perspective
Command-line arguments:  -os win32 -ws win32 -arch x86_64 -perspective it.baeyens.arduino.application.perspective

!ENTRY org.eclipse.egit.ui 2 0 2016-02-22 16:56:24.485
!MESSAGE Warning: The environment variable HOME is not set. The following directory will be used to store the Git
user global configuration and to define the default location to store repositories: 'C:\Users\jan'. If this is
not correct please set the HOME environment variable and restart Eclipse. Otherwise Git for Windows and
EGit might behave differently since they see different configuration options.
This warning can be switched off on the Team > Git > Confirmations and Warnings preference page.

!ENTRY org.eclipse.equinox.p2.core 4 0 2016-02-22 16:58:13.278
!MESSAGE Provisioning exception
!STACK 1
org.eclipse.equinox.p2.core.ProvisionException: No repository found at http://eclipse.baeyens.it/V3.
	at org.eclipse.equinox.internal.p2.repository.helpers.AbstractRepositoryManager.fail(AbstractRepositoryManager.java:395)
	at org.eclipse.equinox.internal.p2.repository.helpers.AbstractRepositoryManager.loadRepository(AbstractRepositoryManager.java:692)
	at org.eclipse.equinox.internal.p2.metadata.repository.MetadataRepositoryManager.loadRepository(MetadataRepositoryManager.java:96)
	at org.eclipse.equinox.internal.p2.metadata.repository.MetadataRepositoryManager.loadRepository(MetadataRepositoryManager.java:92)
	at org.eclipse.equinox.p2.ui.LoadMetadataRepositoryJob.doLoad(LoadMetadataRepositoryJob.java:117)
	at org.eclipse.equinox.p2.ui.LoadMetadataRepositoryJob.runModal(LoadMetadataRepositoryJob.java:102)
	at org.eclipse.equinox.internal.p2.ui.sdk.PreloadingRepositoryHandler$2.runModal(PreloadingRepositoryHandler.java:83)
	at org.eclipse.equinox.p2.operations.ProvisioningJob.run(ProvisioningJob.java:177)
	at org.eclipse.core.internal.jobs.Worker.run(Worker.java:55)
!SUBENTRY 1 org.eclipse.equinox.p2.metadata.repository 4 1000 2016-02-22 16:58:13.279
!MESSAGE No repository found at http://eclipse.baeyens.it/V3.

!ENTRY org.eclipse.equinox.p2.core 4 0 2016-02-22 16:58:35.646
!MESSAGE Provisioning exception
!STACK 1
org.eclipse.equinox.p2.core.ProvisionException: No repository found at http://eclipse.baeyens.it/V3.
	at org.eclipse.equinox.internal.p2.repository.helpers.AbstractRepositoryManager.fail(AbstractRepositoryManager.java:395)
	at org.eclipse.equinox.internal.p2.repository.helpers.AbstractRepositoryManager.loadRepository(AbstractRepositoryManager.java:653)
	at org.eclipse.equinox.internal.p2.metadata.repository.MetadataRepositoryManager.loadRepository(MetadataRepositoryManager.java:96)
	at org.eclipse.equinox.internal.p2.metadata.repository.MetadataRepositoryManager.loadRepository(MetadataRepositoryManager.java:92)
	at org.eclipse.equinox.p2.ui.LoadMetadataRepositoryJob.doLoad(LoadMetadataRepositoryJob.java:117)
	at org.eclipse.equinox.p2.ui.LoadMetadataRepositoryJob.runModal(LoadMetadataRepositoryJob.java:102)
	at org.eclipse.equinox.internal.p2.ui.sdk.PreloadingRepositoryHandler$2.runModal(PreloadingRepositoryHandler.java:83)
	at org.eclipse.equinox.p2.operations.ProvisioningJob.run(ProvisioningJob.java:177)
	at org.eclipse.core.internal.jobs.Worker.run(Worker.java:55)
!SUBENTRY 1 org.eclipse.equinox.p2.metadata.repository 4 1000 2016-02-22 16:58:35.646
!MESSAGE No repository found at http://eclipse.baeyens.it/V3.

!ENTRY org.eclipse.equinox.p2.core 4 0 2016-02-22 16:58:41.677
!MESSAGE Provisioning exception
!STACK 1
org.eclipse.equinox.p2.core.ProvisionException: No repository found at http://eclipse.baeyens.it/V3.
	at org.eclipse.equinox.internal.p2.repository.helpers.AbstractRepositoryManager.fail(AbstractRepositoryManager.java:395)
	at org.eclipse.equinox.internal.p2.repository.helpers.AbstractRepositoryManager.loadRepository(AbstractRepositoryManager.java:653)
	at org.eclipse.equinox.internal.p2.metadata.repository.MetadataRepositoryManager.loadRepository(MetadataRepositoryManager.java:96)
	at org.eclipse.equinox.internal.p2.metadata.repository.MetadataRepositoryManager.loadRepository(MetadataRepositoryManager.java:92)
	at org.eclipse.equinox.p2.ui.LoadMetadataRepositoryJob.doLoad(LoadMetadataRepositoryJob.java:117)
	at org.eclipse.equinox.p2.ui.LoadMetadataRepositoryJob.runModal(LoadMetadataRepositoryJob.java:102)
	at org.eclipse.equinox.internal.p2.ui.sdk.PreloadingRepositoryHandler$2.runModal(PreloadingRepositoryHandler.java:83)
	at org.eclipse.equinox.p2.operations.ProvisioningJob.run(ProvisioningJob.java:177)
	at org.eclipse.core.internal.jobs.Worker.run(Worker.java:55)
!SUBENTRY 1 org.eclipse.equinox.p2.metadata.repository 4 1000 2016-02-22 16:58:41.677
[... 39 lines cut ...]
	at org.eclipse.osgi.container.Module.start(Module.java:434)
	at org.eclipse.osgi.container.ModuleContainer$ContainerStartLevel.dispatchEvent(ModuleContainer.java:1498)
[... cut ...]
!MESSAGE Warning: The environment variable HOME is not set. The following directory will be used to store the Git
user global configuration and to define the default location to store repositories: 'C:\Users\jan'. If this is
not correct please set the HOME environment variable and restart Eclipse. Otherwise Git for Windows and
EGit might behave differently since they see different configuration options.
This warning can be switched off on the Team > Git > Confirmations and Warnings preference page.
`

jantje, 2016-02-22:
in the windopws case above the framework decides to take the last (and as such correct one)
On my linux system the framework decides to take an old version.
The second case is very confusing as you have the latest version on your system but the framework decides to use an old version. You need to look in the error windows to find some cryptical message to understand what is going on.

jantje, 2016-02-23:
If I understand correctly this should fix the problem when I try to update tomorrow?
Can you elaborate on the startup sequence relation? It sounds like hocus pocus to me.

wimjongman, 2016-02-24:
Yes.

Startlevels determine which bundles gets activated and when. Normally bundles load lazy, meaning that they only start (i.e. Activator class is run) when referenced.

Some bundles do housekeeping and must be started early hence the startlevel and autostart boolean. When you now think, "hey, I have a use for it": This concept is broken and leads to severe headaches. It should be avoided as much as possible.

jantje, 2016-02-24:
I know I used this to try to fix a problem a long time ago. Maybe we should get rid of it all together.
As I reported on your pull request this -unfortunately- did not fix the issue (that is if it should fix broken installations)

wimjongman, 2016-02-24:
No you should leave the startlevels as they are now. They are needed for proper running of Eclipse. I will continue to try to find the problem.

jantje, 2016-02-24:
txs wim.
think about the mega :-)

jantje, 2016-02-28:
I think I found the root cause of the issue :)
I'm currenty assuming this is only a problem wit the product.
When you look at help->About the arduino eclipse id->installation detail
You will see the Arduino eclipse extensions and the eclipse arduino IDE (it.baeyens.arduino.product).
The IDE is V2.4.0 and the feature is V3.YY
I tried to upgrade the product but he doesn't find a update.
I tried uninstalling the product but that killed my installation (binary is gone).
I didn't believe that so I tried uninstalling another install and again my installation was corrupted.

The reason I'm pretty sure this is the root cause is because there are also plenty of non arduino plugins that have the same issue. If this was a "general eclipse issue" google would have found something already.

[26 more replies]

commit 19204ffd (2016-02-29): together with org.eclipse.update.reconcile=false this seems to fix #321

Installation of the plugin in java ide is now possible
Upgrade from 2.4 works 
Product works
commit 3bc5517d (2016-03-10): Revert "together with org.eclipse.update.reconcile=false this seems to fix #321"

This reverts commit 19204ffd413472da4f21ae51ab80fb8391b182c0.
closed 2016-04-01 by jantje