Sloeber/arduino-eclipse-plugin #323: EclipseArduino Lockup When Serial Port in Monitor Opened a Second Time - Linux

rmerriam, 2015-11-05:
Hi,

Running Ubuntu 14.04 with the Eclipse 4.4.2, CDT 8.60, Arduion Eclipse Extension 2.4.0. The Arduino IDE is 1.6.5-r5. I've seen this problem previously with earlier versions.

The serial monitor is open to a port. You click + to add another port and a dialog comes up: Serial Port is in use by another applications.... Method Name: open port

A similar lockup occurs if you select a non-existent port that for some reason appears on the list. 

No matter what is clicked, except Details, Eclipse and serial monitor are locked and the dialog will not go away. A End process on the Java instance generates a dialog box with the following text below.

Rud

JVM terminated. Exit code=143
/usr/bin/java
-Dosgi.requiredJavaVersion=1.7
-Xms40m
-Xmx512m
-XX:MaxPermSize=256M
-jar /opt/eclipseArduino//plugins/org.eclipse.equinox.launcher_1.3.0.v20140415-2008.jar
-os linux
-ws gtk
-arch x86_64
-showsplash
-launcher /opt/eclipseArduino/eclipseArduinoIDE
-name EclipseArduinoIDE
--launcher.library /opt/eclipseArduino//plugins/org.eclipse.equinox.launcher.gtk.linux.x86_64_1.1.200.v20150204-1316/eclipse_1607.so
-startup /opt/eclipseArduino//plugins/org.eclipse.equinox.launcher_1.3.0.v20140415-2008.jar
--launcher.appendVmargs
-exitdata aa28015
-perspective it.baeyens.arduino.application.perspective
-vm /usr/bin/java
-vmargs
-Dosgi.requiredJavaVersion=1.7
-Xms40m
-Xmx512m
-XX:MaxPermSize=256M
-jar /opt/eclipseArduino//plugins/org.eclipse.equinox.launcher_1.3.0.v20140415-2008.jar

TrippyLighting, 2015-11-05:
I am wondering if this has to do with the same topic I've decribed a fix for on my Blog.

http://trippylighting.com/teensy-arduino-ect/arduino-eclipse-plugin/the-serial-monitor-on-mac-osx/

The post has links to the the original sources of the fix that explain the details better.

jantje, 2015-11-06:
@all 
I decided to no longer respond to github issues -and other questions- due to a lack of support.
As a patron you can post a message on the patreon forum stating you support a github issue. This allows me to support people who support me.
https://www.patreon.com/posts/3700469
.....
I myself support this issue because Rud is the author of a nice article on hackaday. Patreon seems down so I can't log it there at the moment.

@rmerriam 
Since I developed my bridge for the yun I havn't used the serial monitor a lot but .... I have never seen something like you describe.
Does the pop-up come up when you press the + or when you press ok to try to add the port?
When you press the + the plugin tries to list all the com ports on the system. There have been report on arduino IDE in regards to bluetooth devices causing long delays.

Please also try what @TrippyLighting is proposing. Even though this is from the "rxtx" days it still may work. Who knows?

I'll try the "bad com port name" as you described next weekend.

rud-had, 2015-11-06:
I checked and there is a /var/lock directory owned by root on my system. It has all permissions set. 

The error occurs when clicking on "Ok" in the dialog. Basically I'm telling the serial monitor to open the port a second time. It brings up the error dialog and that locks up everything.

jantje, 2015-11-06:
I just checked this on my windows machine. I can confirm that trying to open the same com port twice results in a hand.

jantje, 2015-11-06:
I found the bug in the code. I started a new build from a blank workspace. Should be available in a couple of hours.

rmerriam, 2015-11-06:
Very good work, Jantje.

jantje, 2015-11-07:
txs for confirming this is fixed

closed 2015-11-07 by jantje