Sloeber/arduino-eclipse-plugin #325: Still a little "convenience" thing

universpharmacie, 2015-11-07:
The Eclipse Arduino IDE remembers well on one side the last active project at restart/relaunch, but the last attached com port in the serial monitor seems not to be saved and one has to re-attach it at every launch of the IDE. Would be nice to have the last setting restored here, too.

jantje, 2015-11-07:
do you mean saving the last opened serial port name or restoring the serial connection that were open when eclipse closed?
Thinking about this. Clicking the serial port icon may also open the serial monitor connection (if I store/request the last used baud rate per project)

universpharmacie, 2015-11-07:
Let me describe my actual workflow to see better what I mean: 
I have got an idea on how to improve my code. Thus I launch the Eclipse Arduino IDE which takes much time and makes the fans of my MacBook turn high during the C/C++ indexing (...swear words...), I attach the Teensy, I do some code changes and I click the upload icon. Then I click on the serial icon, eager to see the result. But there is nothing, (...more swear words...) the COM port settings are empty and I have to re-attach the COM port with the (+) icon. Press the upload icon again, get the "no project selected" error since the serial monitor has still the focus (...still more swear words...), close it, the error console opens but I don't care, I click into the code window, click the upload icon again, click on the serial tab again, and Ill finally see the result... :-)

jantje, 2015-11-07:
It is unclear to me why you have to do the following step "Press the upload icon again, get the "no project selected" error"

Next to that:
If issue #324 is fixed and if The project configuration knows the baud rate(this issue) and you close the error window(I do as it annoys me to much popping up due to other plugins) the workflow would be like this:
I have got an idea on how to improve my code. Thus I launch the Eclipse Arduino IDE which takes much time and makes the fans of my MacBook turn high during the C/C++ indexing (...swear words...), I attach the Teensy, I do some code changes and I click the upload icon. Then I click on the serial icon in the toolbar, the serial monitor connects with the teensy based on the com port and baud rate specified in the currently selected project properties :-)

universpharmacie, 2015-11-07:
Ok, the "upload again" is not forcibly needed, I took the habit to do so to make sure that I didn't miss anything. But you are right, that step is most times obsolete.

Out of that, the workflow you describe seems very sexy if it can be realized :-)

jantje, 2015-11-07:
I plan to do some major changes after the next release. I would not see why these two issues could not be part of it.

universpharmacie, 2015-11-07:
Great, thank you! Looking forward to it. :+1:

jantje, 2015-12-30:
I think about adding a baud rate to the arduino configuration.
When the serial monitor icon is clicked: for all selected projects where the baud rate is present and no serial connection is already in the monitor a serial connection is tried to be made with the baud rate.

What do you think?

jantje, 2016-01-20:
I didn't add the baud rate to the properties. I read it from the arduino code :-)
cool isn't it?

closed 2016-04-01 by jantje