Sloeber/arduino-eclipse-plugin #326: Imported folders may not contain characters like minus or dot.

george-hawkins, 2015-11-07:
If you go to Import &rarr; Arduino &rarr; Import a folder... and then try to import a directory with a name that contains a character like `.` (dot) or `-` (minus) then it fails with the exception:
```
org.eclipse.core.runtime.CoreException: Path variable name cannot contain character: -.
	at org.eclipse.core.internal.resources.PathVariableManager.checkIsValidName(PathVariableManager.java:68)
	at org.eclipse.core.internal.resources.PathVariableManager.setValue(PathVariableManager.java:220)
	at org.eclipse.core.internal.resources.PathVariableManager.setURIValue(PathVariableManager.java:322)
	at it.baeyens.arduino.tools.ArduinoHelpers.addCodeFolder(ArduinoHelpers.java:304)
	at it.baeyens.arduino.ui.Import_Source_Folder.performFinish(Import_Source_Folder.java:170)
	at org.eclipse.jface.wizard.WizardDialog.finishPressed(WizardDialog.java:853)
	at org.eclipse.jface.wizard.WizardDialog.buttonPressed(WizardDialog.java:438)
```

If you look at the Eclipse validation code involved ([`PathVariableManager.java:281`](http://git.eclipse.org/c/platform/eclipse.platform.resources.git/tree/bundles/org.eclipse.core.resources/src/org/eclipse/core/internal/resources/PathVariableManager.java#n281)) it looks like basically only letters and digits are allowed.

It seems odd that the parent directories can contain whatever characters one wants - these restrictions only apply to the name of the last directory in the path being checked.

So `/xxx/fo-o/bar` is legal but `/xxx/foo/ba-r` is not.

I know the validation code is from Eclipse but it's being called by `ArduinoHelpers` - is it really appropriate in this situation to do so? I seem to be able to link to existing folders containing source by using the classic Eclipse functionality like so:

* Right click on my project and go to New &rarr; Folder &rarr; Advanced.
* Click "Link to alternate location" and browse to the required folder.
* Press Finish and then right click on the project again and go to Properties &rarr; C/C++ General &rarr; Paths and Symbols.
* Click on the Source Location tab, click Add Folder and add my newly linked folder.

If I do things this way then unlike the Arduino import functionality it doesn't complain about characters like dot or minus.

In my case the directory name was something like `test.2` and this resulted in an initially very confusing error from Eclipse - `Path variable name cannot contain character: ..` - I read `..` as meaning the parent directory, in the normal UNIX way, and didn't understand what was meant at all. Always adding a full stop at the end of a sentence may be good grammar but in this case it was pretty confusing :smile:

jantje, 2015-11-09:
@george-hawkins 
txs for the input. I'm not sure what to do about this.

jantje, 2015-12-17:
FYI
I'm currently rewriting the plugin to contain a boards an library manager. This to make it run without arduino ide (and as such without the need of configuration)
The path variables were there to make swapping arduino IDE possible. So the path variables are out and that should fix this issue.

closed 2016-04-01 by jantje