Sloeber/arduino-eclipse-plugin #327: Separate private and provided libraries.

TrippyLighting, 2015-11-15:
IN previous versions of the plugin/product, when adding a library to a project it was easy to determine if it was a private library or one of the provided standard libraries becasue these were sorted into two categories. It was possible to have the same library with the same file and folder names in the santard and private library locations

 Now they are all mixed together in one big list and one has to change one of the folder names.

In essence if there are two libraries with the same folder name, one located in the Arduino standard libraries and one in the private libraries, now the folder name has to be changes in order to separate them.

For example my source code works with an older version of the i2c-t3 library of which I have a copy in my private libraries but it does not work with the most recent i2c_t3 library that's included in  Teensyduino 1.26. When attempting to add the library I am only presented with the i2c_t3 library that comes with Teensyduino but not with the one in my private libraries. I have to re-name one of the folder names for the different name to be available for selection in the  list.

jantje, 2015-11-15:
@all
I decided to no longer respond to github issues -and other questions- due to a lack of support.
As a patron you can post a message on the patreon forum stating you support a github issue. This allows me to support people who support me.
https://www.patreon.com/posts/3700469
.....
TrippyLighting is long lasting supporter of the plugin.
http://trippylighting.com/teensy-arduino-ect/arduino-eclipse-plugin/
An a Patron.
Txs TrippyLighting

@TrippyLighting 
I'm really sorry about this, but except for explaining; there is little I can do about this. I'll explain why later on, let me first tell you how you should handle your problem with the plugin.

In the arduino menu under the item "add a library to the selected project" there is **"add a source folder to the selected project"**. Using the "source folder" menu is the way to go for your situation. Note that there has been some degradation on the original code but it works on my system as long as you add the code to the root. You can move it to a subfolder after the import. Do not use the existing libraries folder as this is the folder with the folders added by "add a library to the selected project". Moving it to a subfolder probably requires a update to the include path.
Feel free to report any issues you have on "add a source folder to the selected project" as I feel this functionality will increase in importance and has had some degradation.

 
**From a C/C++ coding point of view they both do the same.** Basically select a folder; add the folder to the source tree and add the folder to the include path.
**From a ide point of view they are very different.**
*With the "add a library to the selected project" you can only select folders that are "known in the arduino world".* That is the folders in the libraries folder of the arduino installation, the folders in ~/arduino/libraries the folders in the libraries subfolder of the selected hardware and the folders in the private libraries folder.
*Secondly you are not in direct control of which folder is selected.* This is to mimic the arduino IDE functionality of adding a #include statement to the code and let the arduino IDE find the folder to add to the include path and source path.
This means that when you change the hardware (fi teensy->uno) all the folders included with the "add a library to the selected project" may change. Try with SPI and/or WIRE and you'll understand why this is a good thing.

*With the "add a source folder to the selected project" you can add any folder known by your os to the code.* The IDE does not take control over the folders you are in full control.

I hope this explains the difference between the 2 functionalities.
Best regards
Jantje

TrippyLighting, 2015-11-15:
Hi Jantje,

Thanks for the feedback. Yes, this explains the functionality. I had not thought about adding a library as as source. That should work fine for my purposes.

On the other hand, renaming a library folder is also not exactly difficult to do ;-)

I will continue to ask questions through patreon.

jantje, 2015-11-26:
:-)
I'm closing this one.
Best regards
Jantje

closed 2015-11-26 by jantje

TrippyLighting, 2015-11-30:
I just wanted to leave one more comment. As mentioned in my last post, renaming a library is really not a problem and actually a better way to organize oneself.
It is a really nice touch that we now see in the long list of libraries , which one are already part of the project. It is also really nice to be able to select and un-select libraries, wichout having to delete these from the project!

jantje, 2015-11-30:
@TrippyLighting thanks for the comment and feedback 
I understand you as a committed user of the plugin prefer a rename.
The problem I have with "renaming the library as a general solution for version conflicts" is that it is arduino IDE incompatible. This because the arduino IDE looks at the content of the library folder and not the folder name. I know they made changes to make it more robust but I don't know how robust it is or ever will be.

And yes we have come another long way. 
The ability to select and un-select libraries was a great contribution. Thanks to @riban-bw for that :-)