Sloeber/arduino-eclipse-plugin #328: Switch to @PlatformIO cross-platform code builder and library manager

ivankravets, 2015-11-21:
Dear @jantje,

How do you look to switch to @PlatformIO cross-platform code builder and library manager? The one thing which need to implement is "pre-filler" for [platformio.ini](http://docs.platformio.org/en/latest/projectconf.html). 

You can see how [PlatformIO](http://platformio.org) can be integrated with Eclipse http://docs.platformio.org/en/latest/ide/eclipse.html

This plugin with cross-platforms support and more then 150+ embedded boards could be a good candidate for "Eclipse for Embedded developers", not just for Arduino.

WDYT?

jantje, 2015-11-21:
>How do you look to switch to @PlatformIO cross-platform code builder and library manager?
WDYT?

I had not heard of PlatformIO so I don"t look at anything in this area.
Now I know about it I may consider action.
IMHO I see no reason to look at "switching to" If there is anything I should look at it is "integration in" or "making a new plugin". Besides The Arduino eclipse plugin and the Arduino IDE (since version 1.6.5) support a whole bunch of platforms (not only Arduino)
Besides the Arduino architecture looks very mutch like the one of plotformio
[image]

ivankravets, 2015-11-21:
>  If there is anything I should look at it is "integration in" or "making a new plugin".

Are you interested in it?

> Besides The Arduino eclipse plugin and the Arduino IDE (since version 1.6.5) support a whole bunch of platforms (not only Arduino)

Do you mean under "platforms" word "embedded boards"? So far as PlatformIO has support for more than 150+ embedded boards and 10+ development platforms (including desktop platforms and card-sized PC).

PlatformIO allows users to choose different frameworks, OS without any dependencies to external software. The only one requirement is Python interpreter. The behaviour of PlatformIO is the same on the multiple host OS. The single magic command `pip install platformio` allows users to forget about multiple IDEs per platforms, different approaches/configurations on the host OS, etc.

As for the picture with overall architecture, please take a look at [platformio.ini](http://docs.platformio.org/en/latest/projectconf.html), where you can define multiple development platforms, frameworks, build flags, etc and build+upload single source code/firmware to the multiple targets.

jantje, 2015-11-21:
>Are you interested in it?

right now: no. But maybe you can get me interested. How big is the user base? What is the target audience of PlatformIO?

>Do you mean under "platforms" word "embedded boards"? 

Referring to the image above.
The platform.txt and boards.txt allow you to specify the "Arduino commands" to talk to the firmware uploader and the firmware builder.
With "Arduino commands" I mean stuff like "upload" "build"
I would say that in Arduino world the platformio.ini is the selected platform.txt and board from boards.txt and is located above embedded development. This because the external libraries can be platform dependent (typically SPI WIRE ..)

I have made a platform.txt and boards.txt to mimic the arduino boards to compile locally to make it easier to debug arduino code. And that was not difficult.
If you are interested in this you can read more about this here 
http://iloapp.baeyens.it/blog/blog?Home&post=19
http://iloapp.baeyens.it/blog/blog?Home&post=20
http://iloapp.baeyens.it/blog/blog?Home&post=23

For me Arduino in "arduino IDE" and "arduino eclipse plugin" no longer stands for IDE to work with Arduino hardware (or compatibles) but for hardware that comes with arduino configuration files.

Best regards
Jantje

ivankravets, 2015-11-21:
> right now: no. But maybe you can get me interested. How big is the user base? What is the target audience of PlatformIO?

Will email privately as for these questions. Thanks.

> I would say that in Arduino world the platformio.ini is the selected platform.txt and board from boards.txt and is located above embedded development. This because the external libraries can be platform dependent (typically SPI WIRE ..)

Ok, let's re-phrase it from the beginning. Yes, user can compile the same code with multiple targets/boards if prepares appropriate toolchains, upload tools, specify correct build flags, include extra libs end etc. How many hobbyists are ready to do these things?

PlatformIO it isn't only the builder. It is a complete eco-system for embedded developer: beginning with from powerful cross-builder which doesn't depend on OS or Makefile and finishing with continuous integration in "one click" and multi-frameworks library manager with dependencies and etc.

How does it look from user side on the popular OS (Win, Linux 32/64, Linux ARM v6+, Mac OS X)?

1. Install PlatformIO via single command: `pip install platformio`
2. Found for which boards need to initialize new project via `platformio boards` command or using [Embedded Boards Explorer](http://platformio.org/#!/boards)
3. Initialise new project `platformio init --board uno --board teensy31 --xxxx`
4. Build source code and upload firmware to the targets devices via `platformio run --target upload`

The approach with the same commands and behaviour is excellent for embedded teams where each developer decides which OS or IDE he will use. In master repo will be located configured PlatformIO-based project which can be compiled on different OS without additional steps. 

PlatformIO has own repository with precompiled, prepared toolchains, uploaders, framework for the all popular host OS and architectures.

greg-ware, 2015-11-22:
This Platform.io looks interesting, they claim support for Espressif ESP-01 and ESP-12.
One of the features I like the most in the current Arduino plugin for ESP is the management of the COM port, where you don't have to unconnect while uploading...
Does Platform.io also have intelligent SerialMonitor?
Thanks for the pointer,
Greg

ivankravets, 2015-11-23:
Dear @greg-ware,

> One of the features I like the most in the current Arduino plugin for ESP is the management of the COM port, where you don't have to unconnect while uploading...

@PlatformIO has built-in upload port finder. I hope that it is that you mean.

> Does Platform.io also have intelligent SerialMonitor?

Yes, it has. See http://docs.platformio.org/en/latest/userguide/cmd_serialports.html#platformio-serialports-monitor

Regards, Ivan.

jantje, 2016-01-13:
oept that should have said fix for #348

closed 2016-02-15 by jantje