Sloeber/arduino-eclipse-plugin #330: makefile:78: recipe for target 'arduino.ar' failed

codisto, 2015-12-06:
in the generated makefile the in the arduino.ar section the filename is missing.

arduino.ar: $(AR_OBJ)
        @echo 'Starting archiver'
        "/home/ubu/arduino/arduino-1.6.6/hardware/tools/avr/bin/avr-ar" rcs  ""   $(AR_OBJ)
        @echo 'Finished building: $@'
        @echo ' '

I use  eclipse juno with the nighly build.

Eclipse IDE for Java Developers
Version: Juno Service Release 2
Build id: 20130225-0426

Arduino Eclipse Extensions      2.4.0.201512040209 it.baeyens.arduino.feature.feature.group     Jan Baeyens
    C/C++ Development Tools     8.6.0.201502131403      org.eclipse.cdt.feature.group   Eclipse CDT
  AVR Eclipse Plugin    2.4.1   de.innot.avreclipse.feature.group       avr-eclipse.sourceforge.net

If i build from the git sources, with eclipse mars, there is no problem.

jantje, 2015-12-06:
I decided to no longer respond to github issues -and other questions- due to a lack of support.
As a patron you can post a message on the patreon forum stating you support a github issue. This allows me to support people who support me.
https://www.patreon.com/posts/3700469

alexandrezia, 2015-12-12:
I'm having the same issue

rlogiacco, 2015-12-13:
Please upgrade to Mars

closed 2015-12-13 by rlogiacco

brodao, 2018-02-11:
Try change A.BUILD.PATH variable to `${git_work_tree}/${ConfigName}`
[image]

jantje, 2018-02-12:
@brodao With setting all these build variables you actually are not using sloeber.
There are work arounds that require build variables but normally you should have none.