Sloeber/arduino-eclipse-plugin #332: Include ArduinoHardwareLibPath in arduino resourse folder

mattb5906, 2015-12-07:
Include linked ArduinoHardwareLibPath folder (hardware\arduino\avr\libraries) in the auto generated arduino resource folder by default. The SPI, EEPROM etc. would then work as soon as a new project is created. It took me a while to find these libraries as a new user.

**Original:**
[image]

**Suggested:**
[image]

mattb5906, 2015-12-07:
Just saw the patreon thing. Will add my donation soon.

TrippyLighting, 2015-12-07:
Awesome!

jantje, 2015-12-07:
See http://eclipse.baeyens.it/how_to.shtml#/f on how to work import libraries.

mattb5906, 2015-12-08:
Thanks jantje. I completely missed that :)

closed 2015-12-08 by mattb5906

jantje, 2015-12-08:
You're welcome.