Sloeber/arduino-eclipse-plugin #334: upgrade to java 8

jantje, 2015-12-17:
As I needed some java 8 functionality I upgraded to java 8 in the big changes branch.

jantje, 2015-12-24:
I seem to be unable to unable to build with maven.
I tried several things but I can't find the solution.
The root cause is a java 8 command. I checked everything ut it looks like all is set is to use java8 but still ...

@wimjongman do you want to take a look?

>[ERROR] Failed to execute goal org.eclipse.tycho:tycho-compiler-plugin:0.22.0:compile (default-compile) on project it.baeyens.arduino.common: Compilation failure: Compilation failure:
[ERROR] /home/jan/workspaces/eclipse_plugin/ArduinoPluginDevelopment/arduino-eclipse-plugin/it.baeyens.arduino.common/src/it/baeyens/arduino/common/ArduinoInstancePreferences.java:[285]
[ERROR] setGlobalValue(KEY_PRIVATE_LIBRARY_PATHS, String.join("\n", folderName)); //$NON-NLS-1$
[ERROR] ^^^^
[ERROR] The method join(String, String[]) is undefined for the type String
[ERROR] /home/jan/workspaces/eclipse_plugin/ArduinoPluginDevelopment/arduino-eclipse-plugin/it.baeyens.arduino.common/src/it/baeyens/arduino/common/ArduinoInstancePreferences.java:[293]
[ERROR] setGlobalValue(KEY_PRIVATE_HARDWARE_PATHS, String.join("\n", folderName)); //$NON-NLS-1$
[ERROR] ^^^^
[ERROR] The method join(String, String[]) is undefined for the type String
[ERROR] 2 problems (2 errors)
[ERROR] -> [Help 1]
[ERROR] 
[ERROR] To see the full stack trace of the errors, re-run Maven with the -e switch.
[ERROR] Re-run Maven using the -X switch to enable full debug logging.
[ERROR] 
[ERROR] For more information about the errors and possible solutions, please read the following articles:
[ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/MojoFailureException
[ERROR] 
[ERROR] After correcting the problems, you can resume the build with the command
[ERROR]   mvn <goals> -rf :it.baeyens.arduino.common
bash-4.1$

jantje, 2015-12-24:
FYI the code is in the bigChange branch

jantje, 2015-12-24:
After I upgraded the default java on my system to 8 mvn verify works.
Looks like travis is not yet running java 8

jantje, 2015-12-24:
Been nosing in the doc of travis and found a way to force oracle java 8
But that gives another error.
Is there incompatibilities between openjdk 8 and oracle 8?

>Testing for PID=2085
>
>Java HotSpot(TM) 64-Bit Server VM warning: ignoring option MaxPermSize=192m; support was removed in 8.0
>
>PID=2085 still running at 180 seconds
>
>[ERROR] Failed to execute goal org.eclipse.tycho:tycho-compiler-plugin:0.22.0:compile (default-compile) on project it.baeyens.arduino.common: Compilation failure: Compilation failure:
>
>[ERROR] /home/travis/build/jantje/arduino-eclipse-plugin/it.baeyens.arduino.common/src/it/baeyens/arduino/common/Common.java:[337]
>
>[ERROR] IResource resource = epart.getEditorInput().getAdapter(IResource.class);
>
>[ERROR] ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

[>ERROR] Type mismatch: cannot convert from Object to IResource

[ERROR] 1 problem (1 error)

[ERROR] -> [Help 1]

[ERROR]

jantje, 2015-12-25:
Reverted to oracle java 8 and added a cast

jantje, 2015-12-25:
The code that causes the oracle java 8 to fail will be replaced by a fix for #324

GHPS, 2016-03-22:
Building the current development version under Kubuntu 14.04 I also got the error
`The method join(String, String[]) is undefined for the type String`
Installing OpenJDK 8 and activating it solved the problem.

A detailed description can be found here:
http://askubuntu.com/questions/464755/how-to-install-openjdk-8-on-14-04-lts

jantje, 2016-03-22:
I found out the hard way there is a problem in ubuntu 14 and java 8
To  help people I made this video: https://www.youtube.com/watch?v=AmePMUuv0uk

closed 2016-04-01 by jantje