Sloeber/arduino-eclipse-plugin #338: Sugestion to new icon

otzen, 2015-12-28:
Hi 

As i have both eclipse, eclipse arduino, and the arduino environment on my quick lauch, I was annoued that the eclipse arduino uses the same icon as arduino. So I created a fusion of the eclipse and the arduino Icon.
The new icon is attached to this message, please feel free to use it, i claim no copyright on it.
And i am certainly no graphWizz to if anyone can improve or create better one, i Will welcome it.

[image]

jantje, 2015-12-28:
I experienced this as well. So I will discuss with the original icon provider.

jantje, 2015-12-29:
Any graphWizz willing to make the needed images that would differentiate the plugin from the Arduino IDE?
https://github.com/jantje/arduino-eclipse-plugin/tree/master/it.baeyens.arduino.application/icons

wimjongman, 2016-01-03:
Closed with [1]

[1] https://github.com/jantje/arduino-eclipse-plugin/commit/4efa44d9740e79df6bbdf0d9766726d9e7573854

closed 2016-01-03 by wimjongman