Sloeber/arduino-eclipse-plugin #342: Change arduino version path

cho934, 2016-01-06:
Hi,
I use Eclipse Mars.
After installing arduino path to 1.6.7 (not yet compatible), I changed the arduino IDE path, but it seems to not change the different include (to 1.6.7) of my project.
Have I to modify all makefile to change the new path or are there some configuration steps in the eclipse interface ?

BR,

jantje, 2016-01-06:
I decided to no longer respond to github issues -and other questions- due to a lack of support.
As a patron you can post a message on the patreon forum stating you support a github issue. This allows me to support people who support me.
https://www.patreon.com/posts/3700469

cho934, 2016-01-06:
ok. I understand.
In the meantime, I found a solution.
- remove the file DUMP_arduino_avr_[Card]_ArduinoIDE.tmp in the workspace
- configure your new preference
- recreate a new project just to recreate the file with the new configuration
- compile this new project
- clean and recompile the old project, the new settings will be applied.

jantje, 2016-01-06:
Great work. You just upped your understanding of the plugin.
:-)

Closing as "user error" (using unsupported ide) and work around documented

Also: These files should have been deleted when you switched ide version.
I'm working on a release to get rid of these dump files.

closed 2016-01-06 by jantje