Sloeber/arduino-eclipse-plugin #347: [3.0.0] Default project cannot be build

wimjongman, 2016-01-13:
How to reproduce:

* Install 3.0.0
* Open a new workspace
* Create a new sketch with default content
* Verify

Result:

11:48:08 **** Incremental Build of configuration Release for project test ****
"C:\\Users\\jongw\\Programs\\eclipseArduino\\arduinoPlugin\\tools\\make\\make" all 
Building file: ../.ino.cpp
Starting C++ compile
"C:\Users\jongw\git\arduino-eclipse-plugin\it.baeyens.arduino.product\target\products\it.baeyens.arduino.product\win32\win32\x86_64\eclipseArduino\arduinoPlugin\tools\arduino\avr-gcc\4.8.1-arduino5/bin/avr-g++" -c -g -Os -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10606 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR     -I"C:\Users\jongw\Programs\eclipseArduino\arduinoPlugin\packages\arduino\hardware\avr\1.6.9\cores\arduino" -I"C:\Users\jongw\Programs\eclipseArduino\arduinoPlugin\packages\arduino\hardware\avr\1.6.9\variants\standard" -MMD -MP -MF".ino.cpp.d" -MT".ino.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "../.ino.cpp"  -o  ".ino.cpp.o"   -Wall
In file included from C:\Users\jongw\Programs\eclipseArduino\arduinoPlugin\packages\arduino\hardware\avr\1.6.9\cores\arduino/Arduino.h:28:0,
                 from ../.ino.cpp:6:
c:\users\jongw\git\arduino-eclipse-plugin\it.baeyens.arduino.product\target\products\it.baeyens.arduino.product\win32\win32\x86_64\eclipsearduino\arduinoplugin\tools\arduino\avr-gcc\4.8.1-arduino5\avr\include\avr\io.h:99:26: fatal error: avr/sfr_defs.h: No such file or directory
 #include <avr/sfr_defs.h>
                          ^
compilation terminated.
subdir.mk:24: recipe for target '.ino.cpp.o' failed
make: *** [.ino.cpp.o] Error 1

11:48:08 Build Finished (took 418ms)

jantje, 2016-01-13:
It looks like the file  avr/sfr_defs.h is corrupt? Can you check that?

wimjongman, 2016-01-13:
I see that io.h is in the same directory as sfr_defs.h. Should then the
include not be

#include <sfr_defs.h> instead of <avr/sfr_defs.h>

The file itself is fine.

c:\users\jongw\git\arduino-eclipse-plugin\it.baeyens.arduino.product\target\products\it.baeyens.arduino.product\win32\win32\x86_64\eclipsearduino\arduinoplugin\tools\arduino\avr-gcc\4.8.1-arduino5\avr\include\avr\io.h:99:26:
fatal error: avr/sfr_defs.h: No such file or directory
 #include <avr/sfr_defs.h>

jantje, 2016-01-13:
>I see that io.h is in the same directory as sfr_defs.h. Should then the include not be #include <sfr_defs.h> instead of <avr/sfr_defs.h>

That is not nessesary as [path before]/avr/sfr_defs.h is in the include path.

I just opened my linux laptop and upgraded the plugin and tested this and it works fine here.

Also what strikes me is that when the compiler prints an error it prints the whole line and you dumped a line only containing "#include"
and the error marker below the # (//forget about this as it is github related)

When I test here and change the name of sfr_defs.h so it can not be found I get following error

>In file included from /home/jan/.plugin/tools/arduino/avr-gcc/4.8.1-arduino5/avr/include/avr/pgmspace.h:88:0,
                 from /home/jan/.plugin/packages/arduino/hardware/avr/1.6.9/cores/arduino/Arduino.h:28,
                 from ../.ino.cpp:6:
/home/jan/.plugin/tools/arduino/avr-gcc/4.8.1-arduino5/avr/include/avr/io.h:99:70: fatal error: avr/sfr_defsg.h: No such file or directory
 #include <avr/sfr_defsg.h> ///some garbage to see wether this appears
                                                                      ^
compilation terminated.
make: *** [.ino.cpp.o] Fout 1

As you can see even my comments are dumped.

So I am really clueless to what is going on on your system.

Can you run this command from the command prompt?

"C:\Users\jongw\git\arduino-eclipse-plugin\it.baeyens.arduino.product\target\products\it.baeyens.arduino.product\win32\win32\x86_64\eclipseArduino\arduinoPlugin\tools\arduino\avr-gcc\4.8.1-arduino5/bin/avr-g++" -c -g -Os -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10606 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR -I"C:\Users\jongw\Programs\eclipseArduino\arduinoPlugin\packages\arduino\hardware\avr\1.6.9\cores\arduino" -I"C:\Users\jongw\Programs\eclipseArduino\arduinoPlugin\packages\arduino\hardware\avr\1.6.9\variants\standard" -MMD -MP -MF".ino.cpp.d" -MT".ino.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "../.ino.cpp" -o ".ino.cpp.o" -Wall

wimjongman, 2016-01-13:
> include not be #include instead of
> #include ///some garbage to see wether this appears

I moved the plugin from the build location to a fresh new directory. I have
the same problems.

running:

"C:\Users\jongw\git\arduino-eclipse-plugin\it.baeyens.arduino.product\target\products\it.baeyens.arduino.product\win32\win32\x86_64\eclipseArduino\arduinoPlugin\tools\arduino\avr-gcc\4.8.1-arduino5/bin/avr-g++"
-c -g -Os -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections
-fno-threadsafe-statics -MMD -mmcu=atmega328p -DF_CPU=16000000L
-DARDUINO=10606 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR
-I"C:\Users\jongw\Programs\eclipseArduino\arduinoPlugin\packages\arduino\hardware\avr\1.6.9\cores\arduino"
-I"C:\Users\jongw\Programs\eclipseArduino\arduinoPlugin\packages\arduino\hardware\avr\1.6.9\variants\standard"
-MMD -MP -MF".ino.cpp.d" -MT".ino.cpp.o" -D__IN_ECLIPSE__=1 -x c++
"../.ino.cpp"  -o  ".ino.cpp.o"   -Wall

on the command line gives me:

avr-g++: error: ../.ino.cpp: No such file or directory
avr-g++: warning: '-x c++' after last input file has no effect
avr-g++: fatal error: no input files
compilation terminated.

Another project shows this in the log:

18:10:53 **** Incremental Build of configuration Release for project wewe
****
"C:\\Users\\jongw\\Programs\\eclipseArduino\\arduinoPlugin\\tools\\make\\make"
all
Building file:
C:/Users/jongw/Programs/eclipseArduino/arduinoPlugin/packages/arduino/hardware/avr/1.6.9/cores/arduino/CDC.cpp
Starting C++ compile
"C:\Users\jongw\git\arduino-eclipse-plugin\it.baeyens.arduino.product\target\products\it.baeyens.arduino.product\win32\win32\x86_64\eclipseArduino\arduinoPlugin\tools\arduino\avr-gcc\4.8.1-arduino5/bin/avr-g++"
-c -g -Os -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections
-fno-threadsafe-statics -MMD -mmcu=atmega328p -DF_CPU=16000000L
-DARDUINO=10606 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR
-I"C:\Users\jongw\Programs\eclipseArduino\arduinoPlugin\packages\arduino\hardware\avr\1.6.9\cores\arduino"
-I"C:\Users\jongw\Programs\eclipseArduino\arduinoPlugin\packages\arduino\hardware\avr\1.6.9\variants\standard"
-MMD -MP -MF"arduino/CDC.cpp.d" -MT"arduino/CDC.cpp.o" -D__IN_ECLIPSE__=1
-x c++
"C:/Users/jongw/Programs/eclipseArduino/arduinoPlugin/packages/arduino/hardware/avr/1.6.9/cores/arduino/CDC.cpp"
-o  "arduino/CDC.cpp.o"   -Wall
In file included from
C:/Users/jongw/Programs/eclipseArduino/arduinoPlugin/packages/arduino/hardware/avr/1.6.9/cores/arduino/CDC.cpp:19:0:
C:/Users/jongw/Programs/eclipseArduino/arduinoPlugin/packages/arduino/hardware/avr/1.6.9/cores/arduino/USBAPI.h:24:26:
fatal error: avr/pgmspace.h: No such file or directory
 #include <avr/pgmspace.h>
                          ^
compilation terminated.
make: *** [arduino/CDC.cpp.o] Error 1
arduino/core/subdir.mk:98: recipe for target 'arduino/CDC.cpp.o' failed

18:10:54 Build Finished (took 424ms)

[image: Inline image 1]

jantje, 2016-01-13:
my bad for the command failing
For the command to succeed it has to be run in the release folder of the project. You can also mod the "../.ino.cpp" part to the full name

jantje, 2016-01-14:
@wimjongman 
Do you have a update on this?
Best regards
Jantje

wimjongman, 2016-01-14:
Yes, here it is:

C:\Users\jongw\workspaces\arduinotest2\test\Release>"C:\Users\jongw\git\arduino-eclipse-plugin\it.baeyens.arduino.product\target\products\it.baeyens.arduino.product\win32\win32\x86_64\eclipseArduino\arduinoPlugin\tools\arduino\avr-gcc\4.8.1-arduino5/bin/avr-g++" -c -g -Os -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10606 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR     -I"C:\Users\jongw\Programs\eclipseArduino\arduinoPlugin\packages\arduino\hardware\avr\1.6.9\cores\arduino" -I"C:\Users\jongw\Programs\eclipseArduino\arduinoPlugin\packages\arduino\hardware\avr\1.6.9\variants\standard" -MMD -MP -MF"arduino/CDC.cpp.d" -MT"arduino/CDC.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "C:/Users/jongw/Programs/eclipseArduino/arduinoPlugin/packages/arduino/hardware/avr/1.6.9/cores/arduino/CDC.cpp"  -o  "arduino/CDC.cpp.o"   -Wall
In file included from C:/Users/jongw/Programs/eclipseArduino/arduinoPlugin/packages/arduino/hardware/avr/1.6.9/cores/arduino/CDC.cpp:19:0:
C:/Users/jongw/Programs/eclipseArduino/arduinoPlugin/packages/arduino/hardware/avr/1.6.9/cores/arduino/USBAPI.h:24:26: fatal error: avr/pgmspace.h: No such file or directory
 #include <avr/pgmspace.h>
                          ^
compilation terminated.

The ^ points to after the h> in fixed width font

C:\Users\jongw\workspaces\arduinotest2\test\Release>

wimjongman, 2016-01-14:
Or this one:

`
c:\users\jongw\git\arduino-eclipse-plugin\it.baeyens.arduino.product\target\products\it.baeyens.arduino.product\win32\win32\x86_64\eclipsearduino\arduinoplugin\tools\arduino\avr-gcc\4.8.1-arduino5\avr\include\avr\io.h:99:26: fatal error: avr/sfr_defs.h: No such file or directory
 #include <avr/sfr_defs.h>
                          ^
`

compilation terminated.

wimjongman, 2016-01-14:
Hey I wonder why it points to:

  c:\users\jongw\git\arduino-eclipse-plugin\it.baeyens.arduino.product\target\products\it.baeyens.arduino.product\win32\win32\x86_64

That is not where my workspace is and also not where my product is. This is the place where I compile and test the product. However I am not working there at all at the moment.

jantje, 2016-01-14:
I really think there is something special on your system. It would be great if we can find what is special because other people will have the same special thing.
Can you do this?
project properties->C/C++ general->preprocessor include ...->Providers->Arduino compiler settings and in the bottom select allocate a console in the console view.
Goto the arduino board selection page and select apply and ok
there should be 2 extra consoles now. 1 for c++ and one for c
can you send the content from the c++ console?

jantje, 2016-01-14:
>Hey I wonder why it points to:

c:\users\jongw\git\arduino-eclipse-plugin\it.baeyens.arduino.product\target\products\it.baeyens.arduino.product\win32\win32\x86_64

That is not where my workspace is and also not where my product is. This is the place where I compile and test the product. However I am not working there at all at the moment.

In all console output you posted here that is where you are working.
Note that the default location is in the installation folder but this can be overruled. 
You can specify the location on the command line as follows -manager_path:XXXXX where XXXXX is the full path to the different location. 

I see 3 options:
1) you start a different eclipse than you think
2) You use -manager_path without knowing 
3) something else 

Currently I see no way to link any of these 3 options to the compile error.

wimjongman, 2016-01-15:
Yes, To make it more real I installed the resulting zip in another location instead of running win32x64.cmd from the root. So that is why the logs are different.

[image]

09:42:25 **** Running scanner discovery: Arduino Compiler Settings ****
"C:\\Users\\jongw\\git\\arduino-eclipse-plugin\\it.baeyens.arduino.product\\target\\products\\it.baeyens.arduino.product\\win32\\win32\\x86_64\\eclipseArduino\\arduinoPlugin\\tools\\arduino\\avr-gcc\\4.8.1-arduino5\\bin\\avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10606 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR -D__IN_ECLIPSE__=1 -E -P -v -dD C:/Users/jongw/workspaces/arduinotest2/.metadata/.plugins/org.eclipse.cdt.managedbuilder.core/spec.C 
Using built-in specs.
COLLECT_GCC=C:\Users\jongw\git\arduino-eclipse-plugin\it.baeyens.arduino.product\target\products\it.baeyens.arduino.product\win32\win32\x86_64\eclipseArduino\arduinoPlugin\tools\arduino\avr-gcc\4.8.1-arduino5\bin\avr-g++
Target: avr
Configured with: ../gcc-4.8.1/configure --enable-fixed-point --enable-languages=c,c++ --prefix=/c/jenkins/workspace/toolchain-avr-windows/objdir --enable-long-long --disable-nls --disable-checking --disable-libssp --disable-libada --disable-shared --enable-lto --with-avrlibc=yes --with-dwarf2 --disable-doc --target=avr
Thread model: single
gcc version 4.8.1 (GCC) 
COLLECT_GCC_OPTIONS='-c' '-g' '-Os' '-w' '-std=gnu++11' '-fno-exceptions' '-ffunction-sections' '-fdata-sections' '-fno-threadsafe-statics' '-MMD' '-mmcu=atmega328p' '-D' 'F_CPU=16000000L' '-D' 'ARDUINO=10606' '-D' 'ARDUINO_AVR_UNO' '-D' 'ARDUINO_ARCH_AVR' '-D' '__IN_ECLIPSE__=1' '-E' '-P' '-v' '-dD'
 c:/users/jongw/git/arduino-eclipse-plugin/it.baeyens.arduino.product/target/products/it.baeyens.arduino.product/win32/win32/x86_64/eclipsearduino/arduinoplugin/tools/arduino/avr-gcc/4.8.1-arduino5/bin/../libexec/gcc/avr/4.8.1/cc1plus.exe -E -quiet -v -P -imultilib avr5 -iprefix c:\users\jongw\git\arduino-eclipse-plugin\it.baeyens.arduino.product\target\products\it.baeyens.arduino.product\win32\win32\x86_64\eclipsearduino\arduinoplugin\tools\arduino\avr-gcc\4.8.1-arduino5\bin\../lib/gcc/avr/4.8.1/ -MMD spec.d -D F_CPU=16000000L -D ARDUINO=10606 -D ARDUINO_AVR_UNO -D ARDUINO_ARCH_AVR -D __IN_ECLIPSE__=1 C:/Users/jongw/workspaces/arduinotest2/.metadata/.plugins/org.eclipse.cdt.managedbuilder.core/spec.C -mmcu=atmega328p -std=gnu++11 -w -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -g -fworking-directory -Os -fno-rtti -fno-enforce-eh-specs -fno-exceptions -dD
#define __STDC__ 1
#define __cplusplus 201103L
#define __STDC_HOSTED__ 1
#define __GNUC__ 4
#define __GNUC_MINOR__ 8
#define __GNUC_PATCHLEVEL__ 1
#define __VERSION__ "4.8.1"
#define __ATOMIC_RELAXED 0
#define __ATOMIC_SEQ_CST 5
#define __ATOMIC_ACQUIRE 2
#define __ATOMIC_RELEASE 3
#define __ATOMIC_ACQ_REL 4
#define __ATOMIC_CONSUME 1
#define __OPTIMIZE_SIZE__ 1
#define __OPTIMIZE__ 1
#define __FINITE_MATH_ONLY__ 0
#define __SIZEOF_INT__ 2
#define __SIZEOF_LONG__ 4
#define __SIZEOF_LONG_LONG__ 8
#define __SIZEOF_SHORT__ 2
#define __SIZEOF_FLOAT__ 4
#define __SIZEOF_DOUBLE__ 4
#define __SIZEOF_LONG_DOUBLE__ 4
#define __SIZEOF_SIZE_T__ 2
#define __CHAR_BIT__ 8
#define __BIGGEST_ALIGNMENT__ 1
#define __ORDER_LITTLE_ENDIAN__ 1234
#define __ORDER_BIG_ENDIAN__ 4321
#define __ORDER_PDP_ENDIAN__ 3412
#define __BYTE_ORDER__ __ORDER_LITTLE_ENDIAN__
#define __FLOAT_WORD_ORDER__ __ORDER_LITTLE_ENDIAN__
#define __SIZEOF_POINTER__ 2
#define __GNUG__ 4
#define __SIZE_TYPE__ unsigned int
#define __PTRDIFF_TYPE__ int
#define __WCHAR_TYPE__ int
#define __WINT_TYPE__ int
#define __INTMAX_TYPE__ long long int
#define __UINTMAX_TYPE__ long long unsigned int
#define __CHAR16_TYPE__ unsigned int
#define __CHAR32_TYPE__ long unsigned int
#define __SIG_ATOMIC_TYPE__ char
#define __INT8_TYPE__ signed char
#define __INT16_TYPE__ int
#define __INT32_TYPE__ long int
#define __INT64_TYPE__ long long int
#define __UINT8_TYPE__ unsigned char
#define __UINT16_TYPE__ unsigned int
#define __UINT32_TYPE__ long unsigned int
#define __UINT64_TYPE__ long long unsigned int
#define __INT_LEAST8_TYPE__ signed char
#define __INT_LEAST16_TYPE__ int
#define __INT_LEAST32_TYPE__ long int
#define __INT_LEAST64_TYPE__ long long int
#define __UINT_LEAST8_TYPE__ unsigned char
#define __UINT_LEAST16_TYPE__ unsigned int
[... 379 lines cut ...]
#define __BUILTIN_AVR_LLKBITS 1
#define __BUILTIN_AVR_UHKBITS 1
#define __BUILTIN_AVR_UKBITS 1
#define __BUILTIN_AVR_ULKBITS 1
#define __BUILTIN_AVR_ULLKBITS 1
#define __BUILTIN_AVR_ABSFX 1
#define __BUILTIN_AVR_ROUNDFX 1
#define __BUILTIN_AVR_COUNTLSFX 1
#define __INT24_MAX__ 8388607L
#define __INT24_MIN__ (-__INT24_MAX__-1)
#define __UINT24_MAX__ 16777215UL
#define __ELF__ 1
#define F_CPU 16000000L
#define ARDUINO 10606
#define ARDUINO_AVR_UNO 1
#define ARDUINO_ARCH_AVR 1
#define __IN_ECLIPSE__ 1
 c:\users\jongw\git\arduino-eclipse-plugin\it.baeyens.arduino.product\target\products\it.baeyens.arduino.product\win32\win32\x86_64\eclipsearduino\arduinoplugin\tools\arduino\avr-gcc\4.8.1-arduino5\bin\../lib/gcc/avr/4.8.1/include-fixed
 c:\users\jongw\git\arduino-eclipse-plugin\it.baeyens.arduino.product\target\products\it.baeyens.arduino.product\win32\win32\x86_64\eclipsearduino\arduinoplugin\tools\arduino\avr-gcc\4.8.1-arduino5\bin\../lib/gcc/avr/4.8.1/../../../../avr/include
End of search list.
[... cut ...] ts/it.baeyens.arduino.product/win32/win32/x86_64/eclipsearduino/arduinoplugin/tools/arduino/avr-gcc/4.8.1-arduino5/bin/../lib/gcc/avr/4.8.1/../../../../avr/lib/
COLLECT_GCC_OPTIONS='-c' '-g' '-Os' '-w' '-std=gnu++11' '-fno-exceptions' '-ffunction-sections' '-fdata-sections' '-fno-threadsafe-statics' '-MMD' '-mmcu=atmega328p' '-D' 'F_CPU=16000000L' '-D' 'ARDUINO=10606' '-D' 'ARDUINO_AVR_UNO' '-D' 'ARDUINO_ARCH_AVR' '-D' '__IN_ECLIPSE__=1' '-E' '-P' '-v' '-dD'

09:42:25 Build Finished (took 160ms)

jantje, 2016-01-15:
>Yes, To make it more real I installed the resulting zip in another location instead of running win32x64.cmd from the root. So that is why the logs are different.

I'm not following what you are trying to say here.
What I do see is that the console output states 
>"C:\Users\jongw\git\arduino-eclipse-plugin\it.baeyens.arduino.product\***target\products\it.baeyens.arduino.product\win32\win32\x86_64\eclipseArduino\***arduinoPlugin\tools\arduino\avr-gcc\4.8.1-arduino5\bin\avr-g++" 

In other words it looks like you run directly from the compiled resources. Which is different from what I do as I mostly run from debug.

From he loggings you provided you can clearly see that he finds the path
>ignoring duplicate directory "c:/users/jongw/git/arduino-eclipse-plugin/it.baeyens.arduino.product/target/products/it.baeyens.arduino.product/win32/win32/x86_64/eclipsearduino/arduinoplugin/tools/arduino/avr-gcc/4.8.1-arduino5/lib/gcc/../../lib/gcc/avr/4.8.1/../../../../avr/include"
....
c:\users\jongw\git\arduino-eclipse-plugin\it.baeyens.arduino.product\target\products\it.baeyens.arduino.product\win32\win32\x86_64\eclipsearduino\arduinoplugin\tools\arduino\avr-gcc\4.8.1-arduino5\bin../lib/gcc/avr/4.8.1/../../../../avr/include

And that is exactly the same at the end as on my system

I however started thinking
>c:\users\jongw\git\arduino-eclipse-plugin\it.baeyens.arduino.product\target\products\it.baeyens.arduino.product\win32\win32\x86_64\eclipsearduino\arduinoplugin\tools\arduino\avr-gcc\4.8.1-arduino5\bin../lib/gcc/avr/4.8.1/../../../../avr/include/avr/sfr_defs.h

is 260 chars long. Is there still a 255 path length limitation somewhere in windows?
And this is what I got
https://msdn.microsoft.com/en-us/library/windows/desktop/aa365247%28v=vs.85%29.aspx#maxpath
>***Maximum Path Length Limitation***
In the Windows API (with some exceptions discussed in the following paragraphs), the maximum length for a path is MAX_PATH, which is defined as 260 characters. A local path is structured in the following order: drive letter, colon, backslash, name components separated by backslashes, and a terminating null character. For example, the maximum path on drive D is "D:\some 256-character path string<NUL>" where "<NUL>" represents the invisible terminating null character for the current system codepage. (The characters < > are used here for visual clarity and cannot be part of a valid path string.)
Note  File I/O functions in the Windows API convert "/" to "\" as part of converting the name to an NT-style name, except when using the "\\?\" prefix as detailed in the following sections

As you have to add a null that is: ***bingo***. 

I think we found the problem :-)

wimjongman, 2016-01-15:
Yes, I think that must be the problem. However as you can see from the screenshot. I do NOT run from the build directory:

C:\Users\jongw\git\arduino-eclipse-plugin\it.baeyens.arduino.product*target\products\it.baeyens.arduino.product\win32\win32\x86_64\eclipseArduino\

I run from

c:\users\jongw\Programs\eclipseArduino

workspace:

c:\users\jongw\workspaces\arduino2

So why does it still think I am running from 

C:\Users\jongw\git\arduino-eclipse-plugin\it.baeyens.arduino.product*target\products\it.baeyens.arduino.product\win32\win32\x86_64\eclipseArduino\

Do you store the initial installation path somewhere?

jantje, 2016-01-15:
I thought I didn't but apparently I do.
Here is the code. 
https://github.com/jantje/arduino-eclipse-plugin/blob/bigChange/it.baeyens.arduino.common/src/it/baeyens/arduino/common/ConfigurationPreferences.java#L36
As you can see there is the command line overwrite I already mentioned and there is a environment variable.
However I see no place where the environment var is set.

jantje, 2016-01-15:
Thinking about it again.
I probably wanted to use a environment variable to save the setting and then went for the arguments stuff so it could be there has been a version which has set the environment variable in your workspace.

jantje, 2016-01-15:
Anyways I think a warning at startup when the download location is bigger than 150 chars is appropriate

jantje, 2016-01-15:
That is for windows machines

wimjongman, 2016-03-13:
Can be closed.

closed 2016-04-01 by jantje