Sloeber/arduino-eclipse-plugin #348: [3.0.0] Example is not copied in the project

wimjongman, 2016-01-13:
* Create a new project
* Select the blink example

Result, no file is created in the project.

jantje, 2016-01-13:
Are you sure the checkbox was on?
I tested this several times with several sketches?
It did happen quite often I clicked the text which did not put the checkbox and then no code is added.

wimjongman, 2016-01-13:
yes, I am sure.

jantje, 2016-01-13:
I just tested it here and it works fine. Can you reproduce this? Is this only with a specific sample?

wimjongman, 2016-01-13:
I made a video here:

http://screencast.com/t/xgCAiWI9

jantje, 2016-01-13:
I can hardly believe my eyes.
Did the same thing here and it works :-(
[image]
Is there anything in the error log?

jantje, 2016-01-13:
got it
It seems that it doesn't work when you also show the debug page.
What movies are good for :-)

jantje, 2016-01-13:
If you go through to debug and go back you will find that your example selection is gone.
That is the reason why there is no code added.
Now I need to find out why the sample program gets deselected???

jantje, 2016-01-13:
I think it is a eclipse root cause. 
As the list of examples is dependent on the selected board I can only fill the sample page when you enter it.
I did that with this code
```java
    @Override
    public IWizardPage getNextPage(IWizardPage page) {
	// TODO Auto-generated method stub
	if (page.equals(this.mNewArduinoSketchWizardCodeSelectionPage)) {
	    String PlatformPath = NewArduinoSketchWizard.this.mArduinoPage.getPlatformFolder().toString();
	    NewArduinoSketchWizard.this.mNewArduinoSketchWizardCodeSelectionPage.setPlatformPath(PlatformPath);
	}
	return super.getNextPage(page);
    }
```

Apparently when you go from the sample selection page to the debug page this code gets called again.
I'll fix it with a optimalisation (only update when the platformpath has changed)

jantje, 2016-01-13:
This is not a 3.0.0 problem only.
thanks for finding this bug wim. I wouldn't have found it without your video.

closed 2016-04-01 by jantje