Sloeber/arduino-eclipse-plugin #350: Web site install section seems broken

TrippyLighting, 2016-01-16:
It looks like the Install section of the web site  is broken. None of the available download menus lead to a nightly or stable build and I always get this:

Not Found

The requested URL /{{navlinks.osx_nightly.url}} was not found on this server.

Really looking forward to V3 BTW.

jantje, 2016-01-16:
Txs or letting me know. I'll lookinto it.

For V3
It is not so obvious to make Teensy work in V3 as it is boardsmanager oriented and there is no boardsmanager support for Teensy 3.X. I contacted Paul and he is not planning to support the boardsmanager. :-(
I still need to send Paul another mail to ask him if he would be willing to make a change in Platform.txt. 
But even if he does that change there will again be a need for documentation for teensy :-(
I'm still considering support for Platformio #328 which would fix that but I have a hard time explaining to platformio that I need what I need to make things work.

jantje, 2016-01-17:
I pushed big change to the nightly as I did some changes in this area. Lets hope this fixes it.

About V3 and Teensy.
Paul is willing to do the change.
Once that is done one only needs to point to the teensy hardware in the preference.

TrippyLighting, 2016-01-17:
That sounds absolutely awesome!

jantje, 2016-01-17:
The build failed last night due to (me forgetting to move to ) java 8.
As a result the web site was not updated.
I just started the build again.

Is java 8 a problem for mac users?

TrippyLighting, 2016-01-17:
I can only speak for my machine which runs the latest version OSX El Capitan 10.11.2.
Java 8 is part of that.

jantje, 2016-01-17:
Cool
The build and upload has finished and as far as I can see everything is back to normal on the website.
If you agree please close this issue.

And big change is now in the nightly.
For teensy: change platform.txt (change that Paul accepted to do)
I replaced
    compiler.path={runtime.tools.avr-gcc.path}/../
with
    compiler.path={runtime.hardware.path}/../tools/
and in preferences-> arduino add [teensyduino installation path]/hardware/teensy
and select create new sketch
Note that the changed does not impact the arduino IDE.

jantje, 2016-01-18:
it is really fixed now

TrippyLighting, 2016-01-18:
Looks like it. Thanks!
The nightly mc builds from 2016-01-17 & 18 ARE ONLY 45 byte in size.
Now that's what I call code optimization ;-)

jantje, 2016-01-18:
it is like that on the build server as well.
We will need to investigate the problem.

jantje, 2016-01-18:
I think I've found it. eclipsearduino folder seems to be eclipsearduino.app now

jantje, 2016-01-19:
now it is 170mb again. Seems like the optimization does not meet your expectations ;-)

closed 2016-01-19 by jantje