Sloeber/arduino-eclipse-plugin #353: Unable to comlete install of plugin

mr-miky, 2016-01-17:
I installed eclipse and your plugin in a new PC which runs Fedora 23.
Everything seems fine but at each successive restart of eclipse appears an error message:

An internal error occurred during: "Finishing the installation ..".
java.lang.NullPointerException

I had a look at the log and it seems that the plugin look for a file in / opt / eclipse / arduinoPlugin that does not exist

look at the attached log file
[log_file.txt](https://github.com/jantje/arduino-eclipse-plugin/files/93570/log_file.txt)

jantje, 2016-01-17:
I decided to no longer respond to github issues -and other questions- due to a lack of support.
As a patron you can post a message on the patreon forum stating you support a github issue. This allows me to support people who support me.
https://www.patreon.com/posts/3700469

A small tip though: As a newbie use the latest stable

mr-miky, 2016-01-18:
I honestly do not understand your answer ...
I have not asked solutions or demanded anything, I just wanted to notify you the problem with the most information possible.

jantje, 2016-01-18:
I think I provided a fix

wimjongman, 2016-03-13:
Can this be closed?

closed 2016-04-01 by jantje