Sloeber/arduino-eclipse-plugin #354: How to add Teensy boards?

hemantsangwan, 2016-01-21:
Toddler v3.0.0 is cool. As it is independent from Arduino IDE now (#306), I am not able to figure out any way to integrate Teensy Boards. :(

hemantsangwan, 2016-01-21:
Also I am getting error at startup: No boards.txt files found in the arduino hardware folders. I looked in:
/home/hemant/arduino/hardware:/home/hemant/Arduino/hardware:
/home/hemant/Work/arduino-eclipse-plugin/it.baeyens.arduino.product/target/products/it.baeyens.arduino.product/linux/gtk/x86_64/opt/eclipseArduino/arduinoPlugin

jantje, 2016-01-21:
I decided to no longer respond to github issues -and other questions- due to a lack of support.
As a patron you can post a message on the patreon forum stating you support a github issue. This allows me to support people who support me.
https://www.patreon.com/posts/3700469

As I think we are nearing a stable toddler I support this under "testing"

>I am on commit (b74567d) 

for other reader: this is the very latest at the time of writing

***First of all about Teensy: ***
I use Teensy myself so I want Toddler to support Teensy as good as possible, but ....
Due to some problems in platform.txt Toddler can not support the current pre Teensyduino 1.28 releases. Paul will do some testing with my proposed changes and if there are no objections Paul said they will be part of Teensyduino 1.28
My advised way to add teensy to toddler is to add the following path to the private hardware
[teensyduino 1.28 or bigger]/hardware/teensy

That being said: as you added the current teensyduino to the "private hardware": you should be able to select Teensy as a board in project properties and in the create new sketch wizard.
Only the build and upload should fail.

***how toddler works***
Toddler is designed to be "start and play" That means that when you start Toddler for the first time it will try to download the Arduino AVR boards and toolchains.
This download and configuration comes with a visual "please hold the line" representation.

As long as the download is running the create new sketch wizard will report back that the download did not finish. This can take up to 20 minutes.

***your system***
It is clear that the initial download failed. Can you check your memory/log files to see why?
This was reported before on linux in #353 and on windows #347 but all case I know of have been dealt with in the version you use.

can you provide the output of 
>ls /home/hemant/Work/arduino-eclipse-plugin/it.baeyens.arduino.product/target/products/it.baeyens.arduino.product/linux/gtk/x86_64/opt/eclipseArduino/arduinoPlugin

This is where all downloaded info should be

Can you point me to a iso file/vmware file of your os? That way I can try and debug here.

 One of the weird things I see is a space in 
>/home/hemant/teensyduino/hardware:/home/hemant/Arduino***/hardware: /home***/hemant/Work/arduino-eclipse-plugin/it.baeyens.arduino.product/target/products/it.baeyens.arduino.product/linux/gtk/x86_64/opt/eclipseArduino/arduinoPlugin
Did you start with a empty workspace? If not please try with a empty workspace?

Best regards
Jantje

hemantsangwan, 2016-01-30:
Hi, I'm sorry for late late response.

Output of
`ls /home/hemant/Work/arduino-eclipse-plugin/it.baeyens.arduino.product/target/products/it.baeyens.arduino.product/linux/gtk/x86_64/opt/eclipseArduino/arduinoPlugin` is:

```
**downloads**           package_esp8266com_index.json  post_processing_platform.txt
**examples**            package_index.json             pre_processing_boards.txt
**libraries**           **packages**                       pre_processing_platform.txt
library_index.json  post_processing_boards.txt
```
(Bold ones are directories)

I am working on my system. Don't have image. :( 
I have tried both empty and existing workspace.

hemantsangwan, 2016-01-30:
Hi, I'm sorry for **late** response.

Output of
`ls /home/hemant/Work/arduino-eclipse-plugin/it.baeyens.arduino.product/target/products/it.baeyens.arduino.product/linux/gtk/x86_64/opt/eclipseArduino/arduinoPlugin` is:

```
**downloads**           package_esp8266com_index.json  post_processing_platform.txt
**examples**            package_index.json             pre_processing_boards.txt
**libraries**           **packages**                       pre_processing_platform.txt
library_index.json  post_processing_boards.txt
```
(Bold ones are directories)

I am working on my system. Don't have image. :( 
I have tried both empty and existing workspace.

jantje, 2016-01-30:
I guess you have issue #358

hemantsangwan, 2016-02-03:
I am now able to import private hardware. So at least this issue can be closed.

closed 2016-02-03 by hemantsangwan

jantje, 2016-02-03:
txs for feedback. I'm marking this issue as a duplicate of #358