Sloeber/arduino-eclipse-plugin #356: Update sites are not correct

wimjongman, 2016-01-24:
* Update sites still point to luna
* Update site points to the nightly repo

Proposed fixes:

* Point to mars
* Point to release update site
* Add nightly site to repos but as an inactive site.

wimjongman, 2016-01-24:
New update site required:

http://www.baeyens.it/eclipse/V3

jantje, 2016-01-24:
I would keep on using V2.

wimjongman, 2016-01-24:
Let's not take shortcuts and make a v3

Xcss 4 brvty mbl

wimjongman, 2016-01-24:
I am back at my laptop.

I am not sure why you want to keep using V2 but since the software is now
on V3 it is not logical to keep using V2. If you do not want to maintain a
version repo then lets call it release or something like that.

Cheers,

Wim

jantje, 2016-01-24:
ok you win

wimjongman, 2016-01-24:
/V3 or /release ?

TrippyLighting, 2016-01-24:
V3

It's easier to ask for forgiveness than to ask for permission ;-)

lrmoreno007, 2016-01-24:
Too many demands on the work of others :(

jantje, 2016-01-24:
V3

jantje, 2016-01-24:
@wimjongman 
Can you create the V2.4 branch I forgot to create? I'm really bad at git :-(
@TrippyLighting and @lrmoreno007 
Are you aware there is also a developers mailing list? eclipse-arduino-dev@freelists.org

wimjongman, 2016-01-24:
There is already a 2.4 tag. I will make a 2.5

wimjongman, 2016-03-13:
Fixed. See also #389

closed 2016-03-13 by wimjongman