Sloeber/arduino-eclipse-plugin #362: indexer will not start

nicoverduin, 2016-01-29:
I just downloaded the overnight build (28-1) on Windows7. Looks great!!!. However the indexer does not seem to start.

jantje, 2016-01-29:
I experianced this problem as well and I'm looking for a solution.
Open and close the project (not eclipse) to get it running again.
Does not always work....

nicoverduin, 2016-01-29:
I'll check monday. Have this  installation at work :)

jantje, 2016-01-30:
Och you have work? and you use the plugin at work?
coool

nicoverduin, 2016-01-30:
Unfortunately Feb is my last month. And since I have nothing to do.... May as well hobby around.

jantje, 2016-01-30:
a shame indeed. but good for me :-)

nicoverduin, 2016-01-30:
yep :)

nicoverduin, 2016-01-31:
I ran it today on my Macbook running Yosemite with 31-1-2016 version. The index still waits until I close the project and reopen it. Also when I change boards (been switching between UNO and NodeMCU 1.0) I have to close the project the first time. From then on it seems to work every time.

jantje, 2016-01-31:
is that with the very latest version?

nicoverduin, 2016-01-31:
yep

jantje, 2016-01-31:
grrr :-(

nicoverduin, 2016-01-31:
mac64.2016-01-31_03-16-59

jantje, 2016-01-31:
I just had hoped it fixed the issue. I did another fix today with spec.d which cause annoying things but the indexer never hung anymore.
So I don't know anymore what I can do to fix it.

nicoverduin, 2016-01-31:
I'm gonna reload it in a couple of minutes

jantje, 2016-01-31:
It may be better to wait for the fix for tomorrow so the spec.d is also fixed.

wimjongman, 2016-01-31:
Is there a 'copy build id to clipboard' command in the quick access box?

Sent from TypeApp

nicoverduin, 2016-01-31:
Well
I already did a clean installation with a new workspace. Made a simple blink program. Had to close the program to start the indexer. Then I installed the ESP8266 platform. Switched to the ESP8266. Indexer started working straight a way. So it seems to only begin at the creation of the program.

nicoverduin, 2016-01-31:
I'll give it a shot tomorrow at work under Windows7 on my "hooker"  machine (DELL :))

nicoverduin, 2016-02-01:
Tested the latest version today. Unfortunately no difference. I still have to close a project first to get the indexer working. Switching between platforms makes no difference.

jantje, 2016-02-01:
@nicoverduin 
Do I understand correctly that it is better? I assume you have upgraded to the latest version so the spec.d file issue is gone?

@wimjongman 
Sorry I do not understand what you are asking.

jantje, 2016-02-01:
I just upgraded to the latest version and I can't reproduce the indexer hanging problem here.
I'm running on linux.

jantje, 2016-02-04:
I have been running with the nightly and this problem has disappeared for me.

nicoverduin, 2016-02-04:
Cool
I'll give it a shot tormorrow
What was the cause apparantly?
thx

jantje, 2016-02-04:
Well nothing has changed lately.
The only thing I changed is that instead of directly adding libraries I add libraries with a job.
If you test and it is still not working. Can you try to disable the "automatically import libraries based on includes" in windows->preferences->arduino?

nicoverduin, 2016-02-04:
will do!

nicoverduin, 2016-02-05:
Ok did the following:

- fresh installation with new workspace
- generated empty INO indexer **OK**
- renamed to .cpp indexer **NOK**
- close project/open project indexer **OK**
- Change to different platform indexer **OK**
- Automatic import OFF **No change in effects**

jantje, 2016-02-05:
when you renamed to .cpp did you also delete the .ino.cpp file?

nicoverduin, 2016-02-05:
I just prest F2 on the filename and renamed it to .cpp. Nothing deleted though.
Normally I always develop in .cpp straight away and for the customer I copy a version to a sepate folder and rename it to .ino.
In this test I started with a .ino (meaning it was a mistake) which gave me a reason to test a little more extensive.
As for priorities, Jan I can certainly live with the workaround. It is a one time effort to close and reopen the project. On the other hand I am a great disapprover of sympton solotions without knowing the cause. But if time is a limiting factor I would suggest we focus on other possibly more urgent issues. 
I am already addicted to this solution and with all the add-ons.... I think it's great.

jantje, 2016-02-05:
>On the other hand I am a great disapprover of sympton solotions without knowing the cause. 

I am to. So I'd like to understand what is going on. I wanted a mac virtual machine for a long time. I guess it is time to set one up. Unless the .ino to .cpp rename helps me to reproduce the issue.

As to ino files. In the latest version there is no disadvantage anymore for using .ino files in the plugin.
I fixed nearly all remaining issue so there is no real other thing to focus on right now. Apart from release :-)

nicoverduin, 2016-02-06:
Jantje
The test was at work=windows

Verstuurd vanaf mijn iPhone

jantje, 2016-02-06:
I confirm that renaming thje last .ino file to .cpp will make the indexer hang.

jantje, 2016-02-07:
@nicoverduin 
Can you check the issue still exists in the latest version?

nicoverduin, 2016-02-08:
Seems to work now:
a) fresh installation
b) created .ino for UNO Indexer **OK**
c) neamed to .cpp Indexer **OK**
d) loaded ESP8266 platforms
e) Changed to ESP8266 board Indexer **OK**
f) renamed .cpp to .ino Indexer **OK**

jantje, 2016-02-08:
:-)

closed 2016-04-01 by jantje