Sloeber/arduino-eclipse-plugin #365: Add a savefunctionality to the scope

jantje, 2016-02-01:
The scope currently  does not have the possibility to dump the data for processing afterwards.
Having this possibility would be a advantage.

jantje, 2016-02-01:
Using doubleclick on the scope could fire this.
@wimjongman 
Is there already some kind of functionality?

wimjongman, 2016-02-01:
The scope currently does not save any state. It only knows the points currently on screen.

The view could have a button that puts the scope data in a file or something ..

jantje, 2016-02-01:
@wimjongman 
Ok. I need this so I'm working on it.
The idea is to double click select a file and save all the points on screen.
Filtering can be done later.

wimjongman, 2016-02-02:
Sounds like a nice feature for the new release.

jantje, 2016-02-02:
It is handy for post processing of data

closed 2016-04-01 by jantje