Sloeber/arduino-eclipse-plugin #366: Problems with datatypes and Arduino sytax

franman11, 2016-02-02:
Hi!
I downloaded and unpacked the nightly build for 3.0.0 on 1st Feb 2016 (yesterday).

For some weird reason, The IDE cannot recognise datatypes such as uint8_t or uint16_t. In addition to this, my code makes use of Timer interrupts and the editor labels it as a syntax error when I build the project.

I included <Arduino.h> but upon opening it, I found out the code is unable to find the libraries stdint.h etc.
I've given the path for Arduino to Arduino IDE 1.6.7 and have also tried passing 1.6.5 but the problem persists.
I don't have any such issues when using the latest 2.x build.

Please help?

jantje, 2016-02-02:
I guess the auto configuration did not work or you interrupted it.
In V 3.0 you should not reference a Arduino IDE
can you please check the error log for errors in regards to auto configuration?
Also check the arduinoplugin folder content in your eclipse installation
if all seems right 
1 stop eclipse
2 delete the arduinoplugin folder
3 start eclipse
4 wait for the configuration to finish (may take 20minutes)
5 press news ketch button
6 verify
7 upload

franman11, 2016-02-02:
Hi,

Thank you for your quick reply!

I tried doing that, but it still did not work. Also, I forgot to mention I'm using Ubuntu 14.04 and not Windows.

jantje, 2016-02-04:
any feedback?

franman11, 2016-02-08:
No luck.

I am currently using the old v2.4 to continue with my work. v3.0 is still unable to recognise the data types as well as the arduino syntax. The initial set up did not have any errors, except one: it could not find the http://eclipse.baeyens.it/V3/ repository

I tried downloading the package again and starting all over again, but it didn't understand. Any v2.x version I download works smoothly, though.

jantje, 2016-02-08:
That is really bad news as V3.0 is supposed to be so much easier.
Would it be possible to create a video on the issue you are experiencing? In my experience this is the best way to solve problems I can not reproduce.

FYI the V3 repository issue is "normal" as V3 is not yet released and as such the V3 repository is empty.

jantje, 2016-03-09:
@franman11 
What do you suggest we do with this issue?

franman11, 2016-03-11:
@jantje 
I'm so sorry I didn't respond to that. I got busy with another project and I just recently got a new machine. V3.0 seems to work fine on that, after I installed linux.

So I'm guessing the problem must be with how the Arduino IDE was set up. That... and could be that I included "Arduino.h" in the new projects just in case.

Seems to be working fine now...

jantje, 2016-03-11:
Don't worry. Be happy. It is a busy world and we all get sucked up from time to time
I'll close this issue

closed 2016-03-11 by jantje