Sloeber/arduino-eclipse-plugin #367: Creating a sketch from an example remembers the previous example choice

wimjongman, 2016-02-07:
Steps to reproduce:

* Install the nightly from 7 February (it may show with other versions)
* Create a new sketch from the Blink example
* Start the new sketch wizard and select another example, note that Blink is still checked
* Uncheck Blink
* Press finish

Note that Blink.ino is available in this sketch along with the other selected example.

jantje, 2016-02-08:
"Creating a sketch from an example remembers the previous example choice"
is what I want as functionality.

However I noticed some unwanted behaviour (with the blink example) where multiple examples (with the same last folder name) are selected and not expanded.
To fix this the full path should be taken into account (and not only the last segment)
I think adding a list of selected examples (or at least the total number of selected examples) and a "unselect all" should be added.

wimjongman, 2016-02-08:
I think I explained it wrong. It is great that you can select multiple examples sketches for one project, but it should not remember these selections for the next project.

jantje, 2016-02-08:
>it should not remember these selections for the next project.

I think it should because all other settings are remembered as well.
What are your arguments to not remember?

wimjongman, 2016-02-09:
IMHO only sensible settings are remembered. For example you also do not remember the project name. 

Maybe the question should be reversed, why do you assume that I also want to have the previously selected sketches in my next project. 

For "example" sketches, like Blink, it does not make sense but maybe for other things it does. I just made an observation. It is no big deal.

jantje, 2016-02-09:
>IMHO only sensible settings are remembered. For example you also do not remember the project name.

I don't fully agree on this one because you can not have 2 project with the same project name but you can have 2 projects based on the same example.

>Maybe the question should be reversed, why do you assume that I also want to have the previously selected sketches in my next project.

A good argument. For testing it is handy as it allows to just provide a project name and click finish and test compilation.

I think if it will be more visible that the old example is still selected we would not have this discussion. I want to keep this issue as there are some issues with the current implementation
The once I know of are:
1) It is not visible enough that the previous example is still selected.
2) Multiple examples with the same name exist where the current code assumes this is not the case.

wimjongman, 2016-02-09:
Okay. Clear!

Sent from TypeApp

jantje, 2016-02-12:
I forgot to mention I fixed the issue that all blink examples are selected when a blink example was selected the last time.

jantje, 2016-02-22:
@wimjongman 
What do you think of the current look and feel?
Can you live with this?

wimjongman, 2016-02-22:
I saw it briefly in the video. It looks great now. I will download the
latest version tonight and play with it.

jantje, 2016-02-22:
:-)

jantje, 2016-03-09:
Can we close this one?

closed 2016-03-25 by jantje