Sloeber/arduino-eclipse-plugin #368: Update to nightly build stop showing plugin integration.

far5893, 2016-02-09:
Update to nightly build stop showing plugin integration.
No error visble.

  Arduino Eclipse Extensions	3.0.0.201602070209	it.baeyens.arduino.feature.feature.group	Jan Baeyens
    C/C++ Development Tools	8.8.0.201509131935	org.eclipse.cdt.feature.group	Eclipse CDT
      C/C++ Common GDB Support	8.8.0.201509131935	org.eclipse.cdt.gdb.feature.group	Eclipse CDT
      C/C++ Development Platform	8.8.0.201509131935	org.eclipse.cdt.platform.feature.group	Eclipse CDT
      C/C++ DSF GDB Debugger Integration	8.8.0.201509131935	org.eclipse.cdt.gnu.dsf.feature.group	Eclipse CDT
      C/C++ GNU Toolchain Build Support	8.8.0.201509131935	org.eclipse.cdt.gnu.build.feature.group	Eclipse CDT
      C/C++ GNU Toolchain Debug Support	8.8.0.201509131935	org.eclipse.cdt.gnu.debug.feature.group	Eclipse CDT
  Eclipse CVS Client	1.4.200.v20150904-0015	org.eclipse.cvs.feature.group	Eclipse.org
  Eclipse CVS Client Resources	1.4.200.v20150904-0015	org.eclipse.cvs.source.feature.group	Eclipse.org
  Eclipse IDE for C/C++ Developers	4.5.1.20150917-1200	epp.package.cpp	null
    EPP CPP Feature	4.5.1.20150917-1200	org.eclipse.epp.package.cpp.feature.feature.group	Eclipse Packaging Project
  Mylyn Versions Connector: CVS	1.9.0.v20150908-1643	org.eclipse.mylyn.cvs.feature.group	Eclipse Mylyn
    Mylyn Versions	1.9.0.v20150908-1643	org.eclipse.mylyn.versions.feature.group	Eclipse Mylyn

far5893, 2016-02-09:
!SESSION 2016-02-09 12:25:33.111 -----------------------------------------------
eclipse.buildId=4.5.1.M20150904-0015
java.version=1.7.0_95
java.vendor=Oracle Corporation
BootLoader constants: OS=linux, ARCH=x86_64, WS=gtk, NL=it_IT
Framework arguments:  -product org.eclipse.epp.package.cpp.product
Command-line arguments:  -os linux -ws gtk -arch x86_64 -product org.eclipse.epp.package.cpp.product

!ENTRY it.baeyens.arduino.common 4 0 2016-02-09 12:25:35.586
!MESSAGE FrameworkEvent ERROR
!STACK 0
org.osgi.framework.BundleException: Could not resolve module: it.baeyens.arduino.common [806]
  Unresolved requirement: Require-Capability: osgi.ee; filter:="(&(osgi.ee=JavaSE)(version=1.8))"

	at org.eclipse.osgi.container.Module.start(Module.java:434)
	at org.eclipse.osgi.container.ModuleContainer$ContainerStartLevel.incStartLevel(ModuleContainer.java:1582)
	at org.eclipse.osgi.container.ModuleContainer$ContainerStartLevel.incStartLevel(ModuleContainer.java:1561)
	at org.eclipse.osgi.container.ModuleContainer$ContainerStartLevel.doContainerStartLevel(ModuleContainer.java:1533)
	at org.eclipse.osgi.container.ModuleContainer$ContainerStartLevel.dispatchEvent(ModuleContainer.java:1476)
	at org.eclipse.osgi.container.ModuleContainer$ContainerStartLevel.dispatchEvent(ModuleContainer.java:1)
	at org.eclipse.osgi.framework.eventmgr.EventManager.dispatchEvent(EventManager.java:230)
	at org.eclipse.osgi.framework.eventmgr.EventManager$EventThread.run(EventManager.java:340)

!ENTRY it.baeyens.arduino.core 4 0 2016-02-09 12:25:35.591
!MESSAGE FrameworkEvent ERROR
!STACK 0
org.osgi.framework.BundleException: Could not resolve module: it.baeyens.arduino.core [807]
  Unresolved requirement: Import-Package: it.baeyens.arduino.arduino
    -> Export-Package: it.baeyens.arduino.arduino; bundle-version="3.0.0.201602070209"; bundle-symbolic-name="it.baeyens.arduino.common"; version="0.0.0"
       it.baeyens.arduino.common [806]
         Unresolved requirement: Require-Capability: osgi.ee; filter:="(&(osgi.ee=JavaSE)(version=1.8))"

	at org.eclipse.osgi.container.Module.start(Module.java:434)
	at org.eclipse.osgi.container.ModuleContainer$ContainerStartLevel.incStartLevel(ModuleContainer.java:1582)
	at org.eclipse.osgi.container.ModuleContainer$ContainerStartLevel.incStartLevel(ModuleContainer.java:1561)
	at org.eclipse.osgi.container.ModuleContainer$ContainerStartLevel.doContainerStartLevel(ModuleContainer.java:1533)
	at org.eclipse.osgi.container.ModuleContainer$ContainerStartLevel.dispatchEvent(ModuleContainer.java:1476)
	at org.eclipse.osgi.container.ModuleContainer$ContainerStartLevel.dispatchEvent(ModuleContainer.java:1)
	at org.eclipse.osgi.framework.eventmgr.EventManager.dispatchEvent(EventManager.java:230)
	at org.eclipse.osgi.framework.eventmgr.EventManager$EventThread.run(EventManager.java:340)

!ENTRY it.baeyens.arduino.monitor 4 0 2016-02-09 12:25:35.596
!MESSAGE FrameworkEvent ERROR
!STACK 0
org.osgi.framework.BundleException: Could not resolve module: it.baeyens.arduino.monitor [809]
  Unresolved requirement: Require-Bundle: it.baeyens.arduino.common
    -> Bundle-SymbolicName: it.baeyens.arduino.common; bundle-version="3.0.0.201602070209"; singleton:="true"
       it.baeyens.arduino.common [806]
         Unresolved requirement: Require-Capability: osgi.ee; filter:="(&(osgi.ee=JavaSE)(version=1.8))"

	at org.eclipse.osgi.container.Module.start(Module.java:434)
	at org.eclipse.osgi.container.ModuleContainer$ContainerStartLevel.incStartLevel(ModuleContainer.java:1582)
	at org.eclipse.osgi.container.ModuleContainer$ContainerStartLevel.incStartLevel(ModuleContainer.java:1561)
	at org.eclipse.osgi.container.ModuleContainer$ContainerStartLevel.doContainerStartLevel(ModuleContainer.java:1533)
	at org.eclipse.osgi.container.ModuleContainer$ContainerStartLevel.dispatchEvent(ModuleContainer.java:1476)
	at org.eclipse.osgi.container.ModuleContainer$ContainerStartLevel.dispatchEvent(ModuleContainer.java:1)
	at org.eclipse.osgi.framework.eventmgr.EventManager.dispatchEvent(EventManager.java:230)
	at org.eclipse.osgi.framework.eventmgr.EventManager$EventThread.run(EventManager.java:340)

!ENTRY it.baeyens.cdt.refactor 4 0 2016-02-09 12:25:35.600
!MESSAGE FrameworkEvent ERROR
!STACK 0
org.osgi.framework.BundleException: Could not resolve module: it.baeyens.cdt.refactor [810]
  Unresolved requirement: Require-Capability: osgi.ee; filter:="(&(osgi.ee=JavaSE)(version=1.8))"

	at org.eclipse.osgi.container.Module.start(Module.java:434)
	at org.eclipse.osgi.container.ModuleContainer$ContainerStartLevel.incStartLevel(ModuleContainer.java:1582)
	at org.eclipse.osgi.container.ModuleContainer$ContainerStartLevel.incStartLevel(ModuleContainer.java:1561)
	at org.eclipse.osgi.container.ModuleContainer$ContainerStartLevel.doContainerStartLevel(ModuleContainer.java:1533)
	at org.eclipse.osgi.container.ModuleContainer$ContainerStartLevel.dispatchEvent(ModuleContainer.java:1476)
[... 773 lines cut ...]
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.safeCreateGui(PartRenderingEngine.java:654)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.safeCreateGui(PartRenderingEngine.java:746)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.access$0(PartRenderingEngine.java:717)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine$2.run(PartRenderingEngine.java:711)
	at org.eclipse.core.runtime.SafeRunner.run(SafeRunner.java:42)
[... cut ...]
	at org.eclipse.e4.ui.internal.workbench.E4Workbench.createAndRunUI(E4Workbench.java:156)
	at org.eclipse.ui.internal.Workbench$5.run(Workbench.java:654)
	at org.eclipse.core.databinding.observable.Realm.runWithDefault(Realm.java:337)
	at org.eclipse.ui.internal.Workbench.createAndRunWorkbench(Workbench.java:598)
	at org.eclipse.ui.PlatformUI.createAndRunWorkbench(PlatformUI.java:150)
	at org.eclipse.ui.internal.ide.application.IDEApplication.start(IDEApplication.java:139)
	at org.eclipse.equinox.internal.app.EclipseAppHandle.run(EclipseAppHandle.java:196)
	at org.eclipse.

wimjongman, 2016-02-09:
We need java 8. Please install and try again.

On Feb 9, 2016, 13:00, at 13:00, far5893 <notifications@github.com> wrote:
> !SESSION 2016-02-09 12:25:33.111
> -----------------------------------------------
> eclipse.buildId=4.5.1.M20150904-0015
> java.version=1.7.0_95
> java.vendor=Oracle Corporation
> BootLoader constants: [...]

far5893, 2016-02-09:
Ok with java 8 work!!

closed 2016-02-09 by jantje