Sloeber/arduino-eclipse-plugin #369: Update to nightly build breaks Teensyduino compatibility

universpharmacie, 2016-02-09:
Just did a routine update with the nightly build of Eclipse Arduino and my project will not longer compile. Then I saw in the project properties that I could not longer select my Teensy 3.2 board in the list. Up to now, the boards list seemed to be linked to the Arduino IDE with the installed Teensyduino add on. That link seems to be broken --- Help!!!

universpharmacie, 2016-02-09:
Aaaargh! And "Help" - "Installation details" won't let me revert to 2.5: "Problem encountered..." 

Now everything is blocked, I can't continue working on my project.

universpharmacie, 2016-02-09:
fyi: Arduino 1.6.5 and Teensyduino 1.2.6 haven't been modified/updated for at least 3 months since I was always working with EclipseArduino 2.5 nightly without problems. How comes that the 3.0 update obviously messes up the hardware path and project settings compatibility in a way that I even can't restore or find these on my Mac?

wimjongman, 2016-02-09:
Welcome to the bleeding edge. 

In help/about/installation details there is a tab installation history. 

Select the previous installation and press revert.

universpharmacie, 2016-02-09:
Reverting is what I tried. Does not work. I get the following error message : 
```
An error occurred while collecting items to be installed
session context was:(profile=DefaultProfile, phase=org.eclipse.equinox.internal.p2.engine.phases.Collect, operand=, action=).
No repository found containing: org.eclipse.update.feature,it.baeyens.arduino.feature,2.5.0.201512120209
```
Nothing works! It's driving me crazy :-(

universpharmacie, 2016-02-09:
Ok, I'm starting to understand what goes wrong (but I hate losing my time with that) : 
On the Mac, the Teensy hardware information (installed by Teensyduino) is buried in the /Applications/Arduino.app/Contents/Java/hardware folder which can not be selected from the new preferences pane where instead of the previous Arduino app path, now in the new version the path towards the hardware information goes deeper, but since the .app folders on the Mac are special folders, one can not browse inside them from the Eclipse Arduino IDE. Thus, no way to configure the Teensy for the moment. Please fix!

jantje, 2016-02-09:
@universpharmacie 
My advice to you:
First: relax V3 is majorly different from V2 so you have to look at things in a completely different way.
Secondly: blame yourself for not taking a backup before upgrading.
Thirdly: forgive yourself and bite the bullet to go to V3.

Steps I advice to you to go to V3:
There is no need to reference a arduino ide anymore. So whatever is installed on your system is by default not important anymore. (It is a bit different for teensy because Paul does not support the boardsmanager but I'll get to that later)
What I advice you to do is as follows:
Separate the "upgrade a workspace problem" from "working with a upgraded product".
Step 1: Start the arduino eclipseplugin in a new workspace. 
Step 2: Make a new sketch from one of the available boards and compile and see if all is well. 
Achievement award: Working with V3

Step 3: Configure Teensy as explained here: https://www.patreon.com/posts/4192266
Step 4: Make a new sketch based on a teensy board; verify and upload

Achievement award: configuring Teensy with V3

Now it is time to upgrade your existing workspace.
Step 5: backup your production workspace
Step 6: switch to the production workspace
Step 7: for each project create a new empty project (select sample project but do not select a example)
Step 7.1: drag the code from the old project to the new project
Step 7.2: Wait for the indexer and auto tools to finish
Step 7.3: compile and upload
If this is working 
Achievement award: Upgraded to V3

universpharmacie, 2016-02-09:
Step 3 can't be done in that way as described on OSX. When I want to add private hardware path, I get a directory selector dialog bow which doesn't allow me to go to the .../hardware/teensy path, since already the /Applications/Arduino.app folder is greyed out. Thus, problem persists.

jantje, 2016-02-09:
Congratulations with your first achievement award :-)))
The problem you have with step 3 is not a bug but a "mac feature". I remember having read how to unpack/unzip/extract the .app folder but I can't find it back. It is a simple action commonly known in mac world

universpharmacie, 2016-02-09:
I don't think that it's a good moment for your irony, jantje.
I know how to access the content of an .app folder on Mac: In the Finder, right click on the App icon and select "Show package content" form the context menu. But the file/directory dialog box in the Eclipse Arduino plugin which opens when one wants to add a private hardware path in the preferences does not have that context option, thus, it is impossible to access subdirectories within an .app folder. That is the real BUG in your plugin. Did you ever test it on a Mac?

universpharmacie, 2016-02-09:
Here is the dialog box which opens when I go to Preferences - Arduino - Private Hardware Path - New...
When I arrive on the Applications directory, the app icons are all greyed out and I can not go down to the Arduino.app folder and lower. I'm not a Java developer, but I think that it has to do with some parameters which you use to call that dialog box.
[image]

jantje, 2016-02-09:
>I don't think that it's a good moment for your irony, jantje.

There was no irony intended.

>Did you ever test it on a Mac?

No I didn't. This is the nightly so this is not yet tested. 
As for helping out, I do not own a mac nor any mac vm so I can only speak from what I have heard.

Back to the issue:
If the "show package content" is a temporary action (I thought it was some kind of unzip) and the dialog box does not have this feature; you are right. This is a bug.

jantje, 2016-02-09:
just found this in a simular discussion

The file open dialogs on Mac don't let you access the package contents
as you can do in Finder.
As a workaround, you can Alt-Cmd-Drag any folder (e.g. Fiji.app/jars)
out of the .app package into a normal, accessible folder on your disk.
This will create a symlink/alias that is accessible also via the file
chooser dialogs. 

can this be a workarund?

jantje, 2016-02-09:
As it is the PathEditor that does not support the "show package content" I created a bug in eclipse. Please vote if you'd like this issue fixed.
https://bugs.eclipse.org/bugs/show_bug.cgi?id=487534

universpharmacie, 2016-02-09:
Good news:
The symlink trick works and allows the selection of the symlinked teensy directory as private hardware directory. As soon as it has been selected in the Eclipse - Arduino preferences, it is correctly resolved into "/Applications/Arduino.app/Contents/Java/hardware/teensy/" and the symlink may even be removed.
Then, I had to resolve the path problems in platform.txt which was a little bit more trickier. If the teensy directory is selected as above, the modified paths as mentioned on Patreon won't work since these start from the idea that one had selected the parent "hardware" directory. So, the following entries made it working:
`compiler.path={runtime.hardware.path}/../../`
and 
`tools.teensyloader.cmd.path={runtime.hardware.path}/../tools/teensy_post_compile`

But that did still not allow the teensy loader to come up, that needed still the following modification :
`tools.teensyloader.upload.pattern="{cmd.path}" "-file={build.project_name}" "-path={build.path}" "-tools=**{runtime.hardware.path}/../tools**" "-board={build.board}" -reboot`

universpharmacie, 2016-02-09:
Now I have still to check out how I integrate TyQt as an alternative to the Teensy loader, but that is a different issue and not blocking.
The most important is that I can continue working on my project. Thank you everybody for the help and the encouragement!

jantje, 2016-02-09:
Cool. You got it going

> won't work since these start from the idea that one had selected the parent "hardware" directory. So,

This is not how the code works. The hardware directory is calculated based on the platform.txt location.
So I reopen this issue as the symlink trick is a workaround and the changes yu needed to do is not compliant with the changes Paul may do in Teensyduino 1.2.8

universpharmacie, 2016-02-09:
Good to know that it should not work like this... ;-) But it actually works like this. I found it out by looking at the generated wrong paths in the "Problems" console window.

jantje, 2016-02-09:
I do not have this issue in windows and linux :-)
There must be something somewhere in the code. I'll have a look after I finished my first toddler movie
Can you vote for this issue (you only need an account it doesn't cost money ;-) ) https://bugs.eclipse.org/bugs/show_bug.cgi?id=487534

mjmeijer, 2016-12-29:
Hi @universpharmacie,

I noticed that Teensyduino is updated to support version 1.8.0 of the Arduino IDE. Does your problem still occur now that files are no longer placed inside the Arduino.App, but in /Library/Arduino15 and /Documents/Arduino/hardware and /Documents/Arduino/libraries ?

universpharmacie, 2016-12-29:
Hi, thank you for asking. I've still to check that out. Did not yet upgrade to Arduino 1.8 because of "never touch a running system"... But I'll make the step next weekend and report back then.

universpharmacie, 2016-12-29:
OK, trashed everything and did a fresh install of Arduino 1.8 and Teensyduino 1.34. I finally discovered that the Teensyduino files install still inside the Arduino.app package.

But I discovered at this occasion that sloeber's path pick now allows to go into app folders. (See attachment)
[image]

mjmeijer, 2016-12-29:
OK, thank you for trying this. I guess that this issue#369 can now be closed. For the record, you are able to install Teensyduino 1.34 form the Arduino.App 1.8.0 folder using EclipseArduino v3.1, right?

universpharmacie, 2016-12-29:
It might most probably been closed. Not sure because I'm not using EclipseArduino 3.1 but v4 nightly.

jantje, 2016-12-30:
so the latest version of eclipse does have a way to show the package content :-) 
Wauw :-)
@mjmeijer 
I only close issues when they are delivered as part of a stable version. This might change after release 4 ..

mjmeijer, 2016-12-30:
@universpharmacie You use v4 nightly, can you reproduce issue #615 ? 
I cannot get the V4 nightly to work on my Mac, I use Sierra 10.12.2, what is your OS?

universpharmacie, 2016-12-30:
Sierra 10.12.3 beta. Will comment in the issue #615 thread

closed 2017-01-17 by jantje