Sloeber/arduino-eclipse-plugin #370: Arduino 101 fails upload

scottemberley, 2016-02-10:
Uploading the blink sample fails:
```
       Starting upload
       using arduino loader
       Starting reset using 1200bps touch process
       Continuing to use COM12
       Ending reset

       Launching /x86/bin/bash.exe --noprofile    COM12  
       Output:
       failed to upload 
```
It is the correct port and baud rate
Using Arduino 1.6.7 I am able to upload the sample with the same port, cable & system.

Note: the UI is looking great and I cant wait to get started pushing C++ code to the device.

jantje, 2016-02-10:
I don't have a Arduino 101 so I can't test it over here.
So please put on verbose upload in arduino ide and post the console output here.

scottemberley, 2016-02-10:
This is both a verbose compile and upload with Arduino IDE 1.6.7

 

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware "C:\Program Files (x86)\Arduino\hardware" -hardware "C:\Users\Scott\AppData\Local\Arduino15\packages" -tools "C:\Program Files (x86)\Arduino\tools-builder" -tools "C:\Program Files (x86)\Arduino\hardware\tools\avr" -tools "C:\Users\Scott\AppData\Local\Arduino15\packages" -built-in-libraries "C:\Program Files (x86)\Arduino\libraries" -libraries "C:\Users\Scott\Documents\Arduino\libraries" -fqbn=Intel:arc32:arduino_101 -vid-pid=0X8087_0X0AB6 -ide-version=10607 -build-path "C:\Users\Scott\AppData\Local\Temp\build9c9ef3bdfe2fccb480bc6e4bac749e41.tmp" -warnings=none -prefs=build.warn_data_percentage=75 -verbose "C:\Program Files (x86)\Arduino\examples\01.Basics\Blink\Blink.ino"

C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware "C:\Program Files (x86)\Arduino\hardware" -hardware "C:\Users\Scott\AppData\Local\Arduino15\packages" -tools "C:\Program Files (x86)\Arduino\tools-builder" -tools "C:\Program Files (x86)\Arduino\hardware\tools\avr" -tools "C:\Users\Scott\AppData\Local\Arduino15\packages" -built-in-libraries "C:\Program Files (x86)\Arduino\libraries" -libraries "C:\Users\Scott\Documents\Arduino\libraries" -fqbn=Intel:arc32:arduino_101 -vid-pid=0X8087_0X0AB6 -ide-version=10607 -build-path "C:\Users\Scott\AppData\Local\Temp\build9c9ef3bdfe2fccb480bc6e4bac749e41.tmp" -warnings=none -prefs=build.warn_data_percentage=75 -verbose "C:\Program Files (x86)\Arduino\examples\01.Basics\Blink\Blink.ino"

Board Intel:i586:izmir_fg doesn't define a 'build.board' preference. Auto-set to: I586_IZMIR_FG

Board Intel:i586:izmir_fd doesn't define a 'build.board' preference. Auto-set to: I586_IZMIR_FD

Board Intel:i686:izmir_ec doesn't define a 'build.board' preference. Auto-set to: I686_IZMIR_EC

"C:\Users\Scott\AppData\Local\Arduino15\packages\Intel\tools\arc-elf32\1.6.4+1.0/bin/arc-elf32-g++"  -c -mARCv2EM -mav2em -mlittle-endian -g -Os -Wall -fno-reorder-functions -fno-asynchronous-unwind-tables -fno-omit-frame-pointer -fno-defer-pop -Wno-unused-but-set-variable -Wno-main -ffreestanding -fno-stack-protector -mno-sdata -ffunction-sections -fdata-sections -fsigned-char  -fno-rtti -fno-exceptions -D__ARDUINO_ARC__ -std=c++11 -w -x c++ -E -CC -DF_CPU=32000000L -DARDUINO=10607 -DARDUINO_ARC32_TOOLS -DARDUINO_ARCH_ARC32 -D__CPU_ARC__ -DCLOCK_SPEED=32 -DCONFIG_SOC_GPIO_32 -DCONFIG_SOC_GPIO_AON -DINFRA_MULTI_CPU_SUPPORT -DCFW_MULTI_CPU_SUPPORT -DHAS_SHARED_MEM "-IC:\Users\Scott\AppData\Local\Arduino15\packages\Intel\hardware\arc32\1.0.4\system/libarc32_arduino101/common" "-IC:\Users\Scott\AppData\Local\Arduino15\packages\Intel\hardware\arc32\1.0.4\system/libarc32_arduino101/drivers" "-IC:\Users\Scott\AppData\Local\Arduino15\packages\Intel\hardware\arc32\1.0.4\system/libarc32_arduino101/bootcode" "-IC:\Users\Scott\AppData\Local\Arduino15\packages\Intel\hardware\arc32\1.0.4\system/libarc32_arduino101/framework/include"        "-IC:\Users\Scott\AppData\Local\Arduino15\packages\Intel\hardware\arc32\1.0.4\cores\arduino" "-IC:\Users\Scott\AppData\Local\Arduino15\packages\Intel\hardware\arc32\1.0.4\variants\arduino_101" "C:\Users\Scott\AppData\Local\Temp\build9c9ef3bdfe2fccb480bc6e4bac749e41.tmp\sketch\Blink.ino.cpp" -o "nul"

"C:\Users\Scott\AppData\Local\Arduino15\packages\Intel\tools\arc-elf32\1.6.4+1.0/bin/arc-elf32-g++"  -c -mARCv2EM -mav2em -mlittle-endian -g -Os -Wall -fno-reorder-functions -fno-asynchronous-unwind-tables -fno-omit-frame-pointer -fno-defer-pop -Wno-unused-but-set-variable -Wno-main -ffreestanding -fno-stack-protector -mno-sdata -ffunction-sections -fdata-sections -fsigned-char  -fno-rtti -fno-exceptions -D__ARDUINO_ARC__ -std=c++11 -w -x c++ -E -CC -DF_CPU=32000000L -DARDUINO=10607 -DARDUINO_ARC32_TOOLS -DARDUINO_ARCH_ARC32 -D__CPU_ARC__ -DCLOCK_SPEED=32 -DCONFIG_SOC_GPIO_32 -DCONFIG_SOC_GPIO_AON -DINFRA_MULTI_CPU_SUPPORT -DCFW_MULTI_CPU_SUPPORT -DHAS_SHARED_MEM "-IC:\Users\Scott\AppData\Local\Arduino15\packages\Intel\hardware\arc32\1.0.4\system/libarc32_arduino101/common" "-IC:\Users\Scott\AppData\Local\Arduino15\packages\Intel\hardware\arc32\1.0.4\system/libarc32_arduino101/drivers" "-IC:\Users\Scott\AppData\Local\Arduino15\packages\Intel\hardware\arc32\1.0.4\system/libarc32_arduino101/bootcode" "-IC:\Users\Scott\AppData\Local\Arduino15\packages\Intel\hardware\arc32\1.0.4\system/libarc32_arduino101/framework/include"        "-IC:\Users\Scott\AppData\Local\Arduino15\packages\Intel\hardware\arc32\1.0.4\cores\arduino" "-IC:\Users\Scott\AppData\Local\Arduino15\packages\Intel\hardware\arc32\1.0.4\variants\arduino_101" "C:\Users\Scott\AppData\Local\Temp\build9c9ef3bdfe2fccb480bc6e4bac749e41.tmp\sketch\Blink.ino.cpp" -o "nul"

"C:\Users\Scott\AppData\Local\Arduino15\packages\Intel\tools\arc-elf32\1.6.4+1.0/bin/arc-elf32-g++"  -c -mARCv2EM -mav2em -mlittle-endian -g -Os -Wall -fno-reorder-functions -fno-asynchronous-unwind-tables -fno-omit-frame-pointer -fno-defer-pop -Wno-unused-but-set-variable -Wno-main -ffreestanding -fno-stack-protector -mno-sdata -ffunction-sections -fdata-sections -fsigned-char  -fno-rtti -fno-exceptions -D__ARDUINO_ARC__ -std=c++11 -w -x c++ -E -CC -DF_CPU=32000000L -DARDUINO=10607 -DARDUINO_ARC32_TOOLS -DARDUINO_ARCH_ARC32 -D__CPU_ARC__ -DCLOCK_SPEED=32 -DCONFIG_SOC_GPIO_32 -DCONFIG_SOC_GPIO_AON -DINFRA_MULTI_CPU_SUPPORT -DCFW_MULTI_CPU_SUPPORT -DHAS_SHARED_MEM "-IC:\Users\Scott\AppData\Local\Arduino15 [... cut ...]
Download           [=====================    ]  85%        16384 bytesInvalid DFU suffix signature

A valid DFU suffix will be required in a future dfu-util release!!!

can't detach

 

Download           [=======================  ]  95%        17188 bytes

Download           [=========================] 100%        17188 bytes

Download done.

Sent a total of 17188 bytes

state(2) = dfuIDLE, status(0) = No error condition is present

Done!

Resetting USB to switch back to runtime mode

Sketch will execute in about 5 seconds.

jantje, 2016-02-10:
Och you are using arduino eclipse plugin 2.4 with Arduino IDE 1.6.7 which is unsupported

scottemberley, 2016-02-10:
I hope your sentence is truncated and what you really meant was “which is unsupported [at this time]”.  

 

In any case I did find it interesting that the Arduino 101 showed up on the device list and gave it a try.  It appeared after selecting the Intel Curie Boards under the ‘Preferences->Arduino->Platforms and Boards’ configuration in the Eclipse Arduino IDE.  The boards.txt file containing the 101 is located at \eclipseArduino\arduinoPlugin\packages\Intel\hardware\arc32\1.0.4 which I thought was included in the nightly build and not something inherited from my existing setup.  Additionally I may have misunderstood your earlier request for a build and upload dump.  That was done directly from the previously installed ‘Arduino IDE’; did you actually want the dump from the ‘Eclipse Arduino IDE’?  If so tell me where to make the verbose setting and I will generate the data from the ‘Eclipse Arduino IDE’ and send that on.

 

Note that I also had the Intel Edison show-up.  Will this have the same issues as the 101?

 

Sorry if this seems like a lot of chatter but I would love to get support for the intel boards into this IDE.  I think it looks great and will make my efforts much more successful. what needs to happen to get support for 1.6.7?  What can I do to help move it forward?

jantje, 2016-02-10:
Your story is inconsistent. I say so because you say you use V3 but the logs show you don't.
This may give you a better understanding about V3
https://www.youtube.com/watch?v=AmePMUuv0uk

Therefore:
I decided to no longer respond to github issues -and other questions- due to a lack of support.
As a patron you can post a message on the patreon forum stating you support a github issue. This allows me to support people who support me.
https://www.patreon.com/posts/3700469

scottemberley, 2016-02-10:
Let me try to clarify.

I downloaded nightly build win64.2016-02-07_03-16-57 of Toddler (V3) and I run the launcher from the location where I uncompressed the files.  I also downloaded win64.2016-02-10_03-17-23 which I will install shortly.  What indicates that I am running another version?  Could another eclipse install (CDT) be interfering with what I am trying to do?

jantje, 2016-02-10:
Thanks for the support :-)
I just tried here and there is a problem with the upload of Arduino 101 but that does not relate to the description above.
V3 downloads and installs its files in a subfolder of the eclipse installation. The subfolder has the name arduinoPlugin.
As the command to upload (and compile) is a fqn and does not contain arduinoPlugin yu are not using the plugin as should.
I guess you added   "C:\Users\Scott\AppData\Local\Arduino15\packages" to your private hardware path. That is a very bad idea.

What you should have done.
In windows->preferences->platforms and boards put a check in the checkbox "intel curie boards 1.0.4" and press ok.
This will trigger the download of the boards and tools.

The upload fails on my system with the following

```
Starting upload
using arduino loader
Starting reset using 1200bps touch process
Continuing to use tt
Ending reset

Launching /x86/bin/bash.exe --noprofile    tt  
Output:
failed to upload 
```
You can see the path is wrong. I'm working on a fix for that.
I also had to remove a check for the arduino version for the code to compile.

scottemberley, 2016-02-10:
I am starting to see our disconnect.

So I had previously installed 1.6.7 from Arduino.cc and that install pushed the hardware files to the appdata directory.  You saw that location due to me sending you a dump from that installation not from the plugin.  I 'assumed' you wanted to see a successful load from arduino.exe to validate everything was functional from that perspective.

Do you still want to see a verbose upload from the plugin?  I expect not until the path issue is resolved.

Would it help if I did a git of the source and throw some assistance at the path issue or would explaining the fix take longer than just fixing it?

jantje, 2016-02-10:
>Would it help if I did a git of the source and throw some assistance at the path issue or would explaining the fix take longer than just fixing it?

I fear the second. But if you are interested start by reading
http://blog.baeyens.it/#post22 and
http://blog.baeyens.it/#post24

scottemberley, 2016-02-10:
I will take a look.

jantje, 2016-02-11:
I got something working. I will check it in so you can upgrade and test tomorrow.

```
Starting upload
using arduino loader
Starting reset using 1200bps touch process
Continuing to use com1
Ending reset

Launching C:\java\eclipse\arduinoPlugin\tools\Intel\sketchUploader\1.6.4+1.14/x86/bin/bash.exe --noprofile C:\java\eclipse\arduinoPlugin\tools\Intel\sketchUploader\1.6.4+1.14/clupload/cluploadArduino101_win.sh C:\java\eclipse\arduinoPlugin\tools\Intel\sketchUploader\1.6.4+1.14/x86/bin C:\Users\jan\Documents\workspaces\runtime-EclipseApplication\jj/Release/jj.elf com1  
Output:
cygwin warning:
  MS-DOS style path detected: C:\java\eclipse\arduinoPlugin\tools\Intel\sketchUploader\1.6.4+1.14/clupload/cluploadArduino101_win.sh
  Preferred POSIX equivalent is: /cygdrive/c/java/eclipse/arduinoPlugin/tools/Intel/sketchUploader/1.6.4+1.14/clupload/cluploadArduino101_win.sh
  CYGWIN environment variable option "nodosfilewarning" turns off this warning.
  Consult the user's guide for more details about POSIX paths:
    http://cygwin.com/cygwin-ug-net/using.html#using-pathnames
starting download script
Args to shell: C:\java\eclipse\arduinoPlugin\tools\Intel\sketchUploader\1.6.4+1.14/x86/bin C:\Users\jan\Documents\workspaces\runtime-EclipseApplication\jj/Release/jj.elf com1
Serial Port PORT
BIN FILE C:\Users\jan\Documents\workspaces\runtime-EclipseApplication\jj/Release/jj.bin
Waiting for device... 
ERROR: Timed out waiting for Arduino 101.
1.6.4+1.14/x86/bin/bash.exe finished
upload done
```

scottemberley, 2016-02-11:
very cool.

As expected I am still building an environment...

scottemberley, 2016-02-11:
I pulled a copy and it looks much better.  There is a compiler error (see below) but I assume its a configuration issue.  I will try to take a look at it tomorrow.

20:46:20 **** Incremental Build of configuration Release for project BlinkTest ****
"C:\\Users\\Scott\\Source\\Repos\\arduino-eclipse-plugin\\it.baeyens.arduino.product\\target\\products\\it.baeyens.arduino.product\\win32\\win32\\x86_64\\eclipseArduino\\arduinoPlugin\\tools\\make\\make" all 
'Building file: ../.ino.cpp'
'Starting C++ compile'
"C:\Users\Scott\Source\Repos\arduino-eclipse-plugin\it.baeyens.arduino.product\target\products\it.baeyens.arduino.product\win32\win32\x86_64\eclipseArduino\arduinoPlugin\tools\Intel\arc-elf32\1.6.4+1.0/bin/arc-elf32-g++" -c -mARCv2EM -mav2em -mlittle-endian -g -Os -Wall -fno-reorder-functions -fno-asynchronous-unwind-tables -fno-omit-frame-pointer -fno-defer-pop -Wno-unused-but-set-variable -Wno-main -ffreestanding -fno-stack-protector -mno-sdata -ffunction-sections -fdata-sections -fsigned-char -MMD -fno-rtti -fno-exceptions -D__ARDUINO_ARC__ -std=c++11 -DF_CPU=32000000L -DARDUINO=10606 -DARDUINO_ARC32_TOOLS -DARDUINO_ARCH_ARC32 -D__CPU_ARC__ -DCLOCK_SPEED=32 -DCONFIG_SOC_GPIO_32 -DCONFIG_SOC_GPIO_AON -DINFRA_MULTI_CPU_SUPPORT -DCFW_MULTI_CPU_SUPPORT -DHAS_SHARED_MEM "-IC:/Users/Scott/Source/Repos/arduino-eclipse-plugin/it.baeyens.arduino.product/target/products/it.baeyens.arduino.product/win32/win32/x86_64/eclipseArduino/arduinoPlugin/packages/Intel/hardware/arc32/1.0.4/system/libarc32_arduino101/common" "-IC:/Users/Scott/Source/Repos/arduino-eclipse-plugin/it.baeyens.arduino.product/target/products/it.baeyens.arduino.product/win32/win32/x86_64/eclipseArduino/arduinoPlugin/packages/Intel/hardware/arc32/1.0.4/system/libarc32_arduino101/drivers" "-IC:/Users/Scott/Source/Repos/arduino-eclipse-plugin/it.baeyens.arduino.product/target/products/it.baeyens.arduino.product/win32/win32/x86_64/eclipseArduino/arduinoPlugin/packages/Intel/hardware/arc32/1.0.4/system/libarc32_arduino101/bootcode" "-IC:/Users/Scott/Source/Repos/arduino-eclipse-plugin/it.baeyens.arduino.product/target/products/it.baeyens.arduino.product/win32/win32/x86_64/eclipseArduino/arduinoPlugin/packages/Intel/hardware/arc32/1.0.4/system/libarc32_arduino101/framework/include"    -I"C:\Users\Scott\Source\Repos\arduino-eclipse-plugin\it.baeyens.arduino.product\target\products\it.baeyens.arduino.product\win32\win32\x86_64\eclipseArduino\arduinoPlugin\packages\Intel\hardware\arc32\1.0.4\cores\arduino" -I"C:\Users\Scott\Source\Repos\arduino-eclipse-plugin\it.baeyens.arduino.product\target\products\it.baeyens.arduino.product\win32\win32\x86_64\eclipseArduino\arduinoPlugin\packages\Intel\hardware\arc32\1.0.4\variants\arduino_101" -MMD -MP -MF".ino.cpp.d" -MT".ino.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "../.ino.cpp"  -o  ".ino.cpp.o"   -Wall
In file included from ../.ino.cpp:6:0:
C:\Users\Scott\Source\Repos\arduino-eclipse-plugin\it.baeyens.arduino.product\target\products\it.baeyens.arduino.product\win32\win32\x86_64\eclipseArduino\arduinoPlugin\packages\Intel\hardware\arc32\1.0.4\cores\arduino/Arduino.h:24:20: fatal error: stdlib.h: No such file or directory
 #include <stdlib.h>
                    ^
compilation terminated.
make: *** [.ino.cpp.o] Error 1
subdir.mk:24: recipe for target '.ino.cpp.o' failed

20:46:21 Build Finished (took 269ms)

scottemberley, 2016-02-11:
The upload process still kicked off and it looks very promising...

starting download script
Args to shell: C:\Users\Scott\Source\Repos\arduino-eclipse-plugin\it.baeyens.arduino.product\target\products\it.baeyens.arduino.product\win32\win32\x86_64\eclipseArduino\arduinoPlugin\tools\Intel\sketchUploader\1.6.4+1.14/x86/bin C:\Users\Scott\Source\Repos\arduino-eclipse-plugin\workspace\test/Release/test.elf COM12
Serial Port PORT
BIN FILE C:\Users\Scott\Source\Repos\arduino-eclipse-plugin\workspace\test/Release/test.bin
Waiting for device... 
Using dfu-util to send  C:\Users\Scott\Source\Repos\arduino-eclipse-plugin\workspace\test/Release/test.bin
C:/Users/Scott/Source/Repos/arduino-eclipse-plugin/it.baeyens.arduino.product/target/products/it.baeyens.arduino.product/win32/win32/x86_64/eclipseArduino/arduinoPlugin/tools/Intel/sketchUploader/1.6.4+1.14/x86/bin/dfu-util.exe -d,8087:0ABA -D C:\Users\Scott\Source\Repos\arduino-eclipse-plugin\workspace\test/Release/test.bin -v --alt 7 -R
dfu-util 0.8

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2014 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to dfu-util@lists.gnumonks.org

No such file or directory: Could not open file C:\Users\Scott\Source\Repos\arduino-eclipse-plugin\workspace\test/Release/test.bin for reading
Sketch will execute in about 5 seconds.
1.6.4+1.14/x86/bin/bash.exe finished
upload done

jantje, 2016-02-11:
The upload error simply tells you the build failed.
As to the build failing. Didn't you get an error/warning that you are to deep in the path?

scottemberley, 2016-02-11:
The upload error was easy to interpret; nothing to send.   It did however show that the changes you made yesterday improved the process.

I do not remember getting an error that the path was too deep.  I will have a chance to try it again a little later today with the build you released yesterday.  I will keep my eyes open for that message.

scottemberley, 2016-02-12:
The upload process is now functioning properly for the Arduino 101

Launching C:\Software\eclipseArduino\arduinoPlugin\tools\Intel\sketchUploader\1.6.4+1.14/x86/bin/bash.exe --noprofile C:\Software\eclipseArduino\arduinoPlugin\tools\Intel\sketchUploader\1.6.4+1.14/clupload/cluploadArduino101_win.sh C:\Software\eclipseArduino\arduinoPlugin\tools\Intel\sketchUploader\1.6.4+1.14/x86/bin C:\Software\eclipseArduino\workspace\TestBlink/Release/TestBlink.elf COM12  
Output:
cygwin warning:
  MS-DOS style path detected: C:\Software\eclipseArduino\arduinoPlugin\tools\Intel\sketchUploader\1.6.4+1.14/clupload/cluploadArduino101_win.sh
  Preferred POSIX equivalent is: /cygdrive/c/Software/eclipseArduino/arduinoPlugin/tools/Intel/sketchUploader/1.6.4+1.14/clupload/cluploadArduino101_win.sh
  CYGWIN environment variable option "nodosfilewarning" turns off this warning.
  Consult the user's guide for more details about POSIX paths:
    http://cygwin.com/cygwin-ug-net/using.html#using-pathnames
starting download script
Args to shell: C:\Software\eclipseArduino\arduinoPlugin\tools\Intel\sketchUploader\1.6.4+1.14/x86/bin C:\Software\eclipseArduino\workspace\TestBlink/Release/TestBlink.elf COM12
Serial Port PORT
BIN FILE C:\Software\eclipseArduino\workspace\TestBlink/Release/TestBlink.bin
Waiting for device... 
Using dfu-util to send  C:\Software\eclipseArduino\workspace\TestBlink/Release/TestBlink.bin
C:/Software/eclipseArduino/arduinoPlugin/tools/Intel/sketchUploader/1.6.4+1.14/x86/bin/dfu-util.exe -d,8087:0ABA -D C:\Software\eclipseArduino\workspace\TestBlink/Release/TestBlink.bin -v --alt 7 -R
dfu-util 0.8

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2014 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to dfu-util@lists.gnumonks.org

Opening DFU capable USB device...
ID 8087:0aba
Run-time device DFU version 0011
Claiming USB DFU Interface...
Setting Alternate Setting #7 ...
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 0011
Device returned transfer size 2048
Copying data from PC to DFU device

Download	[                         ]   0%            0 bytes
Download	[==                       ]  10%         2048 bytes
Download	[=====                    ]  21%         4096 bytes
Download	[=======                  ]  31%         6144 bytes
Download	[==========               ]  42%         8192 bytes
Download	[=============            ]  53%        10240 bytes
Download	[===============          ]  63%        12288 bytes
Download	[==================       ]  74%        14336 bytes
Download	[=====================    ]  85%        16384 bytes
Download	[=======================  ]  95%        17188 bytes
Download	[=========================] 100%        17188 bytes
Download done.
Sent a total of 17188 bytes
state(2) = dfuIDLE, status(0) = No error condition is present
Done!
Resetting USB to switch back to runtime mode
Invalid DFU suffix signature
A valid DFU suffix will be required in a future dfu-util release!!!
can't detach
Sketch will execute in about 5 seconds.
1.6.4+1.14/x86/bin/bash.exe finished
upload done

jantje, 2016-02-12:
@scottemberley 
Thanks for the feedback. I reopen this one as it is "fixed in the nghtly and as such it is not yet fixed for people using the release version.

closed 2016-04-01 by jantje