Sloeber/arduino-eclipse-plugin #373: Eclipse fails to create .bin file for Arundo 101 project

scottemberley, 2016-02-12:
The arduino-builder is failing to create a bin file for a Aruindo 101 project.

jantje, 2016-02-12:
the platform.txt contains the following new recipe which makes the bin file

```
## Create output (.bin file)
recipe.objcopy.bin.pattern="{compiler.path}{compiler.elf2bin.cmd}" {compiler.elf2bin.flags} {compiler.elf2hex.extra_flags} "{build.path}/{build.project_name}.elf" "{build.path}/{build.project_name}.bin"
```

this recipe is not handled by the plugin. Therefore no bin is created.

The recipe also contains following section I have never seen before
```
## Save hex
recipe.output.tmp_file={build.project_name}.hex
recipe.output.save_file={build.project_name}.{build.variant}.hex
```

So I agree that the plugin does not create a bin file. However the platform.txt contains the following upload recipe. 
`tools.arduino101load.upload.pattern=/bin/bash --noprofile "{cmd.path}" "{runtime.tools.sketchUploader-1.6.4+1.14.path}/x86/bin" {build.path}/{build.project_name}.elf {serial.port} "{upload.verbose}"`

As you can see the elf is used. not the bin.
Why is the bin needed? What error do you get?

scottemberley, 2016-02-12:
When running the upload process a call to the Intel sketchUploader is made to perform the actual transfer of the code to the Arduino.  This process generates an "No such file or directory: Could not open file C:\Software\eclipseArduino\workspace\Blink/Release/Blink.bin for reading" error message.  The output from this process is below.

Starting upload
using arduino loader
Starting reset using 1200bps touch process
Continuing to use COM12
Ending reset

Launching C:\Software\eclipseArduino\arduinoPlugin\tools\Intel\sketchUploader\1.6.4+1.14/x86/bin/bash.exe --noprofile C:\Software\eclipseArduino\arduinoPlugin\tools\Intel\sketchUploader\1.6.4+1.14/clupload/cluploadArduino101_win.sh C:\Software\eclipseArduino\arduinoPlugin\tools\Intel\sketchUploader\1.6.4+1.14/x86/bin C:\Software\eclipseArduino\workspace\Blink/Release/Blink.elf COM12  
Output:
cygwin warning:
  MS-DOS style path detected: C:\Software\eclipseArduino\arduinoPlugin\tools\Intel\sketchUploader\1.6.4+1.14/clupload/cluploadArduino101_win.sh
  Preferred POSIX equivalent is: /cygdrive/c/Software/eclipseArduino/arduinoPlugin/tools/Intel/sketchUploader/1.6.4+1.14/clupload/cluploadArduino101_win.sh
  CYGWIN environment variable option "nodosfilewarning" turns off this warning.
  Consult the user's guide for more details about POSIX paths:
    http://cygwin.com/cygwin-ug-net/using.html#using-pathnames
starting download script
Args to shell: C:\Software\eclipseArduino\arduinoPlugin\tools\Intel\sketchUploader\1.6.4+1.14/x86/bin C:\Software\eclipseArduino\workspace\Blink/Release/Blink.elf COM12
Serial Port PORT
BIN FILE C:\Software\eclipseArduino\workspace\Blink/Release/Blink.bin
Waiting for device... 
Using dfu-util to send  C:\Software\eclipseArduino\workspace\Blink/Release/Blink.bin
C:/Software/eclipseArduino/arduinoPlugin/tools/Intel/sketchUploader/1.6.4+1.14/x86/bin/dfu-util.exe -d,8087:0ABA -D C:\Software\eclipseArduino\workspace\Blink/Release/Blink.bin -v --alt 7 -R
dfu-util 0.8

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2014 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to dfu-util@lists.gnumonks.org

No such file or directory: Could not open file C:\Software\eclipseArduino\workspace\Blink/Release/Blink.bin for reading
Sketch will execute in about 5 seconds.
1.6.4+1.14/x86/bin/bash.exe finished
upload done

jantje, 2016-02-12:
I took a look in C:\Software\eclipseArduino\arduinoPlugin\tools\Intel\sketchUploader\1.6.4+1.14/clupload/cluploadArduino101_win.sh and  there they swap from elf to bin.
I need to do some thinking about the best way to launch the extra command in the toolchain.

jantje, 2016-02-12:
I found a quick and easy workaround :-)
in project properties->C/C++ build->Settings tab build step and the following as post build step command
`${A.RECIPE.OBJCOPY.BIN.PATTERN}`
That works fine on my system

jantje, 2016-02-13:
@scottemberley 
Does it work to upload with the workaround?

scottemberley, 2016-02-13:
I was testing that just now.  It does work with the nightly build win64.2016-02-12_03-28-03.

jantje, 2016-02-13:
can you be more specific?
Is the bin generated?

scottemberley, 2016-02-13:
The BIN is generated and the upload process sends usable code to the Arduino 101.  I have been testing with the Blink example and I have been able to change the blink rate, build and upload the project successfully.  Changing the blink rate and seeing that change on the board proves that the updated code is making it successfully to the board.

jantje, 2016-02-13:
Och I had read "it does not work.."
My bad. Good. I really need a good think on how to get this fixed nicely and this takes the pressure off.

jantje, 2016-02-17:
I'm going to release 3.0 before I tackle the final fix for this one as I already have regression #374

scottemberley, 2016-02-17:
OK - thanks for letting me know what to expect in the release.

jantje, 2016-06-22:
I have found a nice solution for this problem :-). As it affects the core build engine (and as such all boards) I want to do some more test before releasing.

jantje, 2016-07-05:
@scottemberley have you been able to test this yet?

closed 2016-08-22 by jantje