Sloeber/arduino-eclipse-plugin #379: failed to upgrade

jantje, 2016-02-21:
Extract from issue #321 which seems not related

When Upgrading my linux system eclipse refuses with this error.
This looks related to the untrained eye

Cannot complete the install because of a conflicting dependency.
  Software being installed: Arduino Eclipse Extensions 3.0.0.201602191959 (it.baeyens.arduino.feature.feature.group 3.0.0.201602191959)
  Software currently installed: Eclipse e4 Rich Client Platform 1.3.101.v20150204-1353 (org.eclipse.e4.rcp.feature.group 1.3.101.v20150204-1353)
  Only one of the following can be installed at once: 
    Core Runtime 3.10.0.v20140318-2214 (org.eclipse.core.runtime 3.10.0.v20140318-2214)
    Core Runtime 3.11.1.v20150903-1804 (org.eclipse.core.runtime 3.11.1.v20150903-1804)
  Cannot satisfy dependency:
    From: Arduino eclipse plugin 3.0.0.201602191959 (it.baeyens.arduino.core 3.0.0.201602191959)
    To: bundle org.eclipse.core.runtime 3.11.1
  Cannot satisfy dependency:
    From: Arduino Eclipse Extensions 3.0.0.201602191959 (it.baeyens.arduino.feature.feature.group 3.0.0.201602191959)
    To: it.baeyens.arduino.core [3.0.0.201602191959]
  Cannot satisfy dependency:
    From: Eclipse e4 Rich Client Platform 1.3.101.v20150204-1353 (org.eclipse.e4.rcp.feature.group 1.3.101.v20150204-1353)
    To: org.eclipse.core.runtime [3.10.0.v20140318-2214]

jantje, 2016-02-22:
I could upgrade now :-)
However due to #321 the plugin did not actually uprade; :-(

`org.osgi.framework.BundleException: Could not resolve module: it.baeyens.arduino.core [1271]
  Another singleton bundle selected: osgi.identity; osgi.identity="it.baeyens.arduino.core"; type="osgi.bundle"; version:Version="2.4.0.201512061821"; singleton:="true"

	at org.eclipse.osgi.container.Module.start(Module.java:434)
	at org.eclipse.osgi.container.ModuleContainer$ContainerStartLevel.dispatchEvent(ModuleContainer.java:1498)
	at org.eclipse.osgi.container.ModuleContainer$ContainerStartLevel.dispatchEvent(ModuleContainer.java:1)
	at org.eclipse.osgi.framework.eventmgr.EventManager.dispatchEvent(EventManager.java:230)
	at org.eclipse.osgi.framework.eventmgr.EventManager$EventThread.run(EventManager.java:340)`

wimjongman, 2016-03-13:
This is caused by version dependencies. 

> Cannot complete the install because of a conflicting dependency.
>  Software being installed: Arduino Eclipse Extensions 3.0.0.201602191959 
>  (it.baeyens.arduino.feature.feature.group 3.0.0.201602191959)

It says you are trying to install the our feature. Which is true.

>  Software currently installed: Eclipse e4 Rich Client Platform 1.3.101.v20150204-1353 (org.eclipse.e4.rcp.feature.group 1.3.101.v20150204-1353)

Then it lists the stuff it has a problem with

>  Only one of the following can be installed at once: 
>    Core Runtime 3.10.0.v20140318-2214 (org.eclipse.core.runtime 3.10.0.v20140318-2214)

Eclipse E4 RCP requires this version

>    Core Runtime 3.11.1.v20150903-1804 (org.eclipse.core.runtime 3.11.1.v20150903-1804)

Arduino plugins require this version

So there is a version deadlock and you cannot install.

## Why is this
The Eclipse you are working with cannot upgrade to the higher runtime version which is required by the Arduino plugins.

## Do the Arduino plugins really require Core Runtime 3.11.1.v20150903-1804 (org.eclipse.core.runtime 3.11.1.v20150903-1804)
No

## How to resolve
When you add a dependency, Eclipse automatically adds the version number of the plugin in the current running IDE.  We need to remove the version number if we do not require a specific eclipse version (most of the times we don't)

## Is this resolved
I have been wading through the manifests and I cannot see any version specifics anymore. So please try again with the latest nightly.

jantje, 2016-03-13:
Wim the status of this issue contains "fixed in nightly" so indeed in the nightly it is fixed :-)

wimjongman, 2016-03-13:
Right, you use tags. I have installed Mylyn and see only "open" or "closed"
issues. Sorry about that.

jantje, 2016-03-13:
yes the right query is
is:issue is:open -label:"status: fixed in nightly"
The reason is: these issues are open for the last stable.
I change the label "fixed in nightly" to fixed in VXYZ" during release.

jantje, 2016-03-13:
I just updated the readme.md in this respect as well

closed 2016-04-01 by jantje