Sloeber/arduino-eclipse-plugin #383: Typo somewhere in your code

jkwilborn, 2016-02-29:
I recieved this error message that seems to indicate there is a type in your code that causes the failure. Here is the output:

[ERROR] Unknown lifecycle phase "varify". You must specify a valid lifecycle phase or a goal in the format <plugin-prefix>:<goal> or <plugin-group-id>:<plugin-artifact-id>[:<plugin-version>]:<goal>. Available lifecycle phases are: validate, initialize, generate-sources, process-sources, generate-resources, process-resources, compile, process-classes, generate-test-sources, process-test-sources, generate-test-resources, process-test-resources, test-compile, process-test-classes, test, prepare-package, package, pre-integration-test, integration-test, post-integration-test, verify, install, deploy, pre-clean, clean, post-clean, pre-site, site, post-site, site-deploy. -> [Help 1]

As you can see, varify is not verify...

Since it error-ed out, I assume it's in a little used code sequence...

Thanks

Jack Wilborn

jantje, 2016-02-29:
Are you sure this is for the arduino eclipse plugin?
Can you provide a screenshot?

jkwilborn, 2016-02-29:
​Jantje, I sure hope so, I've worked on this so long, maybe I've had a
brain fart.  I'll double check it again.. Wouldn't be the first time.  I
get interrupted and sometimes,  when I return, the stack's broken and I'm
lost.

I'll let you know by e-mail and do what's needed in github to correct it.​
Thanks for the quick reply, you people do a wonderful job.  If I ever get
this in and working, I will donate to your fund.

Thanks,

Jack

jkwilborn, 2016-02-29:
Sorry, I sent you the wrong data file.  I will have to hunt.  Is there any way to kill this entry?  Sorry about that, I try and be very specific before filing a bug.  Can't believe I sent you the wrong stuff.  Wonder who I sent it too?

Thanks, sorry about that.

Jack

jantje, 2016-02-29:
no problem

closed 2016-02-29 by jantje