Sloeber/arduino-eclipse-plugin #385: Missing "A.RUNTIME.TOOLS..." enviroment variables for custom hardware

mike-o-tronic, 2016-03-03:
Hi Jantje,

Congratulations first of to v3!!! Looks BIG! I love it!!!

Unfortunately I ran into a little issue with it. I loaded a custom hardware package, but when I try to compile I get an error because the path to _arm-none-eabi-g++_ is wrong. When I change the hardware to Arduino Due (which my board is based on) the compiler works fine. A quick check showed it's missing enviroment variables to the compiler path (see screenshots).

I already created a new workspace and tried to compile an empty arduino project to make sure nothing in my actual workspace or project is causing that.

I compared the hardware packages, but could find anything which would lead to the different behavior. Both packages have _compiler.path={runtime.tools.arm-none-eabi-gcc.path}/bin/_

The package can be found [here](https://github.com/chrishamm/RepRapFirmware/raw/dev/Hardware/package_reprap_index.json).

I couldn't find out where _runtime.tools.arm-none-eabi-gcc.path_ should be defined, but might do a little more research tomorrow. Or do you have any idea, where it is or why these variables go missing?

Best,
Mike

[Build_Error.txt](https://github.com/jantje/arduino-eclipse-plugin/files/156224/Build_Error.txt)
[image]
[image]

jantje, 2016-03-03:
Thanks for the feedback
IMHO You are missing the tools section in the json file.

mike-o-tronic, 2016-03-04:
I see what you mean... though I don't understand the problem. I thought the json file is for the installation of the hardware package components. If the tool is already installed from another package it should work, shouldn't it!?

I did a quick test under win64 with the Arduino IDE. Here I get different results for v1.6.5-R5, which gives me java errors and won't start up after the package installation. And v1.6.7 does install the package and compiles the code perfectly fine, but gives warnings about deprecated recipes. (see attachments)

I guess the definition for those hardware packages is still in development. I will look more into this. For everybody else, who's interested, here are some links I've found so far:
[Customizing the Arduino IDE](http://playground.arduino.cc/Main/CustomizeArduinoIDE)
[Arduino IDE 1.5 3rd party Hardware specification](https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5-3rd-party-Hardware-specification)
[Arduino IDE 1.6.x package_index.json format specification](https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.6.x-package_index.json-format-specification)

Attachments:
[image]
[Duet Install 1.6.5-R5 win64.txt](https://github.com/jantje/arduino-eclipse-plugin/files/158670/Duet.Install.1.6.5-R5.win64.txt)
[Duet Compile 1.6.7 win64.txt](https://github.com/jantje/arduino-eclipse-plugin/files/158671/Duet.Compile.1.6.7.win64.txt)

jantje, 2016-03-04:
I guess there is still movement in the specification but I do not always agree with the Arduino core team and I stopped discussing with them. Here is my point of view (which may  deviate from the Arduino core point of view):
As you have a board on the market with your own json file; why would you want users to first install due and then your board. 
It makes lots more sense to have a json file which not only references a tool but can also install the tools.
Consider following scenario: due gets a tool-chain upgrade. In that case the installation instructions "install latest due and install my board" fail. 
A solution could be to change the installation instructions "install due version XYZ and my board" but then people will complain they can't use the latest due version (as the Arduino IDE does not allow multiple version installs).
A other solution is to quickly update the tool-chain version in your json file. 

IMHO Adding the needed tools sections from the Arduino json file your json file is a far better solution.

Jantje

PS Are you supporting me on patron? https://www.patreon.com/jantje?ty=h
This is the section I think you need to add:
```
"tools": [
        {
          "name": "arm-none-eabi-gcc",
          "version": "4.8.3-2014q1",
          "systems": [
            {
              "host": "arm-linux-gnueabihf",
              "url": "http://downloads.arduino.cc/gcc-arm-none-eabi-4.8.3-2014q1-arm.tar.bz2",
              "archiveFileName": "gcc-arm-none-eabi-4.8.3-2014q1-arm.tar.bz2",
              "checksum": "SHA-256:ebe96b34c4f434667cab0187b881ed585e7c7eb990fe6b69be3c81ec7e11e845",
              "size": "44423906"
            },
            {
              "host": "i686-mingw32",
              "archiveFileName": "gcc-arm-none-eabi-4.8.3-2014q1-windows.tar.gz",
              "url": "http://downloads.arduino.cc/gcc-arm-none-eabi-4.8.3-2014q1-windows.tar.gz",
              "checksum": "SHA-256:fd8c111c861144f932728e00abd3f7d1107e186eb9cd6083a54c7236ea78b7c2",
              "size": "84537449"
            },
            {
              "host": "x86_64-apple-darwin",
              "url": "http://downloads.arduino.cc/gcc-arm-none-eabi-4.8.3-2014q1-mac.tar.gz",
              "archiveFileName": "gcc-arm-none-eabi-4.8.3-2014q1-mac.tar.gz",
              "checksum": "SHA-256:3598acf21600f17a8e4a4e8e193dc422b894dc09384759b270b2ece5facb59c2",
              "size": "52518522"
            },
            {
              "host": "x86_64-pc-linux-gnu",
              "url": "http://downloads.arduino.cc/gcc-arm-none-eabi-4.8.3-2014q1-linux64.tar.gz",
              "archiveFileName": "gcc-arm-none-eabi-4.8.3-2014q1-linux64.tar.gz",
              "checksum": "SHA-256:d23f6626148396d6ec42a5b4d928955a703e0757829195fa71a939e5b86eecf6",
              "size": "51395093"
            },
            {
              "host": "i686-pc-linux-gnu",
              "url": "http://downloads.arduino.cc/gcc-arm-none-eabi-4.8.3-2014q1-linux32.tar.gz",
              "archiveFileName": "gcc-arm-none-eabi-4.8.3-2014q1-linux32.tar.gz",
              "checksum": "SHA-256:ba1994235f69c526c564f65343f22ddbc9822b2ea8c5ee07dd79d89f6ace2498",
              "size": "51029223"
            }
          ]
        },
        {
          "name": "bossac",
          "version": "1.3a-arduino",
          "systems": [
            {
              "host": "i686-linux-gnu",
              "url": "http://downloads.arduino.cc/tools/bossac-1.3a-arduino-i686-linux-gnu.tar.bz2",
              "archiveFileName": "bossac-1.3a-arduino-i686-linux-gnu.tar.bz2",
              "checksum": "SHA-256:d6d10362f40729a7877e43474fcf02ad82cf83321cc64ca931f5c82b2d25d24f",
              "size": "147359"
            },
            {
              "host": "x86_64-pc-linux-gnu",
              "url": "http://downloads.arduino.cc/tools/bossac-1.3a-arduino-x86_64-pc-linux-gnu.tar.bz2",
              "archiveFileName": "bossac-1.3a-arduino-x86_64-pc-linux-gnu.tar.bz2",
              "checksum": "SHA-256:c1daed033251296768fa8b63ad283e053da93427c0f3cd476a71a9188e18442c",
              "size": "26179"
            },
            {
              "host": "i686-mingw32",
              "url": "http://downloads.arduino.cc/tools/bossac-1.3a-arduino-i686-mingw32.tar.bz2",
              "archiveFileName": "bossac-1.3a-arduino-i686-mingw32.tar.bz2",
              "checksum": "SHA-256:a37727622e0f86cb4f2856ad0209568a5d804234dba3dc0778829730d61a5ec7",
              "size": "265647"
            },
            {
              "host": "i386-apple-darwin11",
              "url": "http://downloads.arduino.cc/tools/bossac-1.3a-arduino-i386-apple-darwin11.tar.bz2",
              "archiveFileName": "bossac-1.3a-arduino-i386-apple-darwin11.tar.bz2",
              "checksum": "SHA-256:40770b225753e7a52bb165e8f37e6b760364f5c5e96048168d0178945bd96ad6",
              "size": "39475"
            }
          ]
        }
```

mike-o-tronic, 2016-03-04:
I totally agree with you! Especially for a commercial product a complete package is a must!

My thoughts were more spinning around a private hardware package, where I actually don't want to use the json file for web install. I just want to copy a hardware package into the folder and use already installed tools.

jantje, 2016-03-04:
> My thoughts were more spinning around a private hardware package, where I actually don't want to use the json file for web install. I just want to copy a hardware package into the folder and use already installed tools.

IMHO: The private hardware using "already installed tools" way of working will become harder and harder over time.

mike-o-tronic, 2016-03-05:
I did start from scratch in both systems and now it's working. Both Win64 and Linux find the compiler. I'm not sure what I did different. Maybe it was the restart after installing the package with the tools.

I did get another error while I was setting up the Windows system though. It was saying the zip file was truncated, but continued the installation of the package. When I checked afterwards the package folder was incomplete, but the zip file was completely downloaded. Maybe a timeout occurs, but the installation is continuing with half the zip file?

closed 2016-03-05 by mike-o-tronic

mike-o-tronic, 2016-03-05:
Ok... maybe I was too quick closing it. I ran into that error again. I don't know how I created this error in the first place, but after a lot of testing I found out what's the issue. Something is happening when the eclipse is downloading or updating the json file.

So just placing the json in arduinoPlugin folder and the hardware package in the appropriate folder doesn't work. Which is an issue if you want to quick local testing like duplicating a package. Or maybe restoring a package while being offline. At least that's what I thought... file://[path] to the backup or development location does the job :wink: