Sloeber/arduino-eclipse-plugin #389: Add category to update site

wimjongman, 2016-03-13:
The repository does not have a category.

wimjongman, 2016-03-13:
Ah, I see. You deleted the update site. I remember that I saw something flying by but I think we need to revert this. Can you explain again why you removed it?

4b9c34dd3afaf46e0e4ceb4a84e6e2a79740105e

jantje, 2016-03-13:
Because the update site did not contain the info needed for updating the product.
On the other hand the product build also creates a update site that does contain all info at it.baeyens.arduino.product/target/repository/
So I switched to using this update site which makes the update site project was no longer used.

jantje, 2016-03-13:
The update site never had a category by the way. All instructions to install state you have to unclick "group by category"

wimjongman, 2016-03-13:
I see. I think we need a proper category. I will play with it a little bit.

wimjongman, 2016-03-13:
[image]

jantje, 2016-03-13:
You did it :-)

universpharmacie, 2016-03-13:
I don't know if it's in the right place here, but what about these annoying warnings about unsigned content each time when you install an update?

jantje, 2016-03-13:
>I don't know if it's in the right place here, but what about these annoying warnings about unsigned content each time when you install an update?

fixing this costs money. lots of money.
imho the alternative (self signed) is not better

universpharmacie, 2016-03-13:
Ah, ok, sorry. I'll already close my mouth.

jantje, 2016-03-13:
No problem. Wim explained me less than a month ago on the developers list.

jantje, 2016-03-26:
Wim do I need to copy  the update site now from the update site again?
Now I copy the update site from the product.

wimjongman, 2016-03-26:
Yes from the update site. The one from the product is in raw format.

jantje, 2016-03-27:
OK I'll fix that

jantje, 2016-03-28:
I changed to the update site. I cleaned the whole nightly so no old things were still lingering around.
But I see 2 things I don't like:
1) The product V3 is no longer there (so people will not be able to upgrade V2.4 to V3 via the update site)
2) There are still c++ downloads in the list
[image]

wimjongman, 2016-03-28:
Yes, the product include is an interesting thing. I guess it is harmless to include it in the feature. I will make the changes.

jantje, 2016-03-28:
Should it not be a different feature?

wimjongman, 2016-03-29:
When the product is installed in for example Eclipse CC then the user can also run it in Arduino Dev mode with -product xxxxx

jantje, 2016-03-29:
>When the product is installed in for example Eclipse CC then the user can also run it in Arduino Dev mode with -product xxxxx

Am I missing something?

wimjongman, 2016-03-29:
I have a hard time including the product in the update site.

jantje, 2016-03-29:
Then forget about it.

wimjongman, 2016-03-30:
It is not need for a V2 to V3 product upgrade. I have tested it and found no problems (apart from the fact that bash is installed on my system ;)

jantje, 2016-03-30:
another test on bash :-) 
Maybe the test is to restrictive but given the impact I prefer it this way.

wimjongman, 2016-03-30:
I would like the text to change so that people are more informed.

jantje, 2016-03-30:
I'm open to suggestions.

wimjongman, 2016-03-31:
"we have found programs in the path that might conflict with our external
builder. The conflicting programs are "xxxx" and "yyyy". The program might
still function but if you get strange build errors you know where to look"

[x] Don't show this message again

jantje, 2016-03-31:
I finally finished the upgrade video.
Just in time ;-)
https://www.youtube.com/watch?v=h-rz2FR3H6Y

closed 2016-04-01 by jantje